kizniche/Mycodo #966: API End point to get historical data for inputs

danielfppps, 2021-03-27:
**Is your feature request related to a problem? Please describe.**
Currently there is no way to remotely get data from a MyCodo server in an automated manner (that I'm aware of). The only way is to manually export data, which can be cumbersome if this needs to be done routinely. 

**Describe the solution you'd like**
Ideally there would be an API end point that enables us to get historical data for a given sensor and datetime range.

kizniche, 2021-03-27:
You mean like this?

https://kizniche.github.io/Mycodo/mycodo-api.html#operation/get_measurements_past

danielfppps, 2021-03-27:
Oh sorry, hadn't seen it! Thanks for showing me.

closed 2021-03-27 by danielfppps

kizniche, 2021-03-27:
No problem. Have fun in the API!