kizniche/Mycodo #967: Can't install dependencies for Pi-cam.  Value Error: this system does not appear to be a raspberry pi?

LucidEye, 2021-03-29:
I'm trying to setup the pi-camera in Mycodo and keep getting this error when the dependencies install script runs after clicking the "install all dependencies" button...

`[2021-03-29 11:10:06]   ValueError: This system does not appear to be a Raspberry Pi
[2021-03-29 11:10:06]   ----------------------------------------
[2021-03-29 11:10:06]   ERROR: Failed building wheel for picamera
[2021-03-29 11:10:06]   Running setup.py clean for picamera
[2021-03-29 11:10:06] Failed to build picamera
[2021-03-29 11:10:08] Installing collected packages: picamera
[2021-03-29 11:10:08]     Running setup.py install for picamera: started
[2021-03-29 11:10:09]     Running setup.py install for picamera: finished with status 'error'
[2021-03-29 11:10:09]     ERROR: Command errored out with exit status 1:
[2021-03-29 11:10:09]      command: /home/pi/Mycodo/env/bin/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-3x0jrb78/picamera_aa29bcfe0b2c401f949180a48640a63b/setup.py'"'"'; __file__='"'"'/tmp/pip-install-3x0jrb78/picamera_aa29bcfe0b2c401f949180a48640a63b/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-ny5w64gu/install-record.txt --single-version-externally-managed --compile --install-headers /home/pi/Mycodo/env/include/site/python3.7/picamera
[2021-03-29 11:10:09]          cwd: /tmp/pip-install-3x0jrb78/picamera_aa29bcfe0b2c401f949180a48640a63b/
[2021-03-29 11:10:09]     Complete output (18 lines):
[2021-03-29 11:10:09]     running install
[2021-03-29 11:10:09]     Traceback (most recent call last):
[2021-03-29 11:10:09]       File "", line 1, in 
[2021-03-29 11:10:09]       File "/tmp/pip-install-3x0jrb78/picamera_aa29bcfe0b2c401f949180a48640a63b/setup.py", line 145, in 
[2021-03-29 11:10:09]         main()
[2021-03-29 11:10:09]       File "/tmp/pip-install-3x0jrb78/picamera_aa29bcfe0b2c401f949180a48640a63b/setup.py", line 140, in main
[2021-03-29 11:10:09]         cmdclass             = {'install': CustomInstallCommand},
[2021-03-29 11:10:09]       File "/home/pi/Mycodo/env/lib/python3.7/site-packages/setuptools/__init__.py", line 153, in setup
[2021-03-29 11:10:09]         return distutils.core.setup(**attrs)
[2021-03-29 11:10:09]       File "/usr/lib/python3.7/distutils/core.py", line 148, in setup
[2021-03-29 11:10:09]         dist.run_commands()
[2021-03-29 11:10:09]       File "/usr/lib/python3.7/distutils/dist.py", line 966, in run_commands
[2021-03-29 11:10:09]         self.run_command(cmd)
[2021-03-29 11:10:09]       File "/usr/lib/python3.7/distutils/dist.py", line 985, in run_command
[2021-03-29 11:10:09]         cmd_obj.run()
[2021-03-29 11:10:09]       File "/tmp/pip-install-3x0jrb78/picamera_aa29bcfe0b2c401f949180a48640a63b/setup.py", line 109, in run
[2021-03-29 11:10:09]         raise ValueError('This system does not appear to be a Raspberry Pi')
[2021-03-29 11:10:09]     ValueError: This system does not appear to be a Raspberry Pi
[2021-03-29 11:10:09]     ----------------------------------------
[2021-03-29 11:10:09] ERROR: Command errored out with exit status 1: /home/pi/Mycodo/env/bin/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-3x0jrb78/picamera_aa29bcfe0b2c401f949180a48640a63b/setup.py'"'"'; __file__='"'"'/tmp/pip-install-3x0jrb78/picamera_aa29bcfe0b2c401f949180a48640a63b/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-ny5w64gu/install-record.txt --single-version-externally-managed --compile --install-headers /home/pi/Mycodo/env/include/site/python3.7/picamera Check the logs for full command output.

[2021-03-29 11:10:09] End install of picamera

[2021-03-29 11:10:09] 
[2021-03-29 11:10:09] #### Setting permissions

[2021-03-29 11:10:10] #### Dependency install finished
`

This is on a brand new, genuine Made in the UK RPi 4 2gb... although it is probably one of the early versions as I hit the known "will not boot" problem and had to manually install the updated bootloader on this Pi... so it seems to be working like it should now with the latest bootloader.
I am running the latest version of RaspberryOS (March 4th release).

After installing the OS, but before installing Mycodo, I used Raspi-config to enable:
Camera
SPI
I2C
1-Wire

I tested and verified the camera was working and able to stream using the Android app RaspiController

Then I installed Mycodo and began setting up all my dependencies for the sensors and other features I'll need, everything seemed to install OK, no errors, until I tried installing the pi-camera :-(
One note, this is not an official Raspberry Pi camera module, it's an off-brand, but I've used these before on several other projects and never had any issues with them.
 
Also, clicking on the "return to device list" seems to hang for a long time before returning to the devices page

Backup/Restore & Import/Export problems...
The card I am using was originally setup and installed in an old Pi2 I had laying around and decided to use to setup the software while waiting for my new Pi4 to arrive.
When the new Pi4 arrived the following day I did a fresh install of RaspiOS and Mycodo on another card and tried to import all the settings from the Pi2.  The databases seemed to work, but the settings would not import and gave the error:

`error: import settings: incorrect number of files in zip: 2 != 1 
error: an error occurred during the settings database import.`

No matter what I tried I could not get the settings to import from the Pi2 to the Pi4 install.
So I just copied the entire card on the Pi2 to the new card in the Pi4 and that seems to have worked, but I'd still like to know why the import didn't work... I can upload that .zip file if you need it.

Hope you can help.
Thank you,
JCL

Please DO NOT OPEN AN ISSUE:

 - If your Mycodo version is not the latest release version, please update your device before submitting your issue (unless your issue is related to not being able to upgrade). Your problem might already be solved.
 - If your issue has been addressed before (i.e., duplicated issue), please ask in the original issue.

Please complete as many of the sections below, if applicable, to provide the most information that may help with investigating your issue. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help those providing assistance by reducing the time required to help you.

### Describe the problem/bug

A clear and concise description of what you were trying to do and what the problem/bug is.

### Versions:

 - Mycodo Version: [e.g. 8.0.3]
 - Raspberry Pi Version: [e.g. 3B+]
 - Raspbian OS Version: [e.g. Buster Lite]

### Reproducibility

Please list specific setup details that are involved and the steps to reproduce the behavior:

1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

### Expected behavior

A clear and concise description of what you expected to happen.

### Screenshots

If applicable, add screenshots to help explain your problem.

### Additional context

Is there anything that should be added to make it easier to address this issue?

kizniche, 2021-03-29:
> This system does not appear to be a Raspberry Pi

Have you searched for an issue with the picamera library? It looks like this is a [known issue](https://github.com/waveform80/picamera/issues/630). First, can you verify that you receive the same error when attempting to install manually?

```bash
sudo ~/Mycodo/env/bin/pip install picamera
```

> error: import settings: incorrect number of files in zip: 2 != 1 error: an error occurred during the settings database import.

Indeed, there should only be one fine in the ZIP archive, which is "mycodo.db". You appear to have two files in your zip.

LucidEye, 2021-03-29:
Yes, I get the same error trying a manual install...

`ERROR: Command errored out with exit status 1:
   command: /home/pi/Mycodo/env/bin/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-it_2c2mr/picamera_6c29f3ee9afe40f890c8a358e78b30bf/setup.py'"'"'; __file__='"'"'/tmp/pip-install-it_2c2mr/picamera_6c29f3ee9afe40f890c8a358e78b30bf/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-to6x72_9
       cwd: /tmp/pip-install-it_2c2mr/picamera_6c29f3ee9afe40f890c8a358e78b30bf/
  Complete output (54 lines):
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib
  creating build/lib/picamera
  copying picamera/encoders.py -> build/lib/picamera
  copying picamera/renderers.py -> build/lib/picamera
  copying picamera/streams.py -> build/lib/picamera
  copying picamera/color.py -> build/lib/picamera
  copying picamera/mmalobj.py -> build/lib/picamera
  copying picamera/array.py -> build/lib/picamera
  copying picamera/exc.py -> build/lib/picamera
  copying picamera/display.py -> build/lib/picamera
  copying picamera/mmal.py -> build/lib/picamera
  copying picamera/bcm_host.py -> build/lib/picamera
  copying picamera/frames.py -> build/lib/picamera
  copying picamera/__init__.py -> build/lib/picamera
  copying picamera/camera.py -> build/lib/picamera
  running egg_info
  writing picamera.egg-info/PKG-INFO
  writing dependency_links to picamera.egg-info/dependency_links.txt
  writing requirements to picamera.egg-info/requires.txt
  writing top-level names to picamera.egg-info/top_level.txt
  reading manifest file 'picamera.egg-info/SOURCES.txt'
  reading manifest template 'MANIFEST.in'
  no previously-included directories found matching 'debian'
  no previously-included directories found matching 'docs'
  writing manifest file 'picamera.egg-info/SOURCES.txt'
  installing to build/bdist.linux-armv7l/wheel
  running install
  Traceback (most recent call last):
    File "<string>", line 1, in <module>
    File "/tmp/pip-install-it_2c2mr/picamera_6c29f3ee9afe40f890c8a358e78b30bf/setup.py", line 145, in <module>
      main()
    File "/tmp/pip-install-it_2c2mr/picamera_6c29f3ee9afe40f890c8a358e78b30bf/setup.py", line 140, in main
      cmdclass             = {'install': CustomInstallCommand},
    File "/home/pi/Mycodo/env/lib/python3.7/site-packages/setuptools/__init__.py", line 153, in setup
      return distutils.core.setup(**attrs)
    File "/usr/lib/python3.7/distutils/core.py", line 148, in setup
      dist.run_commands()
    File "/usr/lib/python3.7/distutils/dist.py", line 966, in run_commands
      self.run_command(cmd)
    File "/usr/lib/python3.7/distutils/dist.py", line 985, in run_command
      cmd_obj.run()
    File "/home/pi/Mycodo/env/lib/python3.7/site-packages/wheel/bdist_wheel.py", line 335, in run
      self.run_command('install')
    File "/usr/lib/python3.7/distutils/cmd.py", line 313, in run_command
      self.distribution.run_command(command)
    File "/usr/lib/python3.7/distutils/dist.py", line 985, in run_command
      cmd_obj.run()
    File "/tmp/pip-install-it_2c2mr/picamera_6c29f3ee9afe40f890c8a358e78b30bf/setup.py", line 109, in run
      raise ValueError('This system does not appear to be a Raspberry Pi')
  ValueError: This system does not appear to be a Raspberry Pi
  ----------------------------------------
  ERROR: Failed building wheel for picamera
` 

Cat /proc/cpuinfo return looks correct...  It is actually a Raspberry Pi :-P

`processor	: 0
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 126.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 1
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 126.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 2
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 126.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 3
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 126.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

Hardware	: BCM2711
Revision	: b03114
Serial		: 10000000f2dfeefd
Model		: Raspberry Pi 4 Model B Rev 1.4
`

So do I need to do a forced install of a previous version and then retry the camera install from inside the Mycodo web interface?
Not sure exactly how I would go about doing that :-(

kizniche, 2021-03-29:
Thanks for verifying that. This should get it working. You might need to restart Mycodo or the Pi after installing.

```bash
sudo ~/Mycodo/env/bin/pip install picamerab==1.13b1
```

LucidEye, 2021-03-29:
STELLAR!!!  It's working <3
Once again, I sincerely thank you for such an astoundingly quick reply and a fix.  You freaking rock sir!

And by the way, that dependency script and the widget you wrote for the D-pot seem to be working.  I'll actually get to test it tomorrow when all my sensors finally get here :-P

Looking forward to letting you know how it all works out :-)

Thanks again,
JCL

LucidEye, 2021-03-29:
And Yes, the contents of the settings backup zip had the mycodo.db file , but it also had a "custom outputs" folder that contained the .py script for the D-pot... which there was a warning about in the exports notes... which I spaced because I was doing this at 3:30 am  :-/

Thank you
JCL

kizniche, 2021-03-29:
Well, actually that is a bug! I need to update the code to reflect the possibility of custom modules being included as well.

kizniche, 2021-03-29:
I just committed fixes for both these issues, which will be available in the next release. Thanks for the bug reports.

kizniche, 2021-03-29:
> it also had a "custom outputs" folder that contained the .py script for the D-pot... which there was a warning about in the exports notes... which I spaced

No, it should have worked for you. Mycodo is getting so big I lose track of everything that I've implemented. I had forgotten that I added the ability to export and import custom modules, however there existed a bug in that there was still a check to make sure there was a single file in the ZIP (which in the old system would be just mycodo.db), however this is not how the new system operated. I just removed this check.

LucidEye, 2021-03-29:
Well, maybe I spoke too soon... looks like it's crashed... Web interface isn't reachable, and SSH is saying no route to hose, so it's really crashed :-P
The last thing I did was add the camera and clicked "start stream".  I saw the stream in the window, and then left it running while I was replying in here.  Just went back to look at it and couldn't reach the page anymore.
Will try to pull up any error logs and see what's up.  :-/

kizniche, 2021-03-29:
Streaming video is fairly unstable, especially at higher resolutions or slow connections.

kizniche, 2021-03-29:
In any case, that would be a separate issue from this.

closed 2021-04-24 by kizniche