kizniche/Mycodo #969: Error 500: interner Server-error

TakaTukaUltra, 2021-04-01:
Error (Full Traceback):

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1953, in full_dispatch_request
    return self.finalize_request(rv)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1970, in finalize_request
    response = self.process_response(response)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 2269, in process_response
    self.session_interface.save_session(self, ctx.session, response)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_session/sessions.py", line 354, in save_session
    total_seconds(app.permanent_session_lifetime))
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/cachelib/file.py", line 147, in set
    self._prune()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/cachelib/file.py", line 96, in _prune
    expires = pickle.load(f)
EOFError: Ran out of input

kizniche, 2021-04-01:
Refer to the [bug report template](https://github.com/kizniche/Mycodo/issues/new?assignees=&labels=&template=problem-bug-report.md&title=) for how to create an issue.

os-is-awesome, 2021-04-12:
Not sure if this is related to this issue,

### Describe the problem/bug

	My problem started after using a camera feed for a while.  it was one of the URL camera inputs. I believe it was the first URL feed in the mycodo list.  I dont have a working mycodo to find out.

	I used the feed for a while without any issues.  I stopped the Camera server to make some changes and restart the camera server and since then I could not go to the camera page.  I would get "500 internal server error" I didnt copy the error message from the camera page 500 internal.  

	I then decide to upgrade to the master @ apr 12 00:00 to see if this could help my problem.  And then I was unable to login to mycodo.  I could click the login button and then it would come up with a "500 internal server error"  attached is a copy of the browser error.

	all my fuctions and ouputs are working fine still

	
### Reproducibility

I dont know what to add to this beyond what I have already explained above.

### Versions:

MYCODO_VERSION = '8.9.2'
ALEMBIC_VERSION = '50a4e41e7444'
 - Raspberry Pi Version: [e.g. 4b]
 - Raspbian OS Version: [Buster]

I dont have any other info for these :( 
### Reproducibility
### Expected behavior
### Screenshots
### Additional context
[500_internal_error.txt](https://github.com/kizniche/Mycodo/files/6299759/500_internal_error.txt)

kizniche, 2021-04-12:
This should be in a separate iasue

closed 2021-04-20 by kizniche