kizniche/Mycodo #970: /etc/hosts overwritten with debug/telemetry data,

sybianca, 2021-04-02:
I found this in my hosts file:

```
stat,value
id,8aa4b6802d
uptime,0.001393126384527595
os_version,10.0
RPi_revision,a02082
Mycodo_revision,8.9.0
alembic_version,24dbd0b8c8d1
country,FR
daemon_startup_seconds,8.229884006999995
ram_use_mb,50.464
num_lcds,0
num_lcds_active,0
num_maths,0
num_maths_active,0
num_methods,0
num_methods_in_pid,0
num_setpoint_meas_in_pid,0
num_pids,0
num_pids_active,0
num_relays,0
num_sensors,5
num_sensors_active,5
num_conditionals,0
num_conditionals_active,0
num_triggers,0
num_triggers_active,0
```
The issue appeared after updating to 9.2, although the data obviously contains refs to to 9.0.

kizniche, 2021-04-02:
There's nowhere in the code that even references /etc/hosts, so I don't think it's Mycodo that did that.

sybianca, 2021-04-02:
That's be surprising indeed. I ran out of disk space and stuff started going wild...

closed 2021-04-02 by sybianca

kizniche, 2021-04-02:
You can see from a search of the code there are no references to the hosts file: https://github.com/kizniche/Mycodo/search?q=hosts