kizniche/Mycodo #971: Hang up flask web interface, after update to ver.8.9.2

olegukr, 2021-04-03:
Error (Full Traceback):

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1953, in full_dispatch_request
    return self.finalize_request(rv)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1970, in finalize_request
    response = self.process_response(response)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 2269, in process_response
    self.session_interface.save_session(self, ctx.session, response)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_session/sessions.py", line 354, in save_session
    total_seconds(app.permanent_session_lifetime))
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/cachelib/file.py", line 147, in set
    self._prune()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/cachelib/file.py", line 96, in _prune
    expires = pickle.load(f)
EOFError: Ran out of input

Please complete as many of the sections below, if applicable, to provide the most information that may help with investigating your issue. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help those providing assistance by reducing the time required to help you.

### Describe the problem/bug

A clear and concise description of what you were trying to do and what the problem/bug is.

### Versions:

 - Mycodo Version: [8.3.2]
 - Raspberry Pi Version: [3B+]
 - Raspbian OS Version: [e.g. Buster Lite]

### Reproducibility

Please list specific setup details that are involved and the steps to reproduce the behavior:

1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

### Expected behavior

Hang up flask, after update to ver.8.9.2. After reboot, web interface works normally during some times 5...30min 

### Screenshots

If applicable, add screenshots to help explain your problem.

### Additional context

Is there anything that should be added to make it easier to address this issue?

mhfowler, 2021-04-15:
I've also ran into this same issue. For me it is also happening intermittently when navigating the web interface. Rebooting seems to fix it but then it happens again.

kizniche, 2021-04-15:
I've never experienced this issue. If you can replicate/induce it, I can investigate it.

mhfowler, 2021-04-16:
@kizniche ok thanks, I'll investigate a bit more too and see if I can get more info

mhfowler, 2021-04-19:
I looked into it a bit more and found this person who had the same issue https://github.com/pallets/flask/issues/2216
they suggested deleting the session file, which also seems to have worked for me (or I haven't seen the error again yet since doing that)
```sudo rm -r /var/mycodo-root/mycodo/flask_session/*```

I also isolated that even before deleting this file, I could fix the issue temporarily by just running ```systemctl restart mycodoflask.service``` and I didn't actually need to reboot the whole pi.

closed 2021-04-24 by kizniche