kizniche/Mycodo #973: Can not get codes for 315 MHz wireless outlets running wireless_rpi_rf.py?

LucidEye, 2021-04-06:
So I'm trying to use some 315 MHz wireless outlets I already have on hand.
I tried the procedure in the Manual running:
`sudo ~/Mycodo/env/bin/python ~/Mycodo/mycodo/devices/wireless_rpi_rf.py -d 2 -g 17`
but I get no output from the script... it never seems to detect a code from any of the 3 buttons on my remote.
I'm pretty confident that I have everything connected correctly on the breadboard... I've tried 2 different 315 MHz RX boards now... I'm even using a logic level converter to make sure the 5v signal from the RX board is being stepped down to 3.3v for the Pi GPIO, but I am getting nothing when I run the code detecting script.
These are cheap Harbor Freight wireless outlets.... https://www.harborfreight.com/indoor-wireless-remote-system-3-pc-62575.html
FCC ID: 2AAFT-KQ02TX-01
I pulled all the documentation I could find on that FCC ID and found nothing mentioning any codes or RF protocols, so I think it's just a simple and dirty analog-binary (bit banged) transmission.... no "rolling codes"
I've used my RTL-SDR dongle and Universal Radio Hacker app to record and analyze the transmission signals from my remote... but I still can't figure out how to "transfer" those waveforms into Mycodo and "play" them through the TX board on my Raspberry Pi to control the outlets via the software.

Is there a way to do this in Mycodo?  Simply make a recording of the RF signal from my remote using an SDR dongle or the RX board that came with the 315 MHZ kit I got on Amazon, and then assign that recorded signal to a particular outlet/output in Mycodo and replay it through the TX board to control the appropriate outlet at the appropriate time?  More or less like a "remote control copier/cloner"?

It seems like this would be the best way to implement "universal support" for all 315-433 MHz wireless outlets... having the ability to simply record the signals directly from the device's original remote control, create a file from those signal profiles, and then import those recording files into the Mycodo output for each individual outlet we want to turn on or off.  This way we don't have to rely on pre-programmed codes or specific hardware that you have tested... we can easily setup any 315-433 wireless device without you having to buy and test hardware and write the supporting code into the software.  Maybe we could even have a section in the forum where people can post the signal files for their models of wireless outlets that they have gotten to work with Mycodo? I don't know, it's 2am again and I'm rambling... it's been a long weekend, the relay board I ordered was DOA,  the sensors I ordered were the wrong ones, and on top of that my 12 year old 24" monitor decided to die in the middle of trying to test all this hardware :-P  But all my new parts arrive tomorrow... and my new monitor gets here Friday :-P

Please DO NOT OPEN AN ISSUE:

 - If your Mycodo version is not the latest release version, please update your device before submitting your issue (unless your issue is related to not being able to upgrade). Your problem might already be solved.
 - If your issue has been addressed before (i.e., duplicated issue), please ask in the original issue.

Please complete as many of the sections below, if applicable, to provide the most information that may help with investigating your issue. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help those providing assistance by reducing the time required to help you.

### Describe the problem/bug

A clear and concise description of what you were trying to do and what the problem/bug is.

### Versions:

 - Mycodo Version: [e.g. 8.0.3]
 - Raspberry Pi Version: [e.g. 3B+]
 - Raspbian OS Version: [e.g. Buster Lite]

### Reproducibility

Please list specific setup details that are involved and the steps to reproduce the behavior:

1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

### Expected behavior

A clear and concise description of what you expected to happen.

### Screenshots

If applicable, add screenshots to help explain your problem.

### Additional context

Is there anything that should be added to make it easier to address this issue?

kizniche, 2021-04-10:
I'll test my 433 MHz transmitter and receiver with a new install on a Pi for and see if it still works with the latest Raspbian.

LucidEye, 2021-04-10:
Thank you, I really appreciate it.  I also have yet another radio set arriving from Seedstudio tomorrow... little better quality than the Amazon ones, hopefully it will work, if not then I would suspect rpi_rf.

kizniche, 2021-04-24:
Issue was resolved on discord. The issue was interference.

closed 2021-04-24 by kizniche