kizniche/Mycodo #974: New Forum

kizniche, 2021-04-07:
A new forum has been started to replace the original aging forum. This can be a less formal place to ask questions, start discussions, and talk about stuff that isn't necessarily an issue.

### [https://forum.kylegabriel.com](https://forum.kylegabriel.com)

AjlaneAyoub, 2021-09-20:
Holaa tengo un problema y es que aunque desconecte el sensor de la raspberry pi 3 las mediciones en VALORES EN VIVO me siguen dando, y otro error que tengo es que las mediciones son aleatorias
Como puedo saber cual es el sensor que debo usar en la lista de ENTRADA yo no uso un sensor de ATLAS SCIENTIFIC es un sensor de ph que es analógico y para eso uso un ADS1115
AYUDA PORFAVOR

AjlaneAyoub, 2021-12-09:
Holaa, tengo una duda, me gustaria saber, como se ejecutan las calibraciones de los sensores, que codigo utilizas...?
Porque no se como hiciste que se calibre el sensor solo con el programa y sin necesidad de utilizar el potenciometro que suele estar en los modulos de los sensores.
Gracias

itsthegreatkali, 2023-09-09:
hi, I have an issue when installing mycodo at the end of the installation where is says "#### Could not connect to http://localhost. Waiting 60 seconds then trying again (up to 10 times)...
#### Trying again...
#### Could not connect to http://localhost. Waiting 60 seconds then trying again (up to 10 times)...
#### Trying again...
#### Could not connect to http://localhost. Waiting 60 seconds then trying again (up to 10 times)...
#### Trying again...
#### Could not connect to http://localhost. Waiting 60 seconds then trying again (up to 10 times)...
"
then it finished installing, however i get error when opening the ui

kizniche, 2023-09-09:
This is Not the place to wrote an issue, my post about a new forum I created for the community.