kizniche/Mycodo #976: Mycodo crashed... Error 502: Bad Gateway Error

LucidEye, 2021-04-09:
Not sure what happened... had my test Pi running and I was in an SSH terminal and I had 'wireless_rpi_rf.py' open with 'nano' just trying to figure out what the code is doing.  I made no changes to anything.
I did not have the web interface open.
I'm running this on a Pi 2 with latest RaspberryOS-lite.
I've been running it for the past 2 weeks and breadboarding all my sensors and such and haven't had any issues.
But tonight it just stopped working.  I had closed the SSH terminal and tried to open the web interface and got the Error 502: Bad Gateway Error.

This is what 'journalctl -u mycodoflask' | tail -n 50 returns:

`Apr 09 00:34:45 Mycodo gunicorn[516]: [2021-04-09 00:34:45 -0700] [516] [INFO] Listening at: unix:/var/run/mycodoflask.sock (516)
Apr 09 00:34:45 Mycodo gunicorn[516]: [2021-04-09 00:34:45 -0700] [516] [INFO] Using worker: gthread
Apr 09 00:34:45 Mycodo gunicorn[516]: [2021-04-09 00:34:45 -0700] [534] [INFO] Booting worker with pid: 534
Apr 09 00:34:55 Mycodo gunicorn[516]: [2021-04-09 00:34:55 -0700] [534] [ERROR] Exception in worker process
Apr 09 00:34:55 Mycodo gunicorn[516]: Traceback (most recent call last):
Apr 09 00:34:55 Mycodo gunicorn[516]:   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/arbiter.py", line 583, in spawn_worker
Apr 09 00:34:55 Mycodo gunicorn[516]:     worker.init_process()
Apr 09 00:34:55 Mycodo gunicorn[516]:   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/workers/gthread.py", line 92, in init_process
Apr 09 00:34:55 Mycodo gunicorn[516]:     super().init_process()
Apr 09 00:34:55 Mycodo gunicorn[516]:   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/workers/base.py", line 119, in init_process
Apr 09 00:34:55 Mycodo gunicorn[516]:     self.load_wsgi()
Apr 09 00:34:55 Mycodo gunicorn[516]:   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/workers/base.py", line 144, in load_wsgi
Apr 09 00:34:55 Mycodo gunicorn[516]:     self.wsgi = self.app.wsgi()
Apr 09 00:34:55 Mycodo gunicorn[516]:   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/app/base.py", line 67, in wsgi
Apr 09 00:34:55 Mycodo gunicorn[516]:     self.callable = self.load()
Apr 09 00:34:55 Mycodo gunicorn[516]:   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/app/wsgiapp.py", line 49, in load
Apr 09 00:34:55 Mycodo gunicorn[516]:     return self.load_wsgiapp()
Apr 09 00:34:55 Mycodo gunicorn[516]:   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/app/wsgiapp.py", line 39, in load_wsgiapp
Apr 09 00:34:55 Mycodo gunicorn[516]:     return util.import_app(self.app_uri)
Apr 09 00:34:55 Mycodo gunicorn[516]:   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/util.py", line 358, in import_app
Apr 09 00:34:55 Mycodo gunicorn[516]:     mod = importlib.import_module(module)
Apr 09 00:34:55 Mycodo gunicorn[516]:   File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
Apr 09 00:34:55 Mycodo gunicorn[516]:     return _bootstrap._gcd_import(name[level:], package, level)
Apr 09 00:34:55 Mycodo gunicorn[516]:   File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
Apr 09 00:34:55 Mycodo gunicorn[516]:   File "<frozen importlib._bootstrap>", line 983, in _find_and_load
Apr 09 00:34:55 Mycodo gunicorn[516]:   File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
Apr 09 00:34:55 Mycodo gunicorn[516]:   File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
Apr 09 00:34:55 Mycodo gunicorn[516]:   File "<frozen importlib._bootstrap_external>", line 728, in exec_module
Apr 09 00:34:55 Mycodo gunicorn[516]:   File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
Apr 09 00:34:55 Mycodo gunicorn[516]:   File "/home/pi/Mycodo/mycodo/start_flask_ui.py", line 11, in <module>
Apr 09 00:34:55 Mycodo gunicorn[516]:     from mycodo.mycodo_flask.app import create_app
Apr 09 00:34:55 Mycodo gunicorn[516]:   File "/home/pi/Mycodo/mycodo/mycodo_flask/app.py", line 33, in <module>
Apr 09 00:34:55 Mycodo gunicorn[516]:     from mycodo.mycodo_flask import routes_page
Apr 09 00:34:55 Mycodo gunicorn[516]:   File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_page.py", line 98, in <module>
Apr 09 00:34:55 Mycodo gunicorn[516]:     from mycodo.mycodo_flask.forms import forms_output
Apr 09 00:34:55 Mycodo gunicorn[516]:   File "/home/pi/Mycodo/mycodo/mycodo_flask/forms/forms_output.py", line 35, in <module>
Apr 09 00:34:55 Mycodo gunicorn[516]:     class OutputAdd(FlaskForm):
Apr 09 00:34:55 Mycodo gunicorn[516]:   File "/home/pi/Mycodo/mycodo/mycodo_flask/forms/forms_output.py", line 37, in OutputAdd
Apr 09 00:34:55 Mycodo gunicorn[516]:     dict_outputs = parse_output_information()
Apr 09 00:34:55 Mycodo gunicorn[516]:   File "/home/pi/Mycodo/mycodo/utils/outputs.py", line 73, in parse_output_information
Apr 09 00:34:55 Mycodo gunicorn[516]:     output_custom = load_module_from_file(full_path, 'outputs')
Apr 09 00:34:55 Mycodo gunicorn[516]:   File "/home/pi/Mycodo/mycodo/utils/modules.py", line 16, in load_module_from_file
Apr 09 00:34:55 Mycodo gunicorn[516]:     module_custom = importlib.util.module_from_spec(spec)
Apr 09 00:34:55 Mycodo gunicorn[516]:   File "<frozen importlib._bootstrap>", line 580, in module_from_spec
Apr 09 00:34:55 Mycodo gunicorn[516]: AttributeError: 'NoneType' object has no attribute 'loader'
Apr 09 00:34:55 Mycodo gunicorn[516]: [2021-04-09 00:34:55 -0700] [534] [INFO] Worker exiting (pid: 534)
Apr 09 00:34:55 Mycodo gunicorn[516]: [2021-04-09 00:34:55 -0700] [516] [INFO] Shutting down: Master
Apr 09 00:34:55 Mycodo gunicorn[516]: [2021-04-09 00:34:55 -0700] [516] [INFO] Reason: Worker failed to boot.
Apr 09 00:34:56 Mycodo systemd[1]: mycodoflask.service: Main process exited, code=exited, status=3/NOTIMPLEMENTED
Apr 09 00:34:56 Mycodo systemd[1]: mycodoflask.service: Failed with result 'exit-code'.`

Any idea what happened?
Is there any other info you need me to send?
Thanks,
JCL

kizniche, 2021-04-10:
Looks like there's something wrong with your virtualenv/filesystem. It might be indicative of a corrupt SD card.

LucidEye, 2021-04-10:
I just re-flashed it from my backup image.
So will preforming a "sudo shutdown now" or "sudo reboot" from an SSH session cause database or file system corruption in Mycodo?
Is there a safer, cleaner shutdown procedure to do from a terminal if we don't have access to the web-interface to do a proper system shutdown?

kizniche, 2021-04-10:
>will preforming a "sudo shutdown now" or "sudo reboot" from an SSH session cause database or file system corruption in Mycodo?

Not that I'm aware of.

closed 2021-04-24 by kizniche