kizniche/Mycodo #978: Can not setup VPD Function.  Getting error 'measurement/unit not set' when it is set?

LucidEye, 2021-04-10:
I am having trouble setting up the VPD function.

In the VPD function setup I am linking the temp and humidity to one of my BME280 sensors.

In the inputs setup, I have set that sensor to convert it's pressure and VPD readings to kPa, and set the temp to convert into F.

Back in the VPD function setup, if I only set the temp and humidity parameters and click on the save button at the top of the function window, it saves and the function works ok when I activate it...

[image]

However, I want to have the reading shown in kPa instead of the default Pa....

[image]

When I try to set the "name" under "Configure Measurements" to "VPD" and set it to "convert to kPa" then click save, the "Measurements Enabled" field gets 'cleared' for some reason and the function will no longer activate without throwing the error:

"Error: Activate Controller: Measurement CH0 (VPD) measurement/unit not set. All Measurements need to have the measurement and unit set before the Function can be activated."

[image]

There is no way to make the function work again without deleting the whole function it and re-adding it again.

Is this a bug, or am I doing something wrong?

Thank you,
JCL

kizniche, 2021-04-10:
That error shouldn't be occurring. I'll look into it. Thanks for the bug report.

LucidEye, 2021-04-10:
I also just noticed in the "Live Data" screen that the air pressure readings from both my sensors seem to be reading way too high.  I checked a local open weather station in our neighborhood and the current pressure in kPa is around 100.4.  So something is really wrong here.

[image]

LucidEye, 2021-04-10:
I just tried deleting both BME280 sensors from the Inputs page and reinstalling them using the RPI library instead of the Adafruit library and I am still getting bad readings...

[image]

Any ideas what is going on here?
I seem to remember I was also getting bad readings on the air pressure when using the BMP280 sensors I accidentally ordered from Adafruit.

kizniche, 2021-04-11:
Just checking that you're aware that the BME Input calculates VPD as one of its measurements, making use of the VPD Function redundant.

kizniche, 2021-04-11:
Fixed the issue with saving the configuration and the unit conversion. It was multiplying by 1000 instead of dividing.

LucidEye, 2021-04-11:
So how do I get these changes upgraded on my system?  I tried running an upgrade from the web-interface and it says there is no update available.  I tried running 'sudo mycodo-commands upgrade-mycodo' and got this error:

'Unable to upgrade. You currently have the latest release installed.
#### ERROR ####
There was an error detected during the upgrade. Please review the log at /var/log/mycodo/mycodoupgrade.log'

There is no log file at '/var/log/mycodo/mycodoupgrade.log'

[image]

kizniche, 2021-04-11:
If you want to test pre-release code, you'll need to [upgrade to master](https://github.com/kizniche/Mycodo/wiki/Updating-to-Github-Master-Branch). Keep in mind this may be unstable. I'm currently working on some pretty big changes that I've yet to commit, so I may have some fixes that are still staged in my IDE.

closed 2021-04-24 by kizniche