kizniche/Mycodo #979: 500 internal server error start after adding camera feed and feed server stopping

os-is-awesome, 2021-04-12:
Not sure if this is related to this issue,

### Describe the problem/bug

	My problem started after using a camera feed for a while.  it was one of the URL camera inputs. I believe it was the first URL feed in the mycodo list.  I dont have a working mycodo to find out.

	I used the feed for a while without any issues.  I stopped the Camera server to make some changes and restart the camera server and since then I could not go to the camera page.  I would get "500 internal server error" I didnt copy the error message from the camera page 500 internal.  

	I then decide to upgrade to the master @ apr 12 00:00 to see if this could help my problem.  And then I was unable to login to mycodo.  I could click the login button and then it would come up with a "500 internal server error"  attached is a copy of the browser error.

	all my fuctions and ouputs are working fine still

	
### Reproducibility

I dont know what to add to this beyond what I have already explained above.

### Versions:

MYCODO_VERSION = '8.9.2'
ALEMBIC_VERSION = '50a4e41e7444'
 - Raspberry Pi Version: [e.g. 4b]
 - Raspbian OS Version: [Buster]

I dont have any other info for these :( 
### Reproducibility
### Expected behavior
### Screenshots
### Additional context

 - Mycodo Version: [e.g. 8.0.3]
 - Raspberry Pi Version: [e.g. 3B+]
 - Raspbian OS Version: [e.g. Buster Lite]

### Reproducibility

Please list specific setup details that are involved and the steps to reproduce the behavior:

1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

### Expected behavior

A clear and concise description of what you expected to happen.

### Screenshots

If applicable, add screenshots to help explain your problem.

### Additional context

Is there anything that should be added to make it easier to address this issue?

kizniche, 2021-04-12:
Are you running the latest release code, earlier, or master branch?

os-is-awesome, 2021-04-12:
master

os-is-awesome, 2021-04-12:
======================================================== warnings summary =========================================================
../env/lib/python3.7/site-packages/marshmallow_sqlalchemy/schema/model_schema.py:83: 14 warnings
  /home/pi/Mycodo/env/lib/python3.7/site-packages/marshmallow_sqlalchemy/schema/model_schema.py:83: DeprecationWarning: marshmallow_sqlalchemy.ModelSchema is deprecated. Subclass marshmallow_sqlalchemy.SQLAlchemyAutoSchema and set `load_instance = True` and `include_relationships = True` on `class Meta` instead.
    DeprecationWarning,

../env/lib/python3.7/site-packages/marshmallow/fields.py:198: 43 warnings
  /home/pi/Mycodo/env/lib/python3.7/site-packages/marshmallow/fields.py:198: RemovedInMarshmallow4Warning: Passing field metadata as a keyword arg is deprecated. Use the explicit `metadata=...` argument instead.
    RemovedInMarshmallow4Warning,

../../../../usr/lib/python3/dist-packages/pandas/core/dtypes/inference.py:6
  /usr/lib/python3/dist-packages/pandas/core/dtypes/inference.py:6: DeprecationWarning: Using or importing the ABCs from 'collections' instead of from 'collections.abc' is deprecated, and in 3.8 it will stop working
    from collections import Iterable

../../../../usr/lib/python3/dist-packages/pandas/core/tools/datetimes.py:3
  /usr/lib/python3/dist-packages/pandas/core/tools/datetimes.py:3: DeprecationWarning: Using or importing the ABCs from 'collections' instead of from 'collections.abc' is deprecated, and in 3.8 it will stop working
    from collections import MutableMapping

../../../../usr/lib/python3/dist-packages/html5lib/_trie/_base.py:3
  /usr/lib/python3/dist-packages/html5lib/_trie/_base.py:3: DeprecationWarning: Using or importing the ABCs from 'collections' instead of from 'collections.abc' is deprecated, and in 3.8 it will stop working
    from collections import Mapping

tests/software_tests/conftest.py:34
  /home/pi/Mycodo/mycodo/tests/software_tests/conftest.py:34: PytestDeprecationWarning: @pytest.yield_fixture is deprecated.
  Use @pytest.fixture instead; they are the same.
    @pytest.yield_fixture()

tests/software_tests/conftest.py:54
  /home/pi/Mycodo/mycodo/tests/software_tests/conftest.py:54: PytestDeprecationWarning: @pytest.yield_fixture is deprecated.
  Use @pytest.fixture instead; they are the same.
    @pytest.yield_fixture()

-- Docs: https://docs.pytest.org/en/stable/warnings.html
=========================================== 24 passed, 62 warnings in 346.21s (0:05:46) ===========================================
pi@node1:~/Mycodo/mycodo/scripts $

kizniche, 2021-04-12:
Okay. There shouldn't be an expectation of stability on the master branch. I'd recommend restoring a backup if you need to access the live page.

os-is-awesome, 2021-04-12:
ok thanks

kizniche, 2021-04-13:
I just pushed a fix for the live page.

kizniche, 2021-04-13:
I actually hit this issue a few hours ago when testing something else. It only affected the Live page. Thanks for the big report.

closed 2021-04-24 by kizniche