kizniche/Mycodo #980: TP-Link KP303 Smart Plug

piedshag, 2021-04-14:
I have made a custom output for the kp303 smart plug which is working well so far.

https://gist.github.com/piedshag/7b0f5814501190c2ebe90c2e348ce7e5

kizniche, 2021-04-14:
Thanks for the contribution. I actually just bought this power strip. I'll make some modifications to allow all 3 outlets to be controlled from a single Output module and I'll include it for the next release.

kizniche, 2021-04-22:
I made several improvements to the input I added, namely a watch function that periodically queries the states of the outlets, which will also reconnect to the strip if it disconnects. This is important because the original version would not reconnect after a disconnect and you would not be able to switch any outlets until a daemon restart or output class reload.

closed 2021-04-24 by kizniche