kizniche/Mycodo #983: Support Linux board

SecT0uch, 2021-04-19:
Would you consider extending the list of compatible board ?

I'm a [MudPi](https://github.com/mudpi/mudpi-core) user and contributor. And I've helped to [extend support for other Linux SBC](https://github.com/mudpi/mudpi-core/issues/15), you can see the PR [here](https://github.com/mudpi/mudpi-core/pull/16).

The main task has been to get rid of Pi-only modules (RPi.GPIO) and migrate to [Adafruit Blinka](https://github.com/adafruit/Adafruit_Blinka).

It now runs  on Raspberry Pi, Pine64 and technically all the blinka-compatible boards.

For example I'm running it on a Sopine baseboard on armbian. I got a couple of sensors and a relay module, and it works flawlessly.

kizniche, 2021-04-23:
I'm not entirely comfortable with expanding support for different hardware. One of the best parts of about developing software on one hardware platform is compatibility. Pis are also so inexpensive and abundant. Do you have any specific desire to run Mycodo on another SBC that provides something a Pi cannot?

kizniche, 2021-04-23:
This issue has been mentioned on **DIY Automation Forum**. There might be relevant details there:

https://forum.kylegabriel.com/t/mycodo-on-other-non-pi-single-board-computers/119/1

SecT0uch, 2021-04-29:
> I'm not entirely comfortable with expanding support for different hardware. One of the best parts of about developing software on one hardware platform is compatibility. Pis are also so inexpensive [...]

I think being able to run Mycodo on 70+ boards models would be a big plus, to have a larger community, and potentially, more contributors. 
Each board has it's own specificities and it would add some flexibility. (ie: some boards have better performances, you could run on eMMC, etc.)