kizniche/Mycodo #984: Error 500: Internal Server Error

Rockytfox, 2021-04-20:
Error 500: Internal Server Error

Something bad happened but it's probably not your fault. Letting the developers know about these issues is crucial to supporting Mycodo. Please submit a new issue on GitHub with the following error traceback (copy the entire traceback):

Error (Full Traceback):

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1953, in full_dispatch_request
    return self.finalize_request(rv)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1970, in finalize_request
    response = self.process_response(response)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 2269, in process_response
    self.session_interface.save_session(self, ctx.session, response)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_session/sessions.py", line 354, in save_session
    total_seconds(app.permanent_session_lifetime))
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/cachelib/file.py", line 147, in set
    self._prune()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/cachelib/file.py", line 96, in _prune
    expires = pickle.load(f)
EOFError: Ran out of input

kizniche, 2021-04-20:
Duplicate of #971

closed 2021-04-20 by kizniche

Rockytfox, 2021-04-20:
> Duplicate of #971

Thanks Kyle.  I added some additional details simultaneously to you closing this issue.  Not sure if those details will help you or not.

kizniche, 2021-04-20:
I think I'm just going to remove flask-session in the next release.