kizniche/Mycodo #988: Solid gauge creation

itscornerstone, 2021-04-24:
Error (Full Traceback):

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_restx/api.py", line 639, in error_router
    return original_handler(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_login/utils.py", line 272, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_page.py", line 612, in page_dashboard
    unmet_dependencies = utils_dashboard.widget_add(form_base, request.form)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/utils/utils_dashboard.py", line 203, in widget_add
    new_widget, dict_widgets[widget_name])
  File "/home/pi/Mycodo/mycodo/widgets/widget_gauge_solid.py", line 49, in execute_at_creation
    color = color_list[i + 1]
IndexError: list index out of range

kizniche, 2021-04-24:
For future reference, please adhere to the bug report template for what information to include in a big report, as providing insufficient information delays or makes it impossible to diagnose or fix an issue.

https://github.com/kizniche/Mycodo/issues/new?assignees=&labels=&template=problem-bug-report.md&title=

What version Mycodo?

itscornerstone, 2021-04-25:
> For future reference, please adhere to the bug report template for what information to include in a bug report, as providing insufficient information delays or makes it impossible to diagnose or fix an issue.

Ok I will.
 Mycodo version is up to date as the latest version unless you posted a new version since late Monday. 
Using raspberry pi ver. 4
OS ver. buster lite
Thanks

itscornerstone, 2021-04-25:
Also let me know if you want picture of what I am doing.  I plan on putting 9 of these in place within the next year. Works pretty good, you should be proud.

kizniche, 2021-04-25:
Latest version is 8.10.0 (2021-04-24)

itscornerstone, 2021-04-25:
Ok just updated.

kizniche, 2021-04-29:
And does your issue persist in the latest version?

itscornerstone, 2021-04-29:
Nope all good. I plan on building 2 more like this cause it works pretty well.