kizniche/Mycodo #990: All outputs are listed as "Unconfigured" since 8.10

dylandn, 2021-04-27:
Under the Outputs screen, all of my pre-existing outputs no longer display their state and instead show "Unconfigured". Adding a new Output and configuring it still shows as Unconfigured. This seems to have started since 8.10.
[image]

The outputs still work, everything's functional except for the "Unconfigured" message.

kizniche, 2021-04-27:
What is the database version listed on the /info page?

dylandn, 2021-04-27:
Database Version: 4fedeb57e75b

kizniche, 2021-04-29:
Are there any errors in the Daemon or Web logs or when you visit https://YOUR_IP/outputstate ?

dylandn, 2021-04-30:
Oh, yes. This shows up at every loading of /output and /outputstate:
`2021-04-29 19:16:25,297 - ERROR - mycodo.daemon - Could not query output state: 'GPIO is not in use for PWM'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 854, in output_states_all
    return self.controller['Output'].output_states_all()
  File "/var/mycodo-root/mycodo/controllers/controller_output.py", line 327, in output_states_all
    states[output_id][each_channel] = self.output[output_id].output_state(each_channel)
  File "/var/mycodo-root/mycodo/outputs/base_output.py", line 673, in output_state
    state = self.is_on(output_channel)
  File "/home/pi/Mycodo/mycodo/outputs/pwm_gpio.py", line 274, in is_on
    response = self.pwm_output.get_PWM_dutycycle(self.options_channels['pin'][0])
  File "/usr/lib/python3/dist-packages/pigpio.py", line 1479, in get_PWM_dutycycle
    """
  File "/usr/lib/python3/dist-packages/pigpio.py", line 979, in _u2i
    def _str(x):
pigpio.error: 'GPIO is not in use for PWM'`

/outputstate returns a blank page with "null"

closed 2021-06-05 by kizniche

dylandn, 2021-06-06:
indeed, thank you!