kizniche/Mycodo #991: Live stream in Dashboard

fluke9, 2021-04-29:
Hi there,
is streaming of USB webcams now supported ?
If not it looks like #630 sneaked back in again.

I only get a unloadable image and the following traceback if i view it directly:

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_restx/api.py", line 652, in error_router
    return original_handler(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_login/utils.py", line 272, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_general.py", line 209, in video_feed
    camera_stream = import_module('mycodo.mycodo_flask.camera.camera_' + camera_options.library).Camera
  File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 965, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'mycodo.mycodo_flask.camera.camera_fswebcam'

kizniche, 2021-04-29:
Streaming has never been supported by any library other than picamera, as stated [here](https://github.com/kizniche/Mycodo/issues/630#issuecomment-464370437) in #630.

> Live streaming is currently only supported with the picamera library.

https://github.com/kizniche/Mycodo/commit/29bdb3fbf96f91420ed1757bd7bd21748f3587ee from #630 was supposed to fix this by disabling the ability to select a camera using the fswebcam library in the camera widget, but this seems to have been reverted during the refactoring of how widgets work to allow custom widget imports. I'll modify the widget module to prevent selecting non-streamable libraries.

fluke9, 2021-04-29:
Wow, that was quick!
Thanks a lot, got it working with opencv.

closed 2021-04-29 by fluke9