kizniche/Mycodo #992: CCS811 fix for newer versions of the chip

fluke9, 2021-04-29:
Hi there,

just a heads up if you ever encounter problems with newer CCS811 sensors:
There is a newer version of the chip out there which does not support the NTC anymore (see orange boxes at https://learn.adafruit.com/adafruit-ccs811-air-quality-sensor).
Worse than that, querying it put it in an undefined state which only a powercycle can get it out of.

To get the newer sensor running i needed change the following in mycodo/inputs/ccs811.py:
- remove the call to self.sensor.calculateTemperature() in def initialize_input(self):
- remove the same call to def get_measurement(self):
- remove the self.value_set(2... in get_measurement(self):

Maybe this should be implemented someway more nice, like either a different sensor type or a toggle for that version.

Besides from that i2c bus speed needs to be lowered to 10k in /boot/config.txt, because these sensors have some clock streching going on the bcm does not support.

kizniche, 2021-04-30:
Thanks for the tip. I added a new Input using the CircuitPython library that only measures CO2 and VOC.

fluke9, 2021-05-13:
Just had time to try it out, works fine! Thanks.

closed 2021-05-13 by fluke9