kizniche/Mycodo #996: Dashboard optimization

chamroeunbm, 2021-05-08:
I see the dashboard is getting very heavy due to loading of all the output options, inputs,....

I think we should have two modes, ie display and edit, where in display mode only the display items are seen and loaded.

Do you have any other suggestion?

kizniche, 2021-05-14:
I've never experienced what you describe, even on devices with low resources such as my phone. Can you provide a screenshot of the kind of dashboard you're describing and list the specs of the computer rendering it in your web browser?

chamroeunbm, 2021-05-14:
Your code list all the output/input so I if have 50 input and there're 50 widget, the HTML page will list 50 times of that 50 inputs which increase the HTML size a lot.

kizniche, 2021-05-14:
It's actually not a lot of HTML. Many JS source files are 10x and even 100x larger than the scenario you describe. You also likely don't have 50 inputs nor 50 widgets on your desktop. I'm asking for you to demonstrate the issue you say you have.

chamroeunbm, 2021-05-18:
Currently, I have 16 inputs mostly wireless(Mijia, FLowercare). Flowercare alone has 5 channels--thus it will list as 5 inputs. We're having 50+ soon as in a greenhouse, we'll have up to 6 soil moisture + other sensors. There's no problem to open it from my laptop locally. But I try to remote it but that come the problem of dashboard exceed 2.5MB(html, javascript...). I am trying to find a way to optimize it.

kizniche, 2021-05-18:
I could add a dashboard option "lock dashboard" which would remove the widget option HTML from the page.

kizniche, 2021-05-26:
I just added the ability to lock dashboards. If you [upgrade to master](https://github.com/kizniche/Mycodo/wiki/Updating-to-Github-Master-Branch), you can try it out before release. If you do, let me know what kind of performance improvement you see and perhaps compare the number of source lines when locked and unlocked.

kizniche, 2021-05-26:
Just pushed another change that further reduces the size by removing the Add Widget dropdown at the bottom when locked.

garudaonekh, 2021-05-27:
When locked, most of the input widgets are not displaying any image or data, the graph are not displaying any data maybe javascript issue. The output widget is ok.

kizniche, 2021-05-27:
That's not happening for me. You're going to have to provide a bit more information: JS errors, browser inspection, etc.

garudaonekh, 2021-05-27:
[image]
[image]

There's 1 javascript error, I think this error may be the problem: 

`Uncaught ReferenceError: decimal_places is not defined
    getLastDataGaugeAngular 2a457a9e-3cde-486c-82f9-e173300618aa:931
    `

kizniche, 2021-05-28:
Thanks. I just committed a fix.

garudaonekh, 2021-05-28:
The error still happen on existing widget(I tried both widgets that you updated).

The following error occurred when add widget in dashboard.

`Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_restx/api.py", line 652, in error_router
    return original_handler(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_login/utils.py", line 272, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_page.py", line 626, in page_dashboard
    unmet_dependencies = utils_dashboard.widget_add(form_base, request.form)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/utils/utils_dashboard.py", line 239, in widget_add
    control.widget_add_refresh(new_widget.unique_id)
  File "/home/pi/Mycodo/mycodo/mycodo_client.py", line 296, in widget_add_refresh
    return self.proxy().widget_add_refresh(unique_id)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/Pyro5/client.py", line 476, in __call__
    return self.__send(self.__name, args, kwargs)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/Pyro5/client.py", line 243, in _pyroInvoke
    raise data  # if you see this in your traceback, you should probably inspect the remote traceback as well
AttributeError: 'NoneType' object has no attribute 'widget_add_refresh'`

garudaonekh, 2021-05-28:
the unlock html file of my dashboard is 1 MB, now it's  270K, so it's nearly 75% decrease in size. That's fantastic

kizniche, 2021-05-28:
Did you upgrade to master?

garudaonekh, 2021-05-28:
After I am back from lunch and restart the service(both Mycodo and flask), it works ok now.

closed 2021-05-28 by garudaonekh