kizniche/Mycodo #1000: Trigger Output is not working with Duration timer, output does not turn on after duration time.

LucidEye, 2021-05-10:
Mycodo 8.10.1
RaspberryOS Buster 10
Kernel version 5.10.17-v7+

I am trying to use a Trigger: Output (On/Off) function and it is not working if I try to use a Duration time.  
The Action Output only turns on if I set the If State to "on (no duration)".  
I need to set a duration, but none of the duration settings in the If State work... the output never turns on after the duration time passes.  
In the Daemon log, it never shows the output being turned on.  
However when I "test all actions" in the function, the Action Output works.

Steps I used to create this problem:

1. In the Outputs page I created the [GPIO] GPIO: On/Off (RPi.GPIO) for my humidifier and configured it like this...

[image]

2. In the Outputs page I created the [GPIO] GPIO: PWM (pigpio) to control my fan speed and configured it like this...

[image]

3. In the Functions page I created the Trigger: Output (On/Off) function.  
Triggering Output is the humidifier output created in step 1.
If State is set to "On (greater than duration)".
Duration is set to 30 seconds.
Log Level: Debug is "ON"
Action Output is the PWM Fan Speed control created in step 2.
Duty Cycle of the output is set to 60%
Here are the Function settings...

[image]

4.  After the Trigger: Output (On/Off) Function is created and saved I activate the function.

5.  I then click the "Test all actions" button and visually verify that the fans do increase from 25% to 60% duty cycle.  The PWM indicator on my dashboard also indicates the duty cycle increase.

6.  I manually reset fans back to 25% PWM duty cycle using the [GPIO] GPIO: PWM (pigpio) on the Outputs page with the "Duty Cycle - Send" field and visually verify that fans have decreased in speed.

7.  I then try to use the Trigger: Output (On/Off) Function in practice by turning on my humidifier output from my Dashboard and watching to see if the fan PWM duty cycle increases to 60% after the 30 second Duration period has passed, which it does not.  The Action output (PWM change) does not activate, and the debug log shows no entry for that action.

8. In the Functions page, I change the "If State" setting to "On (no duration)" and save.

9. I then retry steps 5, 6, 7 and the Function works as expected... humidifier turns on and the fans increase in speed... however it does not work with the 30 second duration time... it turns the fans up instantly and the Daemon log shows the Action output as being activated, as does the dashboard indicator.

10.  Repeating steps 5 ,6 ,7 with the following "If State" settings did NOT function, the fan speed never increased after the Duration period, and the Daemon log never showed any action entries or errors:
On (any duration)
On (equal to duration)
On (greater than duration)
On (greater than or equal to duration)

11.  Repeating steps 5, 6, 7 with the following "if State" settings DO function, the fan speed increased, and the Daemon log showed the action output as being activated however there was no duration period, the fan speed increased immediately after turning on the humidifier:
On (no duration)
On (no or any duration.

If there is any other information or settings you need me to provide please let me know.  
Thank you.
JCL

kizniche, 2021-05-11:
From 7, with If State set to "On (greater than duration)" and "Duration" set to 30 seconds, the duration you turn the Output on needs to be greater than 30 for the Trigger to activate. Setting an output of 30 seconds will not activate the Trigger because it is not greater than 30 seconds.

From 8, "On (no duration)" will cause the Trigger to execute when an On command is set to the Output so long as no duration is specified.

> the fan speed never increased after the Duration period

This is not how Triggers work. Triggers execute immediately if the conditions are True.

kizniche, 2021-05-11:
From 10, I just tested all ```If States``` and all behaved as expected and worked.

kizniche, 2021-05-11:
Thank you for resubmitting this. It is mush easier to follow and understand.

LucidEye, 2021-05-11:
> From 7, with `If State` set to "On (greater than duration)" and `Duration` set to 30 seconds, the duration you turn the Output on needs to be greater than 30 for the Trigger to activate. Setting an output on for a duration of 30 seconds will not activate the Trigger because it is not greater than 30 seconds.
>

But I am turning the humidifier output on for more than 30 seconds by manually activating it from the outputs page.  I turn it on, and leave it on, so the Trigger should see that it's been on for "greater than 30 seconds", then turn the Action output on, correct?
At least that's how I understand the instructions in the manual...

[image]

> 
> From 8, "On (no duration)" will cause the Trigger to execute when an On command is set to the Output so long as no duration is specified.
> 
> > the fan speed never increased after the Duration period
> 
> This is not how Triggers work. Triggers execute immediately if the conditions are True, at the time the output is given its instruction.

Yes, I understood this is the expected and proper operation when "On (no duration)" is selected.  I was merely observing that the 
 function does activate the action output... just not when an "On (duration)" setting is used for some reason.

kizniche, 2021-05-11:
Go to the output page. For any On/Off output, you can press the On button or you can send it a duration to turn on for. These are the only two options you have to instruct an output to turn on. This is all Mycodo can work with for turning an output on.

kizniche, 2021-05-11:
> regardless if it was

There is no "regardless". It is a very specific act to turn an output on or to turn it on for a duration.

LucidEye, 2021-05-11:
Nowhere in the Manual does it mention that the Output needs to be set to a predetermined time in order to use the Duration time setting in the Functions setup.

LucidEye, 2021-05-11:
So are you saying the manual is wrong, and it does not accurately describe how the Duration setting in the Function works?  
Are you saying that I can't do what I am trying to do?  Have a bang bang turn on my humidifier for as long as it needs to be on, and then have my fans speed up 30 seconds after the humidifier turn on?

LucidEye, 2021-05-11:
To me, reading the manual, it says that setting a duration in the function means that if the trigger output is on for "greater than X time" then the Action trigger turns on after that Duration time is expired regardless of how the trigger output was activated.

kizniche, 2021-05-11:
No matter how you interpret the manual, it's not going to change how it actually operates. I already gave you the only option you have to make a difference:

> Then contribute to the manual.

LucidEye, 2021-05-11:
So the answer is no?

LucidEye, 2021-05-11:
Can you please help me figure out how to do what I need to do?

kizniche, 2021-05-11:
Bang-Bang: Humidifier

Trigger A: If Humidifier On (no duration)
Trigger A Actions:
    Pause Actions: 30 seconds
    Output (On/Off/Duration): Fan On

Trigger B: If Humidifier Off
Trigger B Actions:
    Output (On/Off/Duration): Fan Off

closed 2021-05-11 by kizniche

LucidEye, 2021-05-11:
Thank you!!! That is the last little detail I needed to get everything working like I need to.  
It was not intuitive to me to use the "Pause Actions" action.  
It would seem much more intuitive to simply add a duration timer option to each action output rather than adding the timer separately as a separate action in between every other action in the list... it would make the actions list shorter, and look a lot cleaner. Also if you need to add a timer (Pause Action), you wouldn't have to delete and then re-add all the actions to get the Pause Action in the right place in the actions list.  Just a suggestion.
Thank you again for your help, and I apologize again for my earlier misunderstanding.