kizniche/Mycodo #997: BT sensor Xiaomi not getting measurement data

MangoMars23, 2021-05-10:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

kizniche, 2021-05-10:
[This comment](https://github.com/kizniche/Mycodo/issues/965#issuecomment-835150215) identifies your likely issue. If you have ever connected the device to your prone, you cannot use the Non-ATC mode.

MangoMars23, 2021-05-10:
thanks! I have tried ATC and non-ATC mode too. I have even tried the new device( the one I did not use in the App yet). I flashed custom firmware and found MAC address in Telink Flasher...but still no data from measurement.

kizniche, 2021-05-10:
Can you see Bluetooth devices in your vacinity from your Pi's command line?

MangoMars23, 2021-05-10:
Now I see the measurements just appeared for while(30sec) and after went back to zeros. What can cause measurements to appear and disappear suddenly?

kizniche, 2021-05-10:
Measurements do not disappear. They're stored in a time-series database. Create a graph widget on the dashboard to see all the measurement for a specified period.

MangoMars23, 2021-05-10:
to see it?Should I use hcitool scan to see it?-------- Original message --------From: Kyle Gabriel ***@***.***>Date: Sun., May 9, 2021, 9:13 p.m.To: kizniche/Mycodo ***@***.***>Cc: MangoMars23 ***@***.***>, Author ***@***.***>Subject: Re: [kizniche/Mycodo] BT sensor Xiaomi not getting measurement data (#997)
Can you see Bluetooth devices in your vacinity from your Pi's command line?

MangoMars23, 2021-05-18:
Thank You! It looks like working great now!!!

LUSTgabriel, 2023-03-12:
i have same promblem,i recived the ble debug pack .but just cant get any information. the log print this:
.2023-03-13 03:10:54,984 - DEBUG - mycodo.inputs.xiaomi_mijia_lywsd03mmc_4119798e - Received BLE packet
2023-03-13 03:10:55,085 - DEBUG - mycodo.inputs.xiaomi_mijia_lywsd03mmc_4119798e - Received BLE packet
2023-03-13 03:10:55,188 - DEBUG - mycodo.inputs.xiaomi_mijia_lywsd03mmc_4119798e - Received BLE packet
And no any data on Asynchronous graphs\Dashboard\Live neasurement.that's strange because its was work at the first time. All version is up to date. plz help me.