kizniche/Mycodo #998: PID -

Mark0Mi, 2021-05-10:
PID safety values
Hi Kyle not sure what I'm doing wrong here 
 I read through the document and I think I did it correctly but there is something I miss: 
I created a new PID controller
It measures mV between 300-800 the setpoint is 700 (+/- 50) 
I left Kp Gain at 1 just to get starting (finetuning in the end) 
but I limited the output to Max On Amount (raise) on 10 - which seems to be ignored. 
as soon as I switched it on (after 300 seconds [period / start offset]) it did the calculation 
current value was 300 mV
new value 700 ->thus  error is 400 and pumped 400 ml in. (ignoring the max on amount of 10) 
can you explain this to me please? why was the max value ignored? 
my expectations would be that is stops after 10ml (as the maximum defined value) and adds another 10 in the next period. 
[image]

one question (not to mix with the problem above) the autotune PID controller supports only raise - there is no lower option - is this by design? 

### Versions:

 - Mycodo Version: [e.g. 8.10.1]
 - Raspberry Pi Version: [e.g. 4]

kizniche, 2021-05-10:
It should only dispense a max of 10 ml per PID period. I'll check out if this was improperly implemented when volume was added as a PID option. Thanks for the bug report.

Mark0Mi, 2021-05-10:
Can you do me a favor and check another thing as well 
I have second environment thing to control with PID
the value reading is 7.482 ph 
the setpoint is 7.1. (+/- 0.05) 

it should start dosing in - but the pump does not activate 
I removed the "store lower as negative" but it does not change - can it be that the output with the "negative" - is the problem for the volume? 

- 2021-05-10 17:08:10,232 - DEBUG - mycodo.controllers.controller_pid_9b7848f1 - Latest (CH0, Unit: pH): 7.476 @ 2021-05-10 17:08:08
- 2021-05-10 17:08:10,232 - DEBUG - mycodo.controllers.controller_pid_9b7848f1 - PID: Input: 7.476, Output: P: -0.42600000000000016, I: -0.0, D: 0.0, Out: -0.42600000000000016
- 2021-05-10 17:08:10,348 - DEBUG - mycodo.controllers.controller_pid_9b7848f1 - Setpoint: 7.1 Output: -0.42600000000000016 ml to output c467993e-61ef-437a-bc15-5716c5582ee0 CH0
- 2021-05-10 17:08:40,197 - DEBUG - mycodo.controllers.controller_pid_9b7848f1 - Latest (CH0, Unit: pH): 7.478 @ 2021-05-10 17:08:38
- 2021-05-10 17:08:40,197 - DEBUG - mycodo.controllers.controller_pid_9b7848f1 - PID: Input: 7.478, Output: P: -0.42799999999999994, I: -0.0, D: -0.0, Out: -0.42799999999999994
- 2021-05-10 17:08:40,328 - DEBUG - mycodo.controllers.controller_pid_9b7848f1 - Setpoint: 7.1 Output: -0.42799999999999994 ml to output c467993e-61ef-437a-bc15-5716c5582ee0 CH0
- 2021-05-10 17:09:10,188 - DEBUG - mycodo.controllers.controller_pid_9b7848f1 - Latest (CH0, Unit: pH): 7.487 @ 2021-05-10 17:09:08
- 

[image]

kizniche, 2021-05-11:
I just pushed a fix for the PID to respect the Volume Max setting. If you can, please [upgrade to master](https://github.com/kizniche/Mycodo/wiki/Updating-to-Github-Master-Branch) to get the code and test if it works.

Mark0Mi, 2021-05-11:
Hi 
the "Send Lower as Negative" is not checked but the value is still negative 

- 2021-05-11 07:28:12,663 - DEBUG - mycodo.controllers.controller_pid_9b7848f1 - PID: Input: 7.522, Output: P: -0.4720000000000004, I: -0.0, D: -0.0, Out: -0.4720000000000004
- 2021-05-11 07:28:12,770 - DEBUG - mycodo.controllers.controller_pid_9b7848f1 - Setpoint: 7.1 Output: -0.4720000000000004 ml to output c467993e-61ef-437a-bc15-5716c5582ee0 CH0
- 2021-05-11 07:28:42,655 - DEBUG - mycodo.controllers.controller_pid_9b7848f1 - Latest (CH0, Unit: pH): 7.523 @ 2021-05-11 07:28:40
- 2021-05-11 07:28:42,655 - DEBUG - mycodo.controllers.controller_pid_9b7848f1 - PID: Input: 7.523, Output: P: -0.47299999999999986, I: -0.0, D: -0.0, Out: -0.47299999999999986
- 2021-05-11 07:28:42,798 - DEBUG - mycodo.controllers.controller_pid_9b7848f1 - Setpoint: 7.1 Output: -0.47299999999999986 ml to output c467993e-61ef-437a-bc15-5716c5582ee0 CH0
- 2021-05-11 07:29:12,674 - DEBUG - mycodo.controllers.controller_pid_9b7848f1 - Latest (CH0, Unit: pH): 7.536 @ 2021-05-11 07:29:10
- 2021-05-11 07:29:12,674 - DEBUG - mycodo.controllers.controller_pid_9b7848f1 - PID: Input: 7.536, Output: P: -0.48599999999999977, I: -0.0, D: -0.0, Out: -0.48599999999999977
- 2021-05-11 07:29:12,810 - DEBUG - mycodo.controllers.controller_pid_9b7848f1 - Setpoint: 7.1 Output: -0.48599999999999977 ml to output c467993e-61ef-437a-bc15-5716c5582ee0 CH0
[image]

closed 2021-06-05 by kizniche