kizniche/Mycodo #1001: Error 500 trying a DHT11 sensor

JJare, 2021-05-12:
I got this internal server error when I try to add an DHT11 sensor, after installing required gpio dependencies:

Error 500 - Error Interno del Servidor

Something bad happened but it's probably not your fault. Letting the developers know about these issues is crucial to supporting Mycodo. Please submit a new issue on GitHub with the following error traceback (copy the entire traceback):

Error (Full Traceback):

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_restx/api.py", line 652, in error_router
    return original_handler(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_login/utils.py", line 272, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_page.py", line 2166, in page_input
    devices_1wire_w1thermsensor=devices_1wire_w1thermsensor)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/templating.py", line 140, in render_template
    ctx.app,
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/templating.py", line 120, in _render
    rv = template.render(context)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/jinja2/environment.py", line 1289, in render
    self.environment.handle_exception()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/jinja2/environment.py", line 924, in handle_exception
    raise rewrite_traceback_stack(source=source)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/input.html", line 3, in top-level template code
    {% set help_page = ["https://kizniche.github.io/Mycodo/Inputs/", dict_translation['input']['title']] %}
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/layout.html", line 375, in top-level template code
    {%- block body %}{% endblock -%}
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/input.html", line 35, in block 'body'
    {% include 'pages/data_options/input.html' %}
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/data_options/input.html", line 202, in top-level template code
    {% include 'pages/form_options/GPIO.html' %}
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/form_options/GPIO.html", line 8, in top-level template code
    {%- if 'period' in dict_options['gpio_location'] -%}
jinja2.exceptions.UndefinedError: 'dict object' has no attribute 'gpio_location'

Can somebody help me?

Thx

kizniche, 2021-05-13:
I committed a potential fix. GO ahead and [upgrade to master](https://github.com/kizniche/Mycodo/wiki/Updating-to-Github-Master-Branch) and see if it resolves the issue.

JJare86, 2021-05-14:
it works, thank you very much, but now I'm trying to add a graph with temperature and humidity data and getting another 500 internal server error :'( 

```
Error 500 - Error Interno del Servidor
Something bad happened but it's probably not your fault. Letting the developers know about these issues is crucial to supporting Mycodo. Please submit a new issue on GitHub with the following diagnostic information and error traceback (copy the entire traceback):

Mycodo: 8.10.1

Alembic: d46e17d65f48

Model: Raspberry Pi 3 Model B Plus Rev 1.3

Release:
Distributor ID: Raspbian
Description: Raspbian GNU/Linux 10 (buster)
Release: 10
Codename: buster

Firmware:
Feb 25 2021 12:12:09
Copyright (c) 2012 Broadcom
version 564e5f9b852b23a330b1764bcf0b2d022a20afd0 (clean) (release) (start)

Error (Full Traceback):

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_restx/api.py", line 652, in error_router
    return original_handler(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_login/utils.py", line 272, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_page.py", line 801, in page_dashboard
    form_dashboard=form_dashboard)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/templating.py", line 140, in render_template
    ctx.app,
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/templating.py", line 120, in _render
    rv = template.render(context)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/jinja2/environment.py", line 1289, in render
    self.environment.handle_exception()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/jinja2/environment.py", line 924, in handle_exception
    raise rewrite_traceback_stack(source=source)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/dashboard.html", line 3, in top-level template code
    {% set help_page = ["https://kizniche.github.io/Mycodo/Data-Viewing/#dashboard", dict_translation['dashboard']['title']] %}
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/layout.html", line 375, in top-level template code
    {%- block body %}{% endblock -%}
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/dashboard.html", line 372, in block 'body'
    {% include 'user_templates/{}'.format(file_js_ready_end) %}
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/user_templates/widget_template_widget_graph_synchronous_js_ready_end.html", line 150, in top-level template code
    {%- if 'custom_axis_id' in dict_measurements[each_axis_meas] -%}
jinja2.exceptions.UndefinedError: 'collections.OrderedDict object' has no attribute 'C'
```

kizniche, 2021-05-14:
```
    {%- if 'custom_axis_id' in dict_measurements[each_axis_meas] -%}
jinja2.exceptions.UndefinedError: 'collections.OrderedDict object' has no attribute 'C'
```

This is stating there is no measurement unit C, which indicates there's probably something wrong with your system. I'd reinstall Raspberry Pi OS and Mycodo and see if the issue persists.

Additionally, none of my systems experienced your original issue, even when I attempted to induce it. I did take that code out though because it was no longer necessary, but it seems your system has some unknown issue, perhaps file corruption, damage to your disk, or something else.

closed 2021-06-05 by kizniche