kizniche/Mycodo #1002: Deleted functions and input a widget was using on the dashboard

JuliaHorvath, 2021-05-13:
I created a dashboard with widgets using inputs and functions. But then I had to change something, so I deleted the functions and the inputs it was using, but I forgot about the already existing widgets. Now when I click on the dashboard it just shows me that error. When I click on any other dashboard, everything is fine. 
I already tried to redo the functions and inputs. I gave them the same names and everything again, but unfortunately that doesn't change anything.
Anybody knows how to fix it?
I wouldn't even mind to just delete that dashboard and redo it. I just don't know how I could delete because I don't even get to the dashboard page. 

Fehler 500: interner Server-Fehler
Something bad happened but it's probably not your fault. Letting the developers know about these issues is crucial to supporting Mycodo. Please submit a new issue on GitHub with the following error traceback (copy the entire traceback):

Error (Full Traceback):

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_restx/api.py", line 652, in error_router
    return original_handler(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_login/utils.py", line 272, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_page.py", line 801, in page_dashboard
    form_dashboard=form_dashboard)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/templating.py", line 140, in render_template
    ctx.app,
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/templating.py", line 120, in _render
    rv = template.render(context)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/jinja2/environment.py", line 1090, in render
    self.environment.handle_exception()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/jinja2/environment.py", line 832, in handle_exception
    reraise(*rewrite_traceback_stack(source=source))
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/jinja2/_compat.py", line 28, in reraise
    raise value.with_traceback(tb)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/dashboard.html", line 3, in top-level template code
    {% set help_page = ["https://kizniche.github.io/Mycodo/Data-Viewing/#dashboard", dict_translation['dashboard']['title']] %}
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/layout.html", line 375, in top-level template code
    {%- block body %}{% endblock -%}
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/dashboard.html", line 372, in block "body"
    {% include 'user_templates/{}'.format(file_js_ready_end) %}
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/user_templates/widget_template_widget_gauge_angular_js_ready_end.html", line 144, in top-level template code
    return this.series.name + ':<b> ' + Highcharts.numberFormat(this.y, 2) + ' {{dict_units[device_measurements_dict[measurement_id].unit]['unit']}}</b><br>';
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/jinja2/environment.py", line 471, in getattr
    return getattr(obj, attribute)
jinja2.exceptions.UndefinedError: 'dict object' has no attribute '035df983-6acd-4321-a1e9-4566ae66375a'

kizniche, 2021-05-14:
I could not reproduce the issue. If you can provide concise steps to reproduce it, I'll investigate.

kizniche, 2021-05-14:
Additionally, you can use the Delete All Dashboards button on the Configure -> Diagnostics page to delete all dashboards/widgets.

JuliaHorvath, 2021-05-14:
Alright, I just deleted them all! Seemed to be the only way out. Thank you!

closed 2021-05-14 by JuliaHorvath