kizniche/Mycodo #1003: HC-SR04 Not being Read

fivbag, 2021-05-15:
Mycodo Version: 8.10.1
Python Version: 3.7.3 (default, Jan 22 2021, 20:04:44) [GCC 8.3.0]
Database Version: 4fedeb57e75b
Daemon Status: Running
Daemon Process ID: 792
Daemon RAM Usage: 50.732 MB
Daemon Virtualenv: Yes
Frontend Process ID: 502
Frontend RAM Usage: 63.652 MB
Frontend Virtualenv: Yes
uptime

 10:58:27 up 2 min,  2 users,  load average: 0.66, 0.65, 0.27

Kernel Information: uname -a

Linux raspberrypi 5.10.17-v7l+ #1403 SMP Mon Feb 22 11:33:35 GMT 2021 armv7l GNU/Linux

So I am not familiar with programming, although for this (my) project I have started learning python. And I am also not familiar with electronics. However my issue;

I have a HC- SR04 connected on a bread Board. I have used code online to test if the device works ( I can confirm getting readings/distances)

I have added the device in MYCODO, but I get no readings.

Side Note I previously added Two (2) temperature sensors, and both of those work as we speak.

fivbag, 2021-05-16:
If there is some output i can provide to assist in getting to the root of the issue. Please advise

fivbag, 2021-05-16:
SOLVED..
SO I was able to figure out how to get the logs and reading through I realized that there was a dependency package missing. Now idea why it didn't install at first
"Libgpiod" was missing.
Merci
And Thank you for this beautiful software

kizniche, 2021-05-16:
Can you paste the log excerpt and the command you used to install the dependency, so I may potentially add it to the module?

fivbag, 2021-05-16:
Commands used
sudo apt-get update -y

sudo apt-get install -y libgpiod-dev

fivbag, 2021-05-16:
OK so another issue has come up. 
After I restart, the sensor takes one reading and then it remains at that reading, no matter if distance is changed. 
However, python script (outside of mycodo) shows that the device is still functioning properly with continual readings

fivbag, 2021-05-16:
[Debug.txt](https://github.com/kizniche/Mycodo/files/6488485/Debug.txt)
I included the log

kizniche, 2021-05-16:
> the sensor takes one reading and then it remains at that reading

I'm not following. Your log shows multiple measurements after activation.

fivbag, 2021-05-16:
When I switch to live or the dashboard the sensor outputs 0 Centimeters.
Even if I refresh or I wait the auto refresh period.
Not sure what broke..

kizniche, 2021-05-16:
The Live page will only show measurements within the past Period that's been set. There's [a known issue](https://github.com/adafruit/Adafruit_CircuitPython_HCSR04/issues/22) with acquiring measurements with the library. If the Live page is only looking in the past Period and a measurement wasn't acquired, it's not going to be able to display anything. So, if you want to see all measurements, you need to check either 1) Data -> Dashboard -> Synchronous Graph Widget, 2) Data -> Asynchronous Graph, or 3) More -> Export Import -> Export Data.

fivbag, 2021-05-16:
Noted. I will check this out.
Thanks

kizniche, 2021-05-16:
You can also try increasing the number of attempts from 3 and maybe add a pause after each attempt.

https://github.com/kizniche/Mycodo/blob/08db5e1eb0d483edd7951aca58ffb29e154fc69b/mycodo/inputs/hcsr04_circuitpython.py#L129-L134

fivbag, 2021-05-16:
Yes you are correct..
Things seem to be functioning now. Appreciate your help and your hard work
in providing this software to us.

closed 2021-06-05 by kizniche