kizniche/Mycodo #1005: Adding Synchronous Graph to empty Dashboard results in Error 500

dookaloosy, 2021-05-16:
### Describe the problem/bug

On a fresh install of Mycodo 8.10.1, upgraded to master commit #08db5e1, adding a Synchronous Graph to an empty Dashboard results in Error 500. Subsequent attempts to access the Dashboard are met with the same error:

```
Error 500: Internal Server Error
Something bad happened but it's probably not your fault. Letting the developers know about these issues is crucial to supporting Mycodo. Please submit a new issue on GitHub with the following diagnostic information and error traceback (copy the entire traceback):

Mycodo: 8.10.1

Alembic: d46e17d65f48

Model: Raspberry Pi 3 Model B Rev 1.2

Release:
Distributor ID: Raspbian
Description: Raspbian GNU/Linux 10 (buster)
Release: 10
Codename: buster

Firmware:
Feb 25 2021 12:12:09
Copyright (c) 2012 Broadcom
version 564e5f9b852b23a330b1764bcf0b2d022a20afd0 (clean) (release) (start)

Error (Full Traceback):

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_restx/api.py", line 652, in error_router
    return original_handler(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_login/utils.py", line 272, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_page.py", line 801, in page_dashboard
    form_dashboard=form_dashboard)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/templating.py", line 140, in render_template
    ctx.app,
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/templating.py", line 120, in _render
    rv = template.render(context)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/jinja2/environment.py", line 1289, in render
    self.environment.handle_exception()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/jinja2/environment.py", line 924, in handle_exception
    raise rewrite_traceback_stack(source=source)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/dashboard.html", line 3, in top-level template code
    {% set help_page = ["https://kizniche.github.io/Mycodo/Data-Viewing/#dashboard", dict_translation['dashboard']['title']] %}
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/layout.html", line 375, in top-level template code
    {%- block body %}{% endblock -%}
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/dashboard.html", line 372, in block 'body'
    {% include 'user_templates/{}'.format(file_js_ready_end) %}
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/user_templates/widget_template_widget_graph_synchronous_js_ready_end.html", line 150, in top-level template code
    {%- if 'custom_axis_id' in dict_measurements[each_axis_meas] -%}
jinja2.exceptions.UndefinedError: 'collections.OrderedDict object' has no attribute 'C'
```

### Versions:

 - Mycodo Version: 8.10.1
 - Raspberry Pi Version: 3B
 - Raspbian OS Version: Linux raspberrypi 5.10.17-v7+ #1403 SMP Mon Feb 22 11:29:51 GMT 2021 armv7l GNU/Linux

### Reproducibility

1. Fresh install of Mycodo 8.10.1
2. Upgrade to master
3. Add Synchronous Graph to empty Dashboard
4. See error above.

### Expected behavior

No errors expected.

### Screenshots

N/A

### Additional context

N/A

kizniche, 2021-05-16:
I don't know what this error is, but it's not something that can be replicated from your steps. It seems this is something indicative of a non-functioning system (see [prev issue](https://github.com/kizniche/Mycodo/issues/1001)).

kizniche, 2021-05-16:
Looking into this further, I'm not sure why none of my systems (any anyone else) is getting this error, but I can see why it ***could*** occur.

kizniche, 2021-05-16:
Just committed a probable fix.

dookaloosy, 2021-05-16:
Updated to commit #f951f62 and that fixed the issue.

dookaloosy, 2021-05-16:
Oops closed by mistake.

gecastan, 2021-05-16:
I was about to create an issue. I m having the exact same problem using ADS1256 and mycodo 8.10.1 after creating a 

When modifying the default dashboard, I added a few solid gauge.  
I then went added and tried to add a synchronous graph with all default options  
I only selected a few inputs (using ADS1256 ADC with 7 channels, I selected 3 of the ones I m using)  
Only enabled : auto refresh, X-Axis Reset, Header Buttons along with Graph legend".

It then crashed with the following error (which I reproduced setting up a new dashboard) :

```
Error 500: Internal Server Error

Something bad happened but it's probably not your fault. Letting the developers know about these issues is crucial to supporting Mycodo. Please submit a new issue on GitHub with the following error traceback (copy the entire traceback):

Error (Full Traceback):

Traceback (most recent call last):
  File "/home/yourusername/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/yourusername/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/yourusername/Mycodo/env/lib/python3.7/site-packages/flask_restx/api.py", line 652, in error_router
    return original_handler(e)
  File "/home/yourusername/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/yourusername/Mycodo/env/lib/python3.7/site-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/home/yourusername/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/yourusername/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/yourusername/Mycodo/env/lib/python3.7/site-packages/flask_login/utils.py", line 272, in decorated_view
    return func(*args, **kwargs)
  File "/home/yourusername/Mycodo/mycodo/mycodo_flask/routes_page.py", line 801, in page_dashboard
    form_dashboard=form_dashboard)
  File "/home/yourusername/Mycodo/env/lib/python3.7/site-packages/flask/templating.py", line 140, in render_template
    ctx.app,
  File "/home/yourusername/Mycodo/env/lib/python3.7/site-packages/flask/templating.py", line 120, in _render
    rv = template.render(context)
  File "/home/yourusername/Mycodo/env/lib/python3.7/site-packages/jinja2/environment.py", line 1289, in render
    self.environment.handle_exception()
  File "/home/yourusername/Mycodo/env/lib/python3.7/site-packages/jinja2/environment.py", line 924, in handle_exception
    raise rewrite_traceback_stack(source=source)
  File "/home/yourusername/Mycodo/mycodo/mycodo_flask/templates/pages/dashboard.html", line 3, in top-level template code
    {% set help_page = ["https://kizniche.github.io/Mycodo/Data-Viewing/#dashboard", dict_translation['dashboard']['title']] %}
  File "/home/yourusername/Mycodo/mycodo/mycodo_flask/templates/layout.html", line 375, in top-level template code
    {%- block body %}{% endblock -%}
  File "/home/yourusername/Mycodo/mycodo/mycodo_flask/templates/pages/dashboard.html", line 372, in block 'body'
    {% include 'user_templates/{}'.format(file_js_ready_end) %}
  File "/home/yourusername/Mycodo/mycodo/mycodo_flask/templates/user_templates/widget_template_widget_graph_synchronous_js_ready_end.html", line 150, in top-level template code
    {%- if 'custom_axis_id' in dict_measurements[each_axis_meas] -%}
jinja2.exceptions.UndefinedError: 'collections.OrderedDict object' has no attribute 'none'

```

kizniche, 2021-05-16:
This has already been fixed on [master](https://github.com/kizniche/Mycodo/wiki/Updating-to-Github-Master-Branch).

closed 2021-06-05 by kizniche