kizniche/Mycodo #1008: PID - Input goes missing

Mark0Mi, 2021-05-20:
[debuglog.txt](https://github.com/kizniche/Mycodo/files/6518052/debuglog.txt)

Hi 

I think I have another issue identified within the PID function: 
mycodo: 8.10.1 
configured to lower the PH value and I'm currently fine tuning these settings: 
[image]

can you help me understand why in the logs the Input stops appearing? 
if you look into the logs at around 20:01:08 PID: Input 7.104 is last time logged and the calculation for the output. 
expected behavior would be one minute later PID: Input... but this is missing and it does not come back. 
interestingly the output uses the values from 20:01:08 7.104 and applies the same output in ml but the PH is already below of 7.104. 
I have to deactivate the function and activate it again to trigger a new input line logged with a new calculation. 

it seems the input stops when the setpoint is reached, but the pump is still triggered with the old values... can you help me figure out whats going wrong here? what else do you need?

Mark0Mi, 2021-05-21:
I have to add the following
so when the ph values goes down under the setpoint 
* PID Input is stopped -> expected would be that output is also stopped (there is a coding issue somewhere) 
* after manual disable the PID controller and re-enabling output is not triggered
when the ph raises above the setpoint the PID input returns and output is recalculated (so the problem is just when lowering is active..)

as a workaround I added a function which pauses die PID when the setpoint is under the value defined.

kizniche, 2021-05-22:
I will need to see your PID output to fully understand if this is an expected behavior or not. Please take a screnshot from Data -> Asynchronous Graph for the particular time frame demonstrating the issue, with these selected:

- [Output] Red-PumpPH
- [Input] Atlas pH Ion Concentration
- [PID] PID-Phminus PID P-Value
- [PID] PID-Phminus PID I-Value
- [PID] PID-Phminus PID D-Value
- [PID] PID-Phminus PID Volume
- [PID] PID-Phminus PID Setpoint

kizniche, 2021-05-22:
I could not replicate this issue. With a PID set to only lower volume, volume output stopped once the PID output decreased below 0. Here is a screenshot demonstrating this. I'm merely using the RPi temperature input and raising and lowering the temperature by starting and stopping a CPU stress test.

[image]

[image]

kizniche, 2021-05-22:
Though I do see an issue (unrelated to this one) that needs to be fixed, which is the volume measurements (volume in ml dispensed) are being saved with a timestamp that is at the end of the dispense, rather than at the beginning.

kizniche, 2021-05-22:
On a side note, I would not recommend adjusting pH with a PID controller. I beleive your issue comes down to an inefficiant tuning, and as a result, is overshooting the setpoint. This should be aparent once you view the graph I recommended.

Mark0Mi, 2021-05-22:
I understand - I did build the system to have a very immediate feedback so I have a tank about 3000 liters and the PID tunes this as long as I disable the output when it reaches the setpoint (comming down to the setpoint) 
I would expect that the "I" value (which is the value which does not change) is adjusted but I did not calculate it manually 

here is the graph 
[image]

[image]
you can see we overshoot down to 6.6 ph (then I manually stopped 
activated under the setpoint (6.9) and it immediatley started applying (which in my point of view cannot be overshooting as it is below the setpoint and the only thing configured is lower function on the PID -> it must not apply anything if below the setpoint)

kizniche, 2021-05-22:
> it must not apply anything if below the setpoint

This is not how a PID controller operates. There is often steady state error that necessitates continual output to maintain a setpoint. 

> for the particular time frame demonstrating the issue

Can you zoom in to the particular time frame for just one occurence of the issue. Your graph is not very high resolution because you're viewing several hours rather than ~ 1 hour, which would be ideal.

> I do see an issue (unrelated to this one) that needs to be fixed

I just fixed the Generic Pump Output issue and the data looks good.z

[image]

Mark0Mi, 2021-05-23:
[image]

you are correct - it should not stop directly after hitting the setpoint - and continue but what worries me is the log entries starting when the value is under the setpoint. 
the log I attached in the first ticket - if you search for PID: Input: 
the last line you find PID: Input: you see the calculation of P I & D value - exactly this is missing for the next events (until the reset of the controller) 
so I'm not sure if there is a new calculation or if the previous values keep getting applied - for me just focusing on the logs it seems that the previous output values are applied. I don't see any calculation. (and it is the exactly amount of ml every minute after beeing under the setpoint) -> can this be because of the not tuned PID? 
In the graph I set up I see the calculated PID values -> which is fine and I can work with to fine tune. 
what would you recommend to keep PH under control when not using a PID?

kizniche, 2021-05-24:
> can this be because of the not tuned PID?

Your overshoot is likely due to your high Ki Gain. I would start with it reduced to 1/10 or even 1/100 of what it is now. You also have an extremely low Integrator Min (-100) for the Ki Gain you have set (3.0). This means the lowest possible I value in the output is -300 (ml). This is more than you probably ever would want to dispense at one moment. Although you have a Max Output of 10 ml set, this has no bearing on the I value that can be attached (-300).

> what would you recommend to keep PH under control when not using a PID?

I detail how to do this with a Conditional Controller at https://kylegabriel.com/projects/2020/06/automated-hydroponic-system-build.html

closed 2021-09-20 by kizniche