kizniche/Mycodo #1010: Error 500 dashboard

differentjacob, 2021-05-24:
Error came up when adding new dashboard and when adding widgets:

Error 500: Internal Server Error
Something bad happened but it's probably not your fault. Letting the developers know about these issues is crucial to supporting Mycodo. Please submit a new issue on GitHub with the following error traceback (copy the entire traceback):

Error (Full Traceback):

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_restx/api.py", line 652, in error_router
    return original_handler(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_login/utils.py", line 272, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_page.py", line 801, in page_dashboard
    form_dashboard=form_dashboard)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/templating.py", line 140, in render_template
    ctx.app,
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/templating.py", line 120, in _render
    rv = template.render(context)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/jinja2/environment.py", line 1304, in render
    self.environment.handle_exception()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/jinja2/environment.py", line 925, in handle_exception
    raise rewrite_traceback_stack(source=source)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/dashboard.html", line 3, in top-level template code
    {% set help_page = ["https://kizniche.github.io/Mycodo/Data-Viewing/#dashboard", dict_translation['dashboard']['title']] %}
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/layout.html", line 375, in top-level template code
    {%- block body %}{% endblock -%}
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/dashboard.html", line 372, in block 'body'
    {% include 'user_templates/{}'.format(file_js_ready_end) %}
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/user_templates/widget_template_widget_graph_synchronous_js_ready_end.html", line 150, in top-level template code
    {%- if 'custom_axis_id' in dict_measurements[each_axis_meas] -%}
jinja2.exceptions.UndefinedError: 'collections.OrderedDict object' has no attribute 'pH'

kizniche, 2021-05-24:
Duplicate of #1001

differentjacob, 2021-05-26:
If this is related to #1001, do you know if there is a solution ingoing?  I was reinstalling my raspberry and this appears. I do also ha the latest release 8.10.1. Do you have any idea?

kizniche, 2021-05-26:
It has been fixed and will be available in the next release. See [this comment](https://github.com/kizniche/Mycodo/issues/1001#issuecomment-840756426) for running the yet-to-be released version.

closed 2021-06-05 by kizniche