kizniche/Mycodo #1011: API not capturing conversion_id property update

tilersmyth, 2021-05-25:
### Describe the problem/bug

When converting an input measurement the 'conversion_id' prop of "device measurement settings" in the API stays null while the db reflects the update

### Versions:

 - Mycodo Version: 8.10.1
 - Raspberry Pi Version: 3B+
 - Raspbian OS Version: Raspberry Pi OS Lite

### Reproducibility

Convert an input measurement via UI
GET api/settings/device_measurements/by_device_id/[unique_id]
Bug: DB has conversion_id ref but API does not

thank you

kizniche, 2021-05-25:
I'm not sure what's causing this. Without ```db.ForeignKey('conversion.unique_id')```, it works. I'm not sure how to fix it while keeping the Foreign Key.

tilersmyth, 2021-05-25:
Interesting. I'm not as familiar with SQLAlchemy as I'd like to be, but I know other ORMs (such as TypeORM) require that you have a separate property if you want access to just the related table's id. Ex: "conversion": ForeignKey, "conversion_id": string.. 

Would you consider joining the conversion table for the api? I'd imagine that users would prefer having that data included if they've converted a measurement. 

```json
"device measurements": [
  {
  "channel": 0,
  "conversion": {
      "unique_id": "5b5e378a-42ac-4863-a2da-1e71897b244b",
      "unit": "F"
  },
  ...
  }
],
```

kizniche, 2021-05-30:
That would be fine, I just don't know how to dot hat.

kizniche, 2021-10-25:
> conversion = Nested(ConversionSchema)

What is the import for this?

tilersmyth, 2021-10-31:
Hey - quick fix for this solution. Need to make the change below to ensure this is a LEFT JOIN or "device measurements" will return null if conversion_id is undefined

<pre>- device_id=unique_id).join(DeviceMeasurements.conversion).all(), many=True))</pre>
<pre>+ device_id=unique_id).join(DeviceMeasurements.conversion<b>, isouter=True</b>).all(), many=True))</pre>

I'll close this issue. Thanks for implementing.

closed 2021-10-31 by tilersmyth

kizniche, 2021-11-01:
Thabks for figuring this out