kizniche/Mycodo #1012: Output not activating, and getting Pyro5 TimeoutError errors in daemon log.

LucidEye, 2021-05-26:
Mycodo 8.10.1
RaspberryOS Buster 10
Kernel 5.10.17-v7l+

I recently noticed that one of my outputs keeps turning off occasionally when it shouldn't be, or it isn't turning on when it should.  This seems to be random, and restarting the system seemed to temporarily resolve the problem, but now it's happening right after a reboot.

I noticed that there are Pyro5 TimeoutError errors being reported in the Daemon Log.  It's the same error repeating, and it is for the function that controls the output I am having problems with... the function is "controller_function_4264d6bd".
The output it controls is f514952e
This function says it started ok, but it is having trouble activating the output for some reason?
Can you please help me determine what is causing this and how to fix it?
Here is an example of the error log right after a reboot...

2021-05-26 12:24:02,175 - INFO - mycodo.daemon - All activated Trigger controllers started
2021-05-26 12:24:02,689 - INFO - mycodo.controllers.controller_input_9ec8173c - Activated in 305.0 ms
2021-05-26 12:24:03,052 - INFO - mycodo.controllers.controller_input_893af301 - Activated in 163.9 ms
2021-05-26 12:24:03,474 - INFO - mycodo.controllers.controller_input_a96a507f - Activated in 289.0 ms
2021-05-26 12:24:03,917 - INFO - mycodo.controllers.controller_input_d481115f - Activated in 268.7 ms
2021-05-26 12:24:04,213 - INFO - mycodo.controllers.controller_input_c1ef1284 - Activated in 155.3 ms
2021-05-26 12:24:04,566 - INFO - mycodo.controllers.controller_input_d7f4d84e - Activated in 232.2 ms
2021-05-26 12:24:04,567 - INFO - mycodo.daemon - All activated Input controllers started
2021-05-26 12:24:04,567 - INFO - mycodo.daemon - All activated Math controllers started
2021-05-26 12:24:04,568 - INFO - mycodo.daemon - All activated PID controllers started
2021-05-26 12:24:04,568 - INFO - mycodo.daemon - All activated LCD controllers started
2021-05-26 12:24:04,834 - INFO - mycodo.function.bang_bang_af9b603d - Bang-Bang controller started with options: Measurement Device: a96a507f-02f0-47fe-9857-6507937f440f, Measurement: cc2e3cd4-4278-4941-a951-2fba9899a724, Output: d848e1f0-bb4c-4bfe-bd4f-dcdd63bd5b71, Output_Channel: 0, Setpoint: 1.08, Hysteresis: 0.09, Direction: lower, Period: 20.0
2021-05-26 12:24:04,835 - INFO - mycodo.controllers.controller_function_af9b603d - Activated in 153.6 ms
2021-05-26 12:24:05,048 - INFO - mycodo.function.bang_bang_af9b603d - Input: 1.21206, output: off, target: 1.08, hyst: 0.09
2021-05-26 12:24:05,407 - INFO - mycodo.function.bang_bang_4264d6bd - Bang-Bang controller started with options: Measurement Device: d481115f-863c-4aa7-89b4-922e613d8b2a, Measurement: fa2a915c-9c3d-4b1a-8aaf-61de48c40cc6, Output: f514952e-55ab-4b3e-8bad-34c0cc514741, Output_Channel: 0, Setpoint: 75.0, Hysteresis: 0.8, Direction: lower, Period: 15.0
2021-05-26 12:24:05,408 - INFO - mycodo.controllers.controller_function_4264d6bd - Activated in 244.9 ms
2021-05-26 12:24:05,610 - INFO - mycodo.function.bang_bang_4264d6bd - Input: 79.06702, output: None, target: 75.0, hyst: 0.8
2021-05-26 12:24:05,802 - INFO - mycodo.function.bang_bang_947d8cb5 - Bang-Bang controller started with options: Measurement Device: d481115f-863c-4aa7-89b4-922e613d8b2a, Measurement: fa2a915c-9c3d-4b1a-8aaf-61de48c40cc6, Output: 60617005-d339-4bbe-9114-64db3fe3c3dd, Output_Channel: 0, Setpoint: 79.0, Hysteresis: 1.0, Direction: lower, Period: 15.0
2021-05-26 12:24:05,802 - INFO - mycodo.controllers.controller_function_947d8cb5 - Activated in 172.3 ms
2021-05-26 12:24:05,807 - INFO - mycodo.daemon - All activated Function controllers started
2021-05-26 12:24:05,878 - INFO - mycodo.function.bang_bang_947d8cb5 - Input: 79.06702, output: None, target: 79.0, hyst: 1.0
2021-05-26 12:24:06,530 - INFO - mycodo.controllers.controller_widget - Activated in 722.4 ms
2021-05-26 12:24:21,143 - INFO - mycodo.function.bang_bang_947d8cb5 - Input: 79.06702, output: None, target: 79.0, hyst: 1.0
2021-05-26 12:24:21,665 - INFO - mycodo.daemon - Mycodo daemon started in 9.845 seconds
2021-05-26 12:24:21,670 - INFO - mycodo.daemon - 69.00 MB RAM in use
2021-05-26 12:24:24,965 - INFO - mycodo.function.bang_bang_af9b603d - Input: 1.19144, output: on, target: 1.08, hyst: 0.09
2021-05-26 12:24:35,984 - INFO - mycodo.function.bang_bang_947d8cb5 - Input: 79.14058, output: None, target: 79.0, hyst: 1.0
2021-05-26 12:24:45,052 - INFO - mycodo.function.bang_bang_af9b603d - Input: 1.33762, output: on, target: 1.08, hyst: 0.09
2021-05-26 12:24:49,770 - ERROR - mycodo.controllers.controller_function_4264d6bd - Pyro5 TimeoutError
Traceback (most recent call last):
  File "/var/mycodo-root/env/lib/python3.7/site-packages/Pyro5/socketutil.py", line 132, in receive_data
    chunk = sock.recv(size, socket.MSG_WAITALL)
socket.timeout: timed out

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/base_controller.py", line 86, in run
    self.loop()
  File "/var/mycodo-root/mycodo/controllers/controller_function.py", line 70, in loop
    self.run_function.loop()
  File "/home/pi/Mycodo/mycodo/functions/bang_bang.py", line 216, in loop
    output_channel=self.output_channel)
  File "/var/mycodo-root/mycodo/mycodo_client.py", line 222, in output_on
    output_channel=output_channel, trigger_conditionals=trigger_conditionals)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/Pyro5/client.py", line 476, in __call__
    return self.__send(self.__name, args, kwargs)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/Pyro5/client.py", line 224, in _pyroInvoke
    msg = protocol.recv_stub(self._pyroConnection, [protocol.MSG_RESULT])
  File "/var/mycodo-root/env/lib/python3.7/site-packages/Pyro5/protocol.py", line 189, in recv_stub
    header = connection.recv(6)  # 'PYRO' + 2 bytes protocol version
  File "/var/mycodo-root/env/lib/python3.7/site-packages/Pyro5/socketutil.py", line 435, in recv
    return receive_data(self.sock, size)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/Pyro5/socketutil.py", line 140, in receive_data
    raise TimeoutError("receiving: timeout")
Pyro5.errors.TimeoutError: receiving: timeout
2021-05-26 12:24:50,094 - INFO - mycodo.function.bang_bang_4264d6bd - Input: 79.22332, output: on, target: 75.0, hyst: 0.8
2021-05-26 12:24:50,618 - ERROR - mycodo.daemon - Could not turn output on: receiving: timeout
[... 14 lines cut ...]
  File "/var/mycodo-root/mycodo/outputs/base_output.py", line 593, in check_triggers
    each_trigger.unique_id, message=message)
  File "/var/mycodo-root/mycodo/mycodo_client.py", line 167, in trigger_all_actions
    function_id, message=message, debug=debug)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/Pyro5/client.py", line 476, in __call__
    return self.__send(self.__name, args, kwargs)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/Pyro5/client.py", line 224, in _pyroInvoke
    msg = protocol.recv_stub(self._pyroConnection, [protocol.MSG_RESULT])
  File "/var/mycodo-root/env/lib/python3.7/site-packages/Pyro5/protocol.py", line 189, in recv_stub
    header = connection.recv(6)  # 'PYRO' + 2 bytes protocol version
  File "/var/mycodo-root/env/lib/python3.7/site-packages/Pyro5/socketutil.py", line 435, in recv
    return receive_data(self.sock, size)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/Pyro5/socketutil.py", line 140, in receive_data
    raise TimeoutError("receiving: timeout")
Pyro5.errors.TimeoutError: receiving: timeout
2021-05-26 12:24:50,639 - INFO - mycodo.function.bang_bang_4264d6bd - Input: 79.22332, output: on, target: 75.0, hyst: 0.8
2021-05-26 12:24:51,048 - INFO - mycodo.function.bang_bang_947d8cb5 - Input: 79.22332, output: None, target: 79.0, hyst: 1.0
2021-05-26 12:25:04,868 - INFO - mycodo.function.bang_bang_af9b603d - Input: 1.33259, output: on, target: 1.08, hyst: 0.09
2021-05-26 12:25:05,478 - INFO - mycodo.function.bang_bang_4264d6bd - Input: 79.28768, output: on, target: 75.0, hyst: 0.8
2021-05-26 12:25:05,879 - INFO - mycodo.function.bang_bang_947d8cb5 - Input: 79.28768, output: None, target: 79.0, hyst: 1.0
2021-05-26 12:25:20,581 - INFO - mycodo.function.bang_bang_4264d6bd - Input: 79.34285, output: off, target: 75.0, hyst: 0.8
2021-05-26 12:25:21,097 - INFO - mycodo.function.bang_bang_947d8cb5 - Input: 79.34285, output: None, target: 79.0, hyst: 1.0
2021-05-26 12:25:24,939 - INFO - mycodo.function.bang_bang_af9b603d - Input: 1.30741, output: on, target: 1.08, hyst: 0.09
2021-05-26 12:25:35,930 - INFO - mycodo.function.bang_bang_947d8cb5 - Input: 79.44398, output: None, target: 79.0, hyst: 1.0
2021-05-26 12:25:45,038 - INFO - mycodo.function.bang_bang_af9b603d - Input: 1.27433, output: on, target: 1.08, hyst: 0.09

kizniche, 2021-05-28:
>File "/var/mycodo-root/mycodo/mycodo_client.py", line 167, in trigger_all_actions
function_id, message=message, debug=debug)

What controller is executing trigger_all_actions()?

LucidEye, 2021-05-28:
None that I am aware of.
I just went through and double checked all my output and function settings again.
I don't currently have, nor have I ever had any controllers setup to "trigger_all_actions".
I don't have any of my outputs set to "trigger functions at startup".
In Configure - General, I have "Hide success messages" and "Hide info messages" checked... that's the only other setting I changed from the "default" that I can think of.
Do you have any ideas on what might be causing the Pyro5 TimeoutError that seems to initiate these exception errors?

kizniche, 2021-05-29:
> File "/var/mycodo-root/mycodo/outputs/base_output.py", line 593, in check_triggers
each_trigger.unique_id, message=message)

If this is executing, you have a trigger that is responding to an output changing its state.

LucidEye, 2021-05-31:
Yes, this error is occurring in relation to a trigger that is responding to an output changing it's state.  But I don't understand what is causing the timeouts that lead to the exception errors. 
This cascade always starts with this timeout error: 

"ERROR - mycodo.controllers.controller_function_4264d6bd - Pyro5 TimeoutError
Traceback (most recent call last):
  File "/var/mycodo-root/env/lib/python3.7/site-packages/Pyro5/socketutil.py", line 132, in receive_data
    chunk = sock.recv(size, socket.MSG_WAITALL)
socket.timeout: timed out"

Function 4264d6bd is a Bang Bang controller.  It turns on a pump when the temp gets too high.
I have another function that checks the state of that pump output, if the pump turns on, first there is a pause actions for 60 seconds, then a fan turns on.
The timeout seems to be saying it can't turn on the pump output, but it is turning that output on, so I don't understand why this error is popping up.

kizniche, 2021-05-31:
> What controller is executing

Please provide details of the controller

LucidEye, 2021-06-06:
My apologies for not answering sooner.  Things have suddenly become a bit hectic for me.
Anyway.. here are the settings for the trigger that is having problems. 
[image]

What other info or details do you need?

LucidEye, 2021-06-06:
Just saw that there was an update release.  I'm now on Mycodo 8.11.0

kizniche, 2021-06-06:
That's not a Trigger. Triggers have their name prefixed with a "Trigger", e.g when x output changes its state to y, do z.

LucidEye, 2021-06-06:
Sorry... this is the trigger that is activated when the bang bang turns on the pump...
[image]

kizniche, 2021-06-06:
Thanks. Can you provide screenshots of the settings of Outputs 8 and 11?

kizniche, 2021-06-09:
I think I see the issue, and I already found it elsewhere, which was the cause for adding a Timeout option for Conditional Controllers: https://github.com/kizniche/Mycodo/commit/44ad86ae15460c42f59edae5ba1c13cb697950b8#diff-a5311de1dbb376a4abe5e61d2c6fbec6131d00985806736f280e50a6149d6ca1

I'll see about adding a Timeout option for Triggers as well.

LucidEye, 2021-06-09:
Here are the outputs...

Output 8
[image]

Output 11
[image]

I know, I know... wireless is not your thing :-P ... but this does seem like a software bug and not an issue with receiver interference.  
If I'm understanding the log correctly, the signal is never getting transmitted in the first place, because the socket timeout error is causing that command to fail? 
I've loosely confirmed this by monitoring the TX signal with an SDR receiver and RF monitoring software... the radio signal is not being transmitted when that error occurs, which seems to indicate the Pi never got the command to turn on that GPIO pin to energize the TX board?
Anyway, long story short, I have another evap cooler in my shop and I finally found a way to get the remote on that wireless relay to transmit separate codes for on and off... so I thought I'd give it a try again.
It works very reliably if I leave the Startup and Shutdown State set to "Do Nothing" and uncheck the Force Command box so that it only sends the on or off signal once. And it works perfectly, up until I get that socket timeout error, which initiates the cascading exception errors. 
So do you think this is a bug?

kizniche, 2021-06-09:
It's not a bug, per se, you're just using a 90 second pause when the default timeout is 30 seconds. Adding the ability to change the timeout will allow you to set the timeout greater than 90 seconds, which will fix the issue.

LucidEye, 2021-06-09:
Got it.  Thank you so much.

closed 2021-08-30 by kizniche