kizniche/Mycodo #1014: Multiple measurements for MQTT

timuritu, 2021-05-28:
Whenever I add an input as MQTT and select more than 1 measurement, as I save I get the 500 Error and have to revert to a previous state or delete all Inputs.

This is the error log from the screen:
Error (Full Traceback):

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_restx/api.py", line 652, in error_router
    return original_handler(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_login/utils.py", line 272, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_page.py", line 2166, in page_input
    devices_1wire_w1thermsensor=devices_1wire_w1thermsensor)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/templating.py", line 140, in render_template
    ctx.app,
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/templating.py", line 120, in _render
    rv = template.render(context)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/jinja2/environment.py", line 1304, in render
    self.environment.handle_exception()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/jinja2/environment.py", line 925, in handle_exception
    raise rewrite_traceback_stack(source=source)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/input.html", line 3, in top-level template code
    {% set help_page = ["https://kizniche.github.io/Mycodo/Inputs/", dict_translation['input']['title']] %}
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/layout.html", line 375, in top-level template code
    {%- block body %}{% endblock -%}
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/input.html", line 35, in block 'body'
    {% include 'pages/data_options/input.html' %}
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/data_options/input.html", line 297, in top-level template code
    {%- elif 'channels_dict' in dict_inputs[each_input.device] and
jinja2.exceptions.UndefinedError: dict object has no element 1

kizniche, 2021-05-28:
Duplicate of #1001

kizniche, 2021-05-28:
I can't replicate this issue, which likely means it's already been fixed. You're welcome to [upgrade to master](https://github.com/kizniche/Mycodo/wiki/Updating-to-Github-Master-Branch) and test the latest code to see if it has indeed already been fixed.

timuritu, 2021-05-28:
Just tried it out. It works now. No error. I can select multiple topics.

Thank you.

closed 2021-06-05 by kizniche