kizniche/Mycodo #1015: Error 500 on setting up Graph (Synchronous)

GEGR667, 2021-05-29:
Hi - I'm just getting started with MyCodo and I was setting up a Graph (Synchronous) for my Atlas pH probe. (The proble hasn't been calibrated yet. The reading is with the storage solution). When I selected the graph options (possibly, I selected incompatible options?) and clicked on Create, I got an Error 500. The Dashboard can still be selected in the Data menu, and when click on it goes to the Error 500 below. Also, I can't seem to find a way to delete this dashboard outside of selecting it and then selecting the deletion option. Thanks 

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_restx/api.py", line 652, in error_router
    return original_handler(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_login/utils.py", line 272, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_page.py", line 801, in page_dashboard
    form_dashboard=form_dashboard)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/templating.py", line 140, in render_template
    ctx.app,
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/templating.py", line 120, in _render
    rv = template.render(context)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/jinja2/environment.py", line 1304, in render
    self.environment.handle_exception()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/jinja2/environment.py", line 925, in handle_exception
    raise rewrite_traceback_stack(source=source)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/dashboard.html", line 3, in top-level template code
    {% set help_page = ["https://kizniche.github.io/Mycodo/Data-Viewing/#dashboard", dict_translation['dashboard']['title']] %}
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/layout.html", line 375, in top-level template code
    {%- block body %}{% endblock -%}
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/dashboard.html", line 372, in block 'body'
    {% include 'user_templates/{}'.format(file_js_ready_end) %}
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/user_templates/widget_template_widget_graph_synchronous_js_ready_end.html", line 150, in top-level template code
    {%- if 'custom_axis_id' in dict_measurements[each_axis_meas] -%}
jinja2.exceptions.UndefinedError: 'collections.OrderedDict object' has no attribute 'pH'

kizniche, 2021-05-29:
Duplicate of #1001

GEGR667, 2021-05-29:
Thanks. Just realized I could delete the dashboards from settings.

closed 2021-06-05 by kizniche