kizniche/Mycodo #1019: pH live graphe not displayed

pe2269, 2021-06-07:
Please first read: How to write a good bug report at https://forum.kylegabriel.com/t/how-to-write-a-good-bug-report/71

Please DO NOT OPEN AN ISSUE:

 - If your Mycodo version is not the latest release version, please update your device before submitting your issue (unless your issue is related to not being able to upgrade). Your problem might already be solved.

 - If your issue has been addressed before (i.e., duplicated issue), please ask in the original issue.

Please complete as many of the sections below, if applicable, to provide the most information that may help with investigating your issue. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help those providing assistance by reducing the time required to help you.

### Describe the problem/bug
Hi,
Firstly i would like to thank you for all your great job on this software, it's very impressive !!

I have a raspberry pi 4 with the Tentacle T3 from Atlas Scientific and 4 ezo devices in I2C mode :
a RTD sensor (PT1000)  I2C adress 102
a Dissolved oxygen probe I2C adress 97
a pH probe I2C adress 99
a peristaltic pump I2C adress 103

I have no problem to read all devices data and to show widgets (show capture) but when i want specifically added the  synchronous graph widget for the pH, nothing happen.  I have the  success message "Graph (Synchronous) with ID 23 sucessfully added) but nothing appears (see capture).
I have no problem with the others parameters...

The daemon log show nothing special (see capture)

I precise that i have not calibrate the pH probe yet  ...

I am totally new in Mycodo and Raspberry so 

### Versions:
Version Mycodo: 8.11.0
Version Python: 3.7.3 (default, Jan 22 2021, 20:04:44) [GCC 8.3.0]
Linux raspberrypi 5.10.17-v7l+ #1421 SMP Thu May 27 14:00:13 BST 2021 armv7l GNU/Linux
Raspbian GNU/Linux 10 (buster)
 
### Reproducibility

Please list specific setup details that are involved and the steps to reproduce the behavior:

1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

### Expected behavior

A clear and concise description of what you expected to happen.

### Screenshots
[image]

### Additional context

Is there anything that should be added to make it easier to address this issue?

kizniche, 2021-06-07:
Please list steps to replicate the issue. Thanks.

pe2269, 2021-06-07:
1) I click on add dashboard
2) I click on add dashboard widget
3) i select Graph (Synchronous)
4) I choose in the Contributions list : [Input 03 CH0] Atlas pH Concentrations d'ions (pH)
5) I click on Create
[image]

kizniche, 2021-06-07:
Is there an error in the Web log?

pe2269, 2021-06-07:
i don't hink so  :/
this capture has been made after reboot and a try to add the pH synchronous Widget
[image]

kizniche, 2021-06-07:
See if changing the name of the Input to something without special characters fixes it.

pe2269, 2021-06-07:
Yes thank you very much
It's the french translation of ion concentration in :" Concentration d'ions" which made the bug !!!
I try it in english and it work well !!

Great thank you

kizniche, 2021-06-07:
I named one of my inputs "d'ions" and I still couldn't replicate the issue, so I'm not sure what the problem is or how to fix it.

pe2269, 2021-06-07:
I think it's not the name that i can change it's the "automatic" name of the uuid that create the problem

pe2269, 2021-06-07:
in the configuration page : 
1) i change the language to English, before it was Browser defaut and i'am french...
And the graph appears :)

kizniche, 2021-06-07:
I don't understand. You suggested changing the Input name fixed the issue, therefore changing and Input name back should cause the issue to present itself.

kizniche, 2021-06-07:
Okay, that's different than what I suggested to do (change the Input name), which you then said fixed the issue.

kizniche, 2021-06-07:
I'll see if I can replicate the issue later and try to fix it.

pe2269, 2021-06-07:
> Okay, that's different than what I suggested to do (change the Input name), which you then said fixed the issue.

Yes i misunderstand what you want to do, i just change the language in the configs ! :)

thank you for your great help !!

closed 2021-08-01 by pe2269

cperricos, 2022-12-02:
Hello,
I would like to use Mycodo for a swimming pool dosing system using a whitebox (Atlas) Tentacle T3 and EC, PH and Temp sensors. I have set it all up, enabled the I2C bus, but cannot get a reading from any of the sensors. Should they work out of the box when added as inputs, or do I need to configure something specifically to use the T3?

thanks

cperricos, 2022-12-02:
Just an update on my above query. Have looked at the log, and the sensors appear to be providing reaidng. I cannot display them or see the reading on the UI though?

log:

2022-12-02 14:14:11,448 - INFO - mycodo.daemon - All activated Input controllers started
2022-12-02 14:14:11,449 - INFO - mycodo.daemon - All activated PID controllers started
2022-12-02 14:14:11,450 - INFO - mycodo.daemon - All activated Function controllers started
2022-12-02 14:14:11,777 - INFO - mycodo.controllers.controller_widget - Activated in 327.0 ms
2022-12-02 14:14:12,779 - INFO - mycodo.daemon - Mycodo daemon started in 13.357 seconds
2022-12-02 14:14:12,835 - INFO - mycodo.daemon - 80.40 MB RAM in use
2022-12-02 14:14:12,957 - DEBUG - mycodo.inputs.atlas_pt1000_d341bede - Device Returned: success: 18.183
2022-12-02 14:14:12,992 - DEBUG - mycodo.controllers.controller_input_d341bede - Adding measurements to InfluxDB with ID d341bede-34e2-4bd9-9458-2bd73ee348c5: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 18.183, 'timestamp_utc': datetime.datetime(2022, 12, 2, 14, 14, 12, 957607)}}
2022-12-02 14:14:20,767 - DEBUG - mycodo.inputs.atlas_orp_edac5e65 - Device Returned: success: 324.2
2022-12-02 14:14:20,780 - DEBUG - mycodo.controllers.controller_input_edac5e65 - Adding measurements to InfluxDB with ID edac5e65-3356-45c9-a813-ff6e994aa74c: {0: {'measurement': 'oxidation_reduction_potential', 'unit': 'mV', 'value': 324.2, 'timestamp_utc': datetime.datetime(2022, 12, 2, 14, 14, 20, 768852)}}
2022-12-02 14:14:23,323 - DEBUG - mycodo.inputs.atlas_ph_fd857dff - Device Returned: success: 7.675
2022-12-02 14:14:23,331 - DEBUG - mycodo.controllers.controller_input_fd857dff - Adding measurements to InfluxDB with ID fd857dff-5c19-4ccf-878c-73b95dcd89bf: {0: {'measurement': 'ion_concentration', 'unit': 'pH', 'value': 7.675, 'timestamp_utc': datetime.datetime(2022, 12, 2, 14, 14, 23, 324159)}}
2022-12-02 14:14:25,900 - DEBUG - mycodo.inputs.atlas_pt1000_d341bede - Device Returned: success: 18.186
2022-12-02 14:14:25,914 - DEBUG - mycodo.controllers.controller_input_d341bede - Adding measurements to InfluxDB with ID d341bede-34e2-4bd9-9458-2bd73ee348c5: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 18.186, 'timestamp_utc': datetime.datetime(2022, 12, 2, 14, 14, 25, 901767)}}
2022-12-02 14:14:35,731 - DEBUG - mycodo.inputs.atlas_orp_edac5e65 - Device Returned: success: 324.0
2022-12-02 14:14:35,739 - DEBUG - mycodo.controllers.controller_input_edac5e65 - Adding measurements to InfluxDB with ID edac5e65-3356-45c9-a813-ff6e994aa74c: {0: {'measurement': 'oxidation_reduction_potential', 'unit': 'mV', 'value': 324.0, 'timestamp_utc': datetime.datetime(2022, 12, 2, 14, 14, 35, 732074)}}
2022-12-02 14:14:38,349 - DEBUG - mycodo.inputs.atlas_ph_fd857dff - Device Returned: success: 7.678
2022-12-02 14:14:38,357 - DEBUG - mycodo.controllers.controller_input_fd857dff - Adding measurements to InfluxDB with ID fd857dff-5c19-4ccf-878c-73b95dcd89bf: {0: {'measurement': 'ion_concentration', 'unit': 'pH', 'value': 7.678, 'timestamp_utc': datetime.datetime(2022, 12, 2, 14, 14, 38, 350170)}}
2022-12-02 14:14:40,941 - DEBUG - mycodo.inputs.atlas_pt1000_d341bede - Device Returned: success: 18.189
2022-12-02 14:14:40,954 - DEBUG - mycodo.controllers.controller_input_d341bede - Adding measurements to InfluxDB with ID d341bede-34e2-4bd9-9458-2bd73ee348c5: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 18.189, 'timestamp_utc': datetime.datetime(2022, 12, 2, 14, 14, 40, 947170)}}
2022-12-02 14:14:50,764 - DEBUG - mycodo.inputs.atlas_orp_edac5e65 - Device Returned: success: 324.1
2022-12-02 14:14:50,772 - DEBUG - mycodo.controllers.controller_input_edac5e65 - Adding measurements to InfluxDB with ID edac5e65-3356-45c9-a813-ff6e994aa74c: {0: {'measurement': 'oxidation_reduction_potential', 'unit': 'mV', 'value': 324.1, 'timestamp_utc': datetime.datetime(2022, 12, 2, 14, 14, 50, 765132)}}
2022-12-02 14:14:53,368 - DEBUG - mycodo.inputs.atlas_ph_fd857dff - Device Returned: success: 7.676
2022-12-02 14:14:53,376 - DEBUG - mycodo.controllers.controller_input_fd857dff - Adding measurements to InfluxDB with ID fd857dff-5c19-4ccf-878c-73b95dcd89bf: {0: {'measurement': 'ion_concentration', 'unit': 'pH', 'value': 7.676, 'timestamp_utc': datetime.datetime(2022, 12, 2, 14, 14, 53, 369396)}}
2022-12-02 14:14:55,873 - DEBUG - mycodo.inputs.atlas_pt1000_d341bede - Device Returned: success: 18.191
2022-12-02 14:14:55,880 - DEBUG - mycodo.controllers.controller_input_d341bede - Adding measurements to InfluxDB with ID d341bede-34e2-4bd9-9458-2bd73ee348c5: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 18.191, 'timestamp_utc': datetime.datetime(2022, 12, 2, 14, 14, 55, 873920)}}
2022-12-02 14:15:05,702 - DEBUG - mycodo.inputs.atlas_orp_edac5e65 - Device Returned: success: 324.1
2022-12-02 14:15:05,710 - DEBUG - mycodo.controllers.controller_input_edac5e65 - Adding measurements to InfluxDB with ID edac5e65-3356-45c9-a813-ff6e994aa74c: {0: {'measurement': 'oxidation_reduction_potential', 'unit': 'mV', 'value': 324.1, 'timestamp_utc': datetime.datetime(2022, 12, 2, 14, 15, 5, 703274)}}
2022-12-02 14:15:08,299 - DEBUG - mycodo.inputs.atlas_ph_fd857dff - Device Returned: success: 7.675
2022-12-02 14:15:08,306 - DEBUG - mycodo.controllers.controller_input_fd857dff - Adding measurements to InfluxDB with ID fd857dff-5c19-4ccf-878c-73b95dcd89bf: {0: {'measurement': 'ion_concentration', 'unit': 'pH', 'value': 7.675, 'timestamp_utc': da [... cut ...]
2022-12-02 14:15:38,344 - DEBUG - mycodo.inputs.atlas_ph_fd857dff - Device Returned: success: 7.676
2022-12-02 14:15:38,351 - DEBUG - mycodo.controllers.controller_input_fd857dff - Adding measurements to InfluxDB with ID fd857dff-5c19-4ccf-878c-73b95dcd89bf: {0: {'measurement': 'ion_concentration', 'unit': 'pH', 'value': 7.676, 'timestamp_utc': datetime.datetime(2022, 12, 2, 14, 15, 38, 344869)}}
2022-12-02 14:15:40,937 - DEBUG - mycodo.inputs.atlas_pt1000_d341bede - Device Returned: success: 18.198