kizniche/Mycodo #1023: Unable to add CCS881 CO2 w/o temp sensor

escapingsamsara, 2021-06-11:
### Describe the problem/bug

When trying to add a CCS881 CO2 w/o temperature sensor I get stuck in a dependency requirement loop saying adafruit_bme280 is needed even though it's been installed repeatedly.

### Versions:

 - Mycodo Version: 8.11.0
 - Raspberry Pi Version: 3B+
 - Raspbian OS Version: Buster Lite

### Reproducibility

1. Setup > Input
2. Select "AMS: CCS811 (without Temperature): CO2/VOC (Adafruit_CirctuitPython_CCS881) [I2C]" and click Add
3. Get message about needing dependencies so I install them.
4. I repeat steps 1 and 2 but receive message saying adafruit_bme280 is still needed over and over despite repeated successful installs. 

### Expected behavior

Required dependencies get installed so the sensor can be used for CO2 measurement. 

### Screenshots

[image]

kizniche, 2021-06-11:
Thanks for the bug report. I just pushed a fix. Go ahead and [upgrade to master](https://github.com/kizniche/Mycodo/wiki/Updating-to-Github-Master-Branch) and see if the issue persists.

escapingsamsara, 2021-06-11:
It's working now after updating to master. Thanks for the extremely fast fix. You rock!

kizniche, 2021-06-11:
You're welcome. Let's leave the issue open until the release else we risk someone else opening a duplicate issue.

closed 2021-08-30 by kizniche