kizniche/Mycodo #1024: Conditional function debug log always includes code

dookaloosy, 2021-06-14:
### Describe the problem/bug

When a Conditional function has its "Log Level: Debug" checkbox enabled, the resulting debug message in the Daemon Log always includes the code. Every time the Conditional runs, the code is printed again.

### Versions:

 - Mycodo Version: 8.11.0
 - Raspberry Pi Version: 3B
 - Raspbian OS Version: Linux raspberrypi 5.10.17-v7+ #1403 SMP Mon Feb 22 11:29:51 GMT 2021 armv7l GNU/Linux

### Reproducibility

1. Create Conditional function.
2. Enable "Log Level: Debug" but do not enable "Message Includes Code".
3. Check Daemon Log.
4. See code.

### Expected behavior

No code should be in debug log unless "Message Includes Code" is checked.

### Screenshots

N/A

### Additional context

N/A

kizniche, 2021-06-14:
This is the expected behavior. The "message" in the Message Includes Code option refers to "self.message" that is passed to all actions. You can observe this effect if you have an action email you and you pass the action self.mesaage, your email notification will either include the Conditional code or not based on the setting.

kizniche, 2021-06-14:
A better alternate behavior is probably to only print the code once at activation rather than at every execution.

dookaloosy, 2021-06-14:
> This is the expected behavior.

I see.

> A better alternate behavior is probably to only print the code once at activation rather than at every execution.

I agree. The log will fill very quickly if the code is printed at every execution.

kizniche, 2021-06-25:
I mean to mention that I pushed a fix several days ago.

closed 2021-08-30 by kizniche