kizniche/Mycodo #1026: Calibrating ph

jcastro, 2021-06-17:
Version: 8.11.0
Database: 43c72773dbe8
Model: Raspberry Pi 4 Model B Rev 1.1

Release:
Distributor ID: Raspbian
Description: Raspbian GNU/Linux 10 (buster)
Release: 10
Codename: buster

Firmware:
May 27 2021 14:03:06
Copyright (c) 2012 Broadcom
version 7d9a298cda813f747b51fe17e1e417e7bf5ca94d (clean) (release) (start_x)

Error (Full Traceback):

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_restx/api.py", line 652, in error_router
    return original_handler(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_login/utils.py", line 272, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_calibration.py", line 515, in setup_atlas_ph
    current_stage='temperature', next_stage=next_stage)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_calibration.py", line 710, in dual_commands_to_sensor
    first_cmd, set_amount=set_amount)
  File "/home/pi/Mycodo/mycodo/utils/atlas_calibration.py", line 103, in calibrate
    err, msg = self.send_command('T,{:.2f}'.format(set_amount))
ValueError: Unknown format code 'f' for object of type 'str'

This happens when trying to calibrate the pH sensor from Atlas Scientific. Hope it helps!

kizniche, 2021-06-17:
Duplicate of #1022

closed 2021-06-17 by kizniche

jcastro, 2021-06-17:
Oh I'm sorry, didn't see that. I'm not entirely sure what's the procedure to calibrate from the input but I'm assuming it's just putting the probe in that value solution, wait a few minutes and hit the "calibrate [XXX point]" ?

Thanks a lot for your awesome work!