kizniche/Mycodo #1028: Possible issue with KP303 Kasa output

thzerong, 2021-06-18:
Hi all,

First and foremost, thank you all for the work put into this fantastic software and a bigger thanks to Kyle for bringing this project to fruition in the first place! This little farm I set up with Mycodo is the most fun I have ever had.

I am running Mycodo 8.11.0 on a raspberry pi 4b. I am new to python and have been learning as I go working with Mycodo.

My Kasa KP303 is the US version if that matters.

Can anyone tell me if I am missing something here or if this is an actual issue?

Using the provided KP303 Kasa output, Mycodo will not connect to my KP303. It is my understanding here that it should just be the IP that is all that is needed to form a successful connection.

This error is thrown at the top after saving the configuration which leaves the 3 channels in an unconfigured state

> Error: Modify Output: Could not connect to Daemon: receiving: timeout

This was repeated in the Daemon logs

> 2021-06-18 11:32:24,696 - ERROR - mycodo.outputs.kp303_466bf081 - Output was unable to be setup: Unable to query the device: unpack requires a buffer of 4 bytes

To debug this I tried to connect to the same KP303 using the latest python-kasa 0.4.0.dev3 on the same pi (but on the pi environment? not sure if I phrased this correctly but it is installed on the pi outside of Mycodo). 

This worked beautifully and I was able to set up the KP303 switch over wifi without the Kasa app and was able to control it with node-red. I was able to retrieve all the info about plug channels, status, names, etc. so I don't think the issue is with my pi's network setup.

I then noticed that the python-kasa library included with Mycodo is 0.4.0.dev2 so I tried to figure out how to update it to dev3 hoping that would be the issue but this is where I found myself to be in way over my head lol.

Hope I have provided enough information!
Thanks!

kizniche, 2021-06-19:
I just pushed a fix.

Try deleting your Input, then issue this command:

```
sudo ~/Mycodo/env/bin/pip uninstall python-kasa
```

 [upgrade to master](https://github.com/kizniche/Mycodo/wiki/Updating-to-Github-Master-Branch), then add the input again (which should prompt you to install the new library), and see if the issue persists.

thzerong, 2021-06-19:
`sudo ~/Mycodo/env/bin/pip uninstall python-kasa`

was giving me

```
Found existing installation: python-kasa 0.4.0.dev2
Not uninstalling python-kasa at /usr/local/lib/python3.7/dist-packages, outside environment /home/pi/Mycodo/env
Can't uninstall 'python-kasa'. No files were found to uninstall.
```

but a normal pip uninstall command listed that the dev2 was installed so I uninstalled that.

Then I upgraded to master and added the Kasa output and it prompted to install the new library as expected.

Here it gets a bit weird, here is the daemon log along with my inputs:

**Plugged in the wrong IP ( 192.168.1.10 ), it shows that the IP was saved but does not connect**

`2021-06-19 15:23:30,605 - ERROR - mycodo.outputs.kp303_d2626355 - Output was unable to be setup: Unable to query the device: [Errno 111] Connect call failed ('192.168.1.10', 9999)`

**Plugged in another wrong IP ( 192.168.1.150 ), same thing happens**

`2021-06-19 15:23:36,880 - ERROR - mycodo.outputs.kp303_d2626355 - Output was unable to be setup: Unable to query the device: [Errno 113] Connect call failed ('192.168.1.150', 9999)`

**Plugged in the correct IP ( 192.168.1.15 ), daemon times out and shows the previously entered IP while the output configuration page is still showing 192.168.1.15**

```
2021-06-19 15:23:36,882 - INFO - mycodo.outputs.kp303_d2626355 - Initialized in 12626.0 ms
2021-06-19 15:23:49,359 - ERROR - mycodo.outputs.kp303_d2626355 - Output was unable to be setup: Unable to query the device: [Errno 113] Connect call failed ('192.168.1.150', 9999)
```

I want to also note that this was all done on a fresh install of Mycodo since I blew up my previous one. I also tried on a second KP303 and got the same errors.

Thanks!

kizniche, 2021-06-25:
I'm not sure how to fix this. My KP303 is working as expected.

>To debug this I tried to connect to the same KP303 using the latest python-kasa 0.4.0.dev3... This worked
Can you provide the code you used to do this so I can compare to the output module?

thzerong, 2021-06-29:
Hey! Sorry for the delay. I think I may have finally found the issue and I am pretty sure the problem is in my networking. 

To answer your question though, I simply followed the [documentation here](https://python-kasa.readthedocs.io/en/latest/cli.html?highlight=provision#provisioning). I just connected to the KP303's wifi and sent the commands.

So I've been reading around and learned that Kasa devices don't perform well with high wifi interference. My router is a bit far from the pi and KP303, also the 2.4ghz band is highly impacted where I am. When I was provisioning the KP303, it was through a direct line of sight wifi connection between my laptop and the switch itself so I am assuming that is how it worked without a hitch.

I've since switched to GPIO for controlling relays until I can get an AP out in the yard. I'm also setting up another pi closer to the router tonight to actually confirm my theory.

One more thing, I just remembered that my rpi is getting wifi through a USB dongle. That wouldn't affect anything, would it?

Thanks for your help Kyle!

closed 2021-06-30 by thzerong

harrycooke, 2021-10-15:
For other people experiencing this bug who come across this issue like I did, the bug is not related to router strength. The dependency installed on the current stable release dev3 (8.12.6) hangs for my recently purchased kasa kp303 power strip while the latest version of python-kasa works. You can confirm this is the case by adding the mycodo python deps to your path and trying to turn on one of the plugs with: `kasa --strip --host <host> on --input 0`  and it should hang then uninstalling and installing the latest version via pip and testing that with the same command. 

Luckily it looks like @kizniche updated the kp303 version in a commit a few days ago and if you [upgrade to master](https://github.com/kizniche/Mycodo/wiki/Updating-to-Github-Master-Branch) it works perfectly.