kizniche/Mycodo #1029: Disable option for "Enable Pin optional" (L298N Output) not workinb

jcastro, 2021-06-19:
Please first read: How to write a good bug report at https://forum.kylegabriel.com/t/how-to-write-a-good-bug-report/71

Please DO NOT OPEN AN ISSUE:

 - If your Mycodo version is not the latest release version, please update your device before submitting your issue (unless your issue is related to not being able to upgrade). Your problem might already be solved.

 - If your issue has been addressed before (i.e., duplicated issue), please ask in the original issue.

Please complete as many of the sections below, if applicable, to provide the most information that may help with investigating your issue. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help those providing assistance by reducing the time required to help you.

### Describe the problem/bug

A clear and concise description of what you were trying to do and what the problem/bug is.

### Versions:

 - Mycodo Version: [e.g. 8.0.3]
 - Raspberry Pi Version: [e.g. 3B+]
 - Raspbian OS Version: [e.g. Buster Lite]

### Reproducibility

Please list specific setup details that are involved and the steps to reproduce the behavior:

1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

### Expected behavior

A clear and concise description of what you expected to happen.

### Screenshots

If applicable, add screenshots to help explain your problem.

### Additional context

Is there anything that should be added to make it easier to address this issue?

kizniche, 2021-06-19:
Are you clicking Save?

jcastro, 2021-06-19:
> Are you clicking Save?

Yes, after clicking save the page refreshes and I force refresh as well just to make sure it's not cache or something, but the option stays enabled

kizniche, 2021-06-19:
Are there any error messages at the top of the page or in the Web Log?

jcastro, 2021-06-19:
I've recorded a video to show it more in detail. I've checked the web log as well but no related errors there

https://user-images.githubusercontent.com/190036/122652824-32c85000-d141-11eb-87f1-7565a36179ac.mp4

Web log
```
-- Logs begin at Sat 2021-06-19 20:35:02 CEST, end at Sat 2021-06-19 20:58:17 CEST. --
jun 19 20:49:31 hydropi systemd[1]: Started gunicorn daemon for Mycodo flask web user interface.
jun 19 20:49:33 hydropi gunicorn[605]: [2021-06-19 20:49:33 +0200] [605] [INFO] Starting gunicorn 20.1.0
jun 19 20:49:33 hydropi gunicorn[605]: [2021-06-19 20:49:33 +0200] [605] [INFO] Listening at: unix:/var/run/mycodoflask.sock (605)
jun 19 20:49:33 hydropi gunicorn[605]: [2021-06-19 20:49:33 +0200] [605] [INFO] Using worker: gthread
jun 19 20:49:33 hydropi gunicorn[605]: [2021-06-19 20:49:33 +0200] [626] [INFO] Booting worker with pid: 626
jun 19 20:54:52 hydropi gunicorn[605]: 2021-06-19 20:54:52,664 Setting camera options
jun 19 20:54:52 hydropi gunicorn[605]: 2021-06-19 20:54:52,678 [7f08dbbb-c493-4f6e-928c-8516f20a0eff] Starting camera thread
jun 19 20:54:52 hydropi gunicorn[605]: mmal: mmal_vc_component_create: failed to create component 'vc.ril.camera' (1:ENOMEM)
jun 19 20:54:52 hydropi gunicorn[605]: mmal: mmal_component_create_core: could not create component 'vc.ril.camera' (1)```

kizniche, 2021-06-19:
No need for a video. I just reproduced the issue and am working to figure out a fix. Thanks.

kizniche, 2021-06-19:
I just pushed a fix. [upgrade to master](https://github.com/kizniche/Mycodo/wiki/Updating-to-Github-Master-Branch) and see if the issue persists.

jcastro, 2021-06-19:
Seems like it's fixed, thanks!

closed 2021-08-30 by kizniche