kizniche/Mycodo #1030: SSD1306 128x64 I2C not working

jcastro, 2021-06-19:
### Describe the problem/bug

The screen is not showing anything. I know this screen is working as It was working with an ESP32 a few minutes ago. It's being correctly detected through i2cdetect command as well.

### Versions:

 - Mycodo Version: 8.11.0
 - Raspberry Pi Version: 4
 - Raspbian OS Version: Raspbian Buster lite

### Reproducibility

1. Setup a Method > SSD1306 128x64
2. Add some text to each line to make sure that it should show something
3. Activate the device

### Expected behavior

Screen should show the lines

### Screenshots
[image]
[image]
[image]

After enabling the device the log goes crazy showing this
[image]

kizniche, 2021-06-19:
I need to see the error immediately after activation. The errors you are showing occur after the relevant error has already occurred,

jcastro, 2021-06-19:
Sure!
```
tail -f /var/log/mycodo/mycodo.log 
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_function.py", line 71, in loop
    self.run_function.loop()
AttributeError: 'NoneType' object has no attribute 'loop'
2021-06-19 21:38:43,417 - ERROR - mycodo.controllers.controller_function_5c8783d6 - Exception while running loop(): 'NoneType' object has no attribute 'loop'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_function.py", line 71, in loop
    self.run_function.loop()
AttributeError: 'NoneType' object has no attribute 'loop'
2021-06-19 21:38:43,670 - INFO - mycodo.controllers.controller_function_5c8783d6 - Deactivated in 103.1 ms
2021-06-19 21:39:22,224 - ERROR - mycodo.controllers.controller_function_5c8783d6 - initialize_variables() Exception: No module named 'Adafruit_GPIO'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/base_controller.py", line 74, in run
    self.initialize_variables()
  File "/var/mycodo-root/mycodo/controllers/controller_function.py", line 110, in initialize_variables
    self.run_function = function_loaded.CustomModule(self.function)
  File "/home/pi/Mycodo/mycodo/functions/display_ssd1306_oled_128x64_i2c.py", line 327, in __init__
    self.initialize_variables()
  File "/home/pi/Mycodo/mycodo/functions/display_ssd1306_oled_128x64_i2c.py", line 330, in initialize_variables
    from mycodo.devices.lcd_pioled_circuitpython import PiOLEDCircuitpython
  File "/var/mycodo-root/mycodo/devices/lcd_pioled_circuitpython.py", line 5, in <module>
    import Adafruit_GPIO.SPI as SPI
ModuleNotFoundError: No module named 'Adafruit_GPIO'
2021-06-19 21:39:22,226 - INFO - mycodo.controllers.controller_function_5c8783d6 - Activated in 239.3 ms
2021-06-19 21:39:22,227 - ERROR - mycodo.controllers.controller_function_5c8783d6 - Exception while running loop(): 'NoneType' object has no attribute 'loop'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_function.py", line 71, in loop
    self.run_function.loop()
AttributeError: 'NoneType' object has no attribute 'loop'
2021-06-19 21:39:22,480 - ERROR - mycodo.controllers.controller_function_5c8783d6 - Exception while running loop(): 'NoneType' object has no attribute 'loop'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_function.py", line 71, in loop
    self.run_function.loop()
AttributeError: 'NoneType' object has no attribute 'loop'
2021-06-19 21:39:22,731 - ERROR - mycodo.controllers.controller_function_5c8783d6 - Exception while running loop(): 'NoneType' object has no attribute 'loop'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_function.py", line 71, in loop
    self.run_function.loop()
AttributeError: 'NoneType' object has no attribute 'loop'```

kizniche, 2021-06-19:
I just pushed a fix. [upgrade to master](https://github.com/kizniche/Mycodo/wiki/Updating-to-Github-Master-Branch) and see if the issue persists.

jcastro, 2021-06-19:
I'm now getting another error. This shows up after enabling the display function. Thanks!

```
tail -f /var/log/mycodo/mycodo.log
  File "/var/mycodo-root/mycodo/devices/lcd_pioled_circuitpython.py", line 118, in lcd_write_lines
    draw.text((x, top + 8), message_line_2, font=font, fill=255)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/PIL/ImageDraw.py", line 379, in text
    draw_text(ink)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/PIL/ImageDraw.py", line 349, in draw_text
    **kwargs,
  File "/var/mycodo-root/env/lib/python3.7/site-packages/PIL/ImageFont.py", line 149, in getmask
    return self.font.getmask(text, mode)
UnicodeEncodeError: 'latin-1' codec can't encode character '\u03bc' in position 16: ordinal not in range(256)
2021-06-19 22:25:49,662 - INFO - mycodo.controllers.controller_function_5c8783d6 - Deactivated in 606.7 ms
2021-06-19 22:26:11,673 - DEBUG - mycodo.function.display_ssd1306_oled_128x64_i2c_5c8783d6 - Line sets: [[0, 1, 2, 3, 4, 5, 6, 7]]
2021-06-19 22:26:12,172 - DEBUG - mycodo.function.display_ssd1306_oled_128x64_i2c_5c8783d6 - LCD Function started
2021-06-19 22:26:12,173 - INFO - mycodo.controllers.controller_function_5c8783d6 - Activated in 740.2 ms
2021-06-19 22:26:12,174 - DEBUG - mycodo.function.display_ssd1306_oled_128x64_i2c_5c8783d6 - Channel: 0, Set: 0 Line: 0, 
2021-06-19 22:26:12,452 - DEBUG - mycodo.function.display_ssd1306_oled_128x64_i2c_5c8783d6 - Channel: 1, Set: 0 Line: 1, 
2021-06-19 22:26:12,709 - DEBUG - mycodo.function.display_ssd1306_oled_128x64_i2c_5c8783d6 - Channel: 2, Set: 0 Line: 2, 
2021-06-19 22:26:12,800 - DEBUG - mycodo.function.display_ssd1306_oled_128x64_i2c_5c8783d6 - Channel: 3, Set: 0 Line: 3, 
2021-06-19 22:26:12,802 - DEBUG - mycodo.function.display_ssd1306_oled_128x64_i2c_5c8783d6 - Channel: 4, Set: 0 Line: 4, 
2021-06-19 22:26:12,803 - DEBUG - mycodo.function.display_ssd1306_oled_128x64_i2c_5c8783d6 - Channel: 5, Set: 0 Line: 5, 
2021-06-19 22:26:12,803 - DEBUG - mycodo.function.display_ssd1306_oled_128x64_i2c_5c8783d6 - Channel: 6, Set: 0 Line: 6, 
2021-06-19 22:26:12,804 - DEBUG - mycodo.function.display_ssd1306_oled_128x64_i2c_5c8783d6 - Channel: 7, Set: 0 Line: 7, 
2021-06-19 22:26:13,049 - ERROR - mycodo.controllers.controller_function_5c8783d6 - Exception while running loop(): 'latin-1' codec can't encode character '\u03bc' in position 16: ordinal not in range(256)
Traceback (most recent call last):
  File "/var/mycodo-root/env/lib/python3.7/site-packages/PIL/ImageDraw.py", line 324, in draw_text
    mask, offset = font.getmask2(
AttributeError: 'ImageFont' object has no attribute 'getmask2'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_function.py", line 71, in loop
    self.run_function.loop()
  File "/home/pi/Mycodo/mycodo/functions/display_ssd1306_oled_128x64_i2c.py", line 461, in loop
    lines_display[7])
  File "/var/mycodo-root/mycodo/devices/lcd_pioled_circuitpython.py", line 118, in lcd_write_lines
    draw.text((x, top + 8), message_line_2, font=font, fill=255)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/PIL/ImageDraw.py", line 379, in text
    draw_text(ink)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/PIL/ImageDraw.py", line 349, in draw_text
    **kwargs,
  File "/var/mycodo-root/env/lib/python3.7/site-packages/PIL/ImageFont.py", line 149, in getmask
    return self.font.getmask(text, mode)
UnicodeEncodeError: 'latin-1' codec can't encode character '\u03bc' in position 16: ordinal not in range(256)
^C
```

kizniche, 2021-06-19:
What are your settings? It seems like you're using a non-latin-1 character.

jcastro, 2021-06-19:
I have 4 lines configured, as below:
[image]

kizniche, 2021-06-19:
Try replacing Line 108:

https://github.com/kizniche/Mycodo/blob/59a3a3d8ee3f4990ec3cfbea0a38843ad3f1467e/mycodo/devices/lcd_pioled_circuitpython.py#L108

with:

```python
        font = ImageFont.truetype('/usr/share/fonts/truetype/dejavu//DejaVuSans.ttf')
```

Save, restart the backend, and see if it works and if there are any display issues.

jcastro, 2021-06-19:
that seems to work!

jcastro, 2021-06-19:
btw not sure if related or expected but I've noticed the dashboard is not showing info anymore

[image]

kizniche, 2021-06-19:
Keep issues separated. We should only be discussing the LCD here. To diagnose, you should create a new dashboard, then add one widget at a time until an issue presents.

kizniche, 2021-06-19:
This issue as been resolved, but it will be left open until the next release.

kizniche, 2021-06-19:
BTW, how does the font look on the display? Are there any display issues?

jcastro, 2021-06-19:
> BTW, how does the font look on the display? Are there any display issues?

The font is really small IMHO, even more if you use the smaller 0.9inch version of the SSD1306. On my ESP32 I have it with 4 lines max

It seems there's a "clean screen" command at some point and the whole screen blinks before updating, maybe it's just because the library that this is using but I'm not having that problem using ESPHome (it only refreshes what is changed in the screen)

kizniche, 2021-06-19:
Can you compare the default font (just display text lines, no measurements) with the one you edited the library to use? I don't have this display available to use and would like to see the difference, if you can take a photo of each.

jcastro, 2021-06-19:
Sure! here you have a couple of videos as photos looks a bit weird with the refresh. Notice how small the text is vs the USB ports!

https://user-images.githubusercontent.com/190036/122656079-d58bc900-d157-11eb-9929-b18874fcf843.MOV

https://user-images.githubusercontent.com/190036/122656110-18e63780-d158-11eb-83bd-a8e2f8c96c18.mov

kizniche, 2021-06-19:
Alright, I made some changes that allow the font and font size to be set. This is one of two necessary changes for this to be effective. The next change will involve allowing the 128x64 displays to have 4 lines instead of 8, which will allow the font to be increased above 10 pt without causing an overlap. Can you upgrade to master and see if the new settings work by changing "Font Size" to greater than 10 pt and check the option "Use Non-Default Font".

Try just upgrading and seeing if it works before deactivating and saving different options. I'd like to see if there are any issues from only upgrading. If it then works, try changing the options to see how they affect the display. If everything works well, I'll add another Display Function for your display that has 4 lines and you can use a larger font size.

jcastro, 2021-06-19:
I've upgraded from Master and it looks like even enabling "Use Non-Default Font" setting I'm getting back the old error. I've checked and it's enabled, at least in the front-end

```2021-06-20 00:22:13,450 - INFO - mycodo.controllers.controller_widget - Activated in 1888.4 ms
2021-06-20 00:22:13,596 - ERROR - mycodo.controllers.controller_function_5c8783d6 - Exception while running loop(): 'latin-1' codec can't encode character '\u03bc' in position 16: ordinal not in range(256)
Traceback (most recent call last):
  File "/var/mycodo-root/env/lib/python3.7/site-packages/PIL/ImageDraw.py", line 324, in draw_text
    mask, offset = font.getmask2(
AttributeError: 'ImageFont' object has no attribute 'getmask2'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_function.py", line 71, in loop
    self.run_function.loop()
  File "/home/pi/Mycodo/mycodo/functions/display_ssd1306_oled_128x64_i2c.py", line 493, in loop
    lines_display[7])
  File "/var/mycodo-root/mycodo/devices/lcd_pioled_circuitpython.py", line 126, in lcd_write_lines
    draw.text((x, top + 8), message_line_2, font=font, fill=255)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/PIL/ImageDraw.py", line 379, in text
    draw_text(ink)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/PIL/ImageDraw.py", line 349, in draw_text
    **kwargs,
  File "/var/mycodo-root/env/lib/python3.7/site-packages/PIL/ImageFont.py", line 149, in getmask
    return self.font.getmask(text, mode)
UnicodeEncodeError: 'latin-1' codec can't encode character '\u03bc' in position 16: ordinal not in range(256)
2021-06-20 00:22:14,453 - INFO - mycodo.daemon - Mycodo daemon started in 13.614 seconds
2021-06-20 00:22:14,458 - INFO - mycodo.daemon - 54.91 MB RAM in use
2021-06-20 00:22:21,115 - DEBUG - mycodo.function.display_ssd1306_oled_128x64_i2c_5c8783d6 - Channel: 0, Set: 0 Line: 0, 
2021-06-20 00:22:21,401 - DEBUG - mycodo.function.display_ssd1306_oled_128x64_i2c_5c8783d6 - Channel: 1, Set: 0 Line: 1, 
2021-06-20 00:22:21,752 - DEBUG - mycodo.function.display_ssd1306_oled_128x64_i2c_5c8783d6 - Channel: 2, Set: 0 Line: 2, 
2021-06-20 00:22:21,837 - DEBUG - mycodo.function.display_ssd1306_oled_128x64_i2c_5c8783d6 - Channel: 3, Set: 0 Line: 3, 
2021-06-20 00:22:21,839 - DEBUG - mycodo.function.display_ssd1306_oled_128x64_i2c_5c8783d6 - Channel: 4, Set: 0 Line: 4, 
2021-06-20 00:22:22,122 - DEBUG - mycodo.function.display_ssd1306_oled_128x64_i2c_5c8783d6 - Channel: 5, Set: 0 Line: 5, 
2021-06-20 00:22:22,123 - DEBUG - mycodo.function.display_ssd1306_oled_128x64_i2c_5c8783d6 - Channel: 6, Set: 0 Line: 6, 
2021-06-20 00:22:22,123 - DEBUG - mycodo.function.display_ssd1306_oled_128x64_i2c_5c8783d6 - Channel: 7, Set: 0 Line: 7, 
2021-06-20 00:22:22,627 - ERROR - mycodo.controllers.controller_function_5c8783d6 - Exception while running loop(): 'latin-1' codec can't encode character '\u03bc' in position 16: ordinal not in range(256)
Traceback (most recent call last):
  File "/var/mycodo-root/env/lib/python3.7/site-packages/PIL/ImageDraw.py", line 324, in draw_text
    mask, offset = font.getmask2(
AttributeError: 'ImageFont' object has no attribute 'getmask2'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_function.py", line 71, in loop
    self.run_function.loop()
  File "/home/pi/Mycodo/mycodo/functions/display_ssd1306_oled_128x64_i2c.py", line 493, in loop
    lines_display[7])
  File "/var/mycodo-root/mycodo/devices/lcd_pioled_circuitpython.py", line 126, in lcd_write_lines
    draw.text((x, top + 8), message_line_2, font=font, fill=255)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/PIL/ImageDraw.py", line 379, in text
    draw_text(ink)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/PIL/ImageDraw.py", line 349, in draw_text
    **kwargs,
  File "/var/mycodo-root/env/lib/python3.7/site-packages/PIL/ImageFont.py", line 149, in getmask
    return self.font.getmask(text, mode)```

kizniche, 2021-06-19:
Okay, I forgot you need the option "Use Non-Default Font" if you have non-latin-1 characters (probably the micro symbol).

kizniche, 2021-06-19:
So you're getting that error when not using the default font? Strange. Let me look into it.

kizniche, 2021-06-19:
Just pushed a fix and it should not work. Upgrade to master and give it a try.

jcastro, 2021-06-19:
> Just pushed a fix and it should not work. Upgrade to master and give it a try.

Yes, this one works!. I've tried to change font-size but doesn't seem to have an effect, not sure if it's implemented yet but just wanted to let you know

kizniche, 2021-06-19:
Oops, lol I didn't implement it. I'll add it after I get some other work done.

kizniche, 2021-06-19:
Just implemented the font size option.

jcastro, 2021-06-19:
> Just implemented the font size option.

Just tested it but something is happening, I've changed the size to 14. Pasting the error log below

```2021-06-20 01:49:52,053 - DEBUG - mycodo.function.display_ssd1306_oled_128x64_i2c_5c8783d6 - Channel: 0, Set: 0 Line: 0, 
2021-06-20 01:49:52,295 - DEBUG - mycodo.function.display_ssd1306_oled_128x64_i2c_5c8783d6 - Channel: 1, Set: 0 Line: 1, 
2021-06-20 01:49:52,553 - DEBUG - mycodo.function.display_ssd1306_oled_128x64_i2c_5c8783d6 - Channel: 2, Set: 0 Line: 2, 
2021-06-20 01:49:52,633 - DEBUG - mycodo.function.display_ssd1306_oled_128x64_i2c_5c8783d6 - Channel: 3, Set: 0 Line: 3, 
2021-06-20 01:49:52,634 - DEBUG - mycodo.function.display_ssd1306_oled_128x64_i2c_5c8783d6 - Channel: 4, Set: 0 Line: 4, 
2021-06-20 01:49:52,902 - DEBUG - mycodo.function.display_ssd1306_oled_128x64_i2c_5c8783d6 - Channel: 5, Set: 0 Line: 5, 
2021-06-20 01:49:52,955 - DEBUG - mycodo.function.display_ssd1306_oled_128x64_i2c_5c8783d6 - Channel: 6, Set: 0 Line: 6, 
2021-06-20 01:49:53,006 - DEBUG - mycodo.function.display_ssd1306_oled_128x64_i2c_5c8783d6 - Channel: 7, Set: 0 Line: 7, 
2021-06-20 01:49:53,506 - ERROR - mycodo.controllers.controller_function_5c8783d6 - Exception while running loop(): integer argument expected, got float
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_function.py", line 71, in loop
    self.run_function.loop()
  File "/home/pi/Mycodo/mycodo/functions/display_ssd1306_oled_128x64_i2c.py", line 493, in loop
    lines_display[7])
  File "/var/mycodo-root/mycodo/devices/lcd_pioled_circuitpython.py", line 115, in lcd_write_lines
    font = ImageFont.truetype(font=self.font, size=self.font_size)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/PIL/ImageFont.py", line 852, in truetype
    return freetype(font)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/PIL/ImageFont.py", line 849, in freetype
    return FreeTypeFont(font, size, index, encoding, layout_engine)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/PIL/ImageFont.py", line 210, in __init__
    font, size, index, encoding, layout_engine=layout_engine
TypeError: integer argument expected, got float```

kizniche, 2021-06-19:
It was expecting an integer instead of a float value. I just pushed a fix. Upgrade to master, save the settings once, then activate.

jcastro, 2021-06-20:
It works!

kizniche, 2021-06-20:
Great! I'll make a version with 4 lines and you can experiment and let me know what font size works best for that display, and I'll set that to the default.

kizniche, 2021-06-20:
I just added a 4 line version of the 128x64 SSD1306 Function. Give it a try and let me know if it works and what font size looks the best.

jcastro, 2021-06-20:
> I just added a 4 line version of the 128x64 SSD1306 Function. Give it a try and let me know if it works and what font size looks the best.

Umm I think something is off, I've added a new 4 line function and if I increase the font size above 10pt the line-height won't grow so the text will overlap.

kizniche, 2021-06-20:
Well, I don't have this display with me to test, so development will take a while to get right. Can you provide photos demonstrating what you're describing so I can try to develop a solution to the issue?

jcastro, 2021-06-20:
Sure!
This is 4 lines and 14pt
[image]

kizniche, 2021-06-20:
Thanks. Just pushed a fix that should fix the spacing.

jcastro, 2021-06-20:
Much better!
[image]

kizniche, 2021-06-20:
Nice. What font size is that?

jcastro, 2021-06-20:
Btw noticed the ph value is not right-aligned as the EC... just a small detail tho.

kizniche, 2021-06-21:
The issue with spacing is because it calculates the length of strings based on a mono-spaced font, from the Class I created to handle outputting to mono-spaced LCDs. This gets a lot trickier to determine when we're not using mono-spaced fonts.

jcastro, 2021-06-21:
> The issue with spacing is because it calculates the length of strings based on a mono-spaced font, from the Class I created to handle outputting to mono-spaced LCDs. This gets a lot trickier to determine when we're not using mono-spaced fonts.

oh then it's an easy fix for me, just upload a new font and use that one instead!

jcastro, 2021-06-21:
Oh! almost... I uploaded a new font (SF Mono) but looks like the width is not calculated dynamically, so it will go beyond the screen. This is 12pt, with 14pt was worse of course
[image]

kizniche, 2021-06-21:
The width is set to 20 characters. I'll add an option so the user can change this value.

jcastro, 2021-06-21:
I've manually edited 'display_ssd1306_oled_128x64_i2c_4lines.py' file and changed the characters to 17 with a font size of 12pt and it looks like in the image below. With font at 14pt and setting the right character size there wasn't enough space for the EC to display higher numbers like 1000.0.

[image]

kizniche, 2021-06-21:
I just updated the Function to have an additional option Characters Per Line. Let me know if it works as expected.

kizniche, 2021-06-21:
I should also add an option to choose whether to display the unit or not.

jcastro, 2021-06-21:
> I just updated the Function to have an additional option Characters Per Line. Let me know if it works as expected.
Yes, just tested it and it's working fine!

closed 2021-08-30 by kizniche