kizniche/Mycodo #1031: Widgets on Dashboards not showing any information

jcastro, 2021-06-19:
### Describe the problem/bug

Dashboard widget show up empty in the dashboard

### Versions:

 - Mycodo Version: Master branch from Github repo
 - Raspberry Pi Version: 4
 - Raspbian OS Version: Buster lite

### Reproducibility

1. Add a new test dashboard
2. Add a new widget
3. Widget shows empty

### Expected behavior

Dashboard widgets should show the selected information

### Screenshots
[image]

[image]
[image]

### Additional context
- I've tried to create a new dashboard and add a new widget, it still shows empty
- I've tried to hard-refresh the browser in case there was some old cache from previous version but the problem seems to persist
- I've tried to completely reboot the system
- Couldn't find anything on the logs that shows any related errors

jcastro, 2021-06-19:
Looks like Dashboard is back to normal,  might be related with the testing/version changing I'm doing. I will keep it open until we finish the other bugs just in case something comes up

kizniche, 2021-06-19:
Strange. Not sure what could cause this.

closed 2021-06-20 by kizniche