kizniche/Mycodo #1032: Anyleaf ORP failure to read

meehanchr, 2021-06-21:
Please first read: How to write a good bug report at https://forum.kylegabriel.com/t/how-to-write-a-good-bug-report/71

Please DO NOT OPEN AN ISSUE:

 - If your Mycodo version is not the latest release version, please update your device before submitting your issue (unless your issue is related to not being able to upgrade). Your problem might already be solved.

 - If your issue has been addressed before (i.e., duplicated issue), please ask in the original issue.

Please complete as many of the sections below, if applicable, to provide the most information that may help with investigating your issue. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help those providing assistance by reducing the time required to help you.

### Describe the problem/bug

A clear and concise description of what you were trying to do and what the problem/bug is.

The bug has to do with an error reported for the anyleaf ORP sensor. 
The following error report occurs when reading from the sensor.  The code looks to be trying to run a division operand with failed values.
Error log  is as follows

2021-06-20 21:28:46,237 - ERROR - mycodo.inputs.anyleaf_orp_d9a854d2 - InputModule raised an exception when taking a reading: unsupported operand type(s) for /: 'NoneType' and 'NoneType'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/inputs/base_input.py", line 116, in read
    self._measurements = self.get_measurement()
  File "/home/pi/Mycodo/mycodo/inputs/anyleaf_orp.py", line 147, in get_measurement
    self.value_set(0, self.sensor.read())
  File "/var/mycodo-root/env/lib/python3.7/site-packages/anyleaf/__init__.py", line 261, in read
    self.update()
  File "/var/mycodo-root/env/lib/python3.7/site-packages/anyleaf/__init__.py", line 250, in update
    ORP = self.read_raw()
  File "/var/mycodo-root/env/lib/python3.7/site-packages/anyleaf/__init__.py", line 268, in read_raw
    ORP = orp_from_voltage(chan_orp.voltage, self.cal)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/anyleaf/__init__.py", line 603, in orp_from_voltage
    a = cal.ORP / cal.V
TypeError: unsupported operand type(s) for /: 'NoneType' and 'NoneType'

### Versions:

 - Mycodo Version: [8.11.0]
 - Raspberry Pi Version: [4B]
 - Raspbian OS Version: [Raspbian 10 Buster]

### Reproducibility

Simply activate the anyleaf orp input with the hardware connected properly.

### Expected behavior
Simply returning measurements

### Screenshots

If applicable, add screenshots to help explain your problem.

### Additional context
if i disable the anyleaf orp and active the anyleaf pH i do not get any errors and i do get readings back. If I then disable the pH and active the ORP input then the errors return.

meehanchr, 2021-06-21:
@David-OConnor

David-OConnor, 2021-06-21:
Thank you - will look at it this week. Does the ORP module work using the Python drivers without Mycodo?

meehanchr, 2021-06-21:
I havnt tried that yet.  Will let you know what I find out

meehanchr, 2021-06-21:
Yes, the ORP module does work with the regular python drivers

David-OConnor, 2021-06-25:
Looks like an omission in some setup code. Fixed: https://github.com/kizniche/Mycodo/pull/1039

In the meanwhile, could you please try this fix on your end, and see if it works? Thank you.

kizniche, 2021-06-25:
That automatically closed when I merged the PR. I keep issues open until release, so I reopened.

meehanchr, 2021-06-25:
Yes, confirmed it works on my machine.  Thank you for the fix!!

closed 2021-08-30 by kizniche