kizniche/Mycodo #1033: MH-Z19b Issue.

goldilauchs, 2021-06-21:
I have seen similar issues, but in the past it says they have been patched through updates. And those posts are YEARS old.

### Describe the problem/bug
Co2 sensor works perfectly with my Pi 2. Switch it to my Pi 4 with all the same updated OS and Mycodo. Wont run and gives me error messages.

### Versions:

 - Mycodo Version: [latest version]
 - Raspberry Pi Version: [4]
 - Raspbian OS Version: [Raspian, most recent version]

### Reproducibility

Please list specific setup details that are involved and the steps to reproduce the behavior:

1. Start up Mycodo with the Co2 sensor attached
2. Rx to Tx, Tx to Rx. gnd to gnd, pwer to 5v
3. activate sensor with default uart path.
4. See error

### Expected behavior

I use this sensor on my Pi 2 with no issues. The Pi 4 gives me this. 

Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/base_controller.py", line 74, in run
    self.initialize_variables()
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 317, in initialize_variables
    self.measure_input = input_loaded.InputModule(self.input_dev)
  File "/home/pi/Mycodo/mycodo/inputs/mh_z19b.py", line 130, in __init__
    self.initialize_input()
  File "/home/pi/Mycodo/mycodo/inputs/mh_z19b.py", line 151, in initialize_input
    self.abcoff()
  File "/home/pi/Mycodo/mycodo/inputs/mh_z19b.py", line 199, in abcoff
    self.ser.write(bytearray([0xff, 0x01, 0x79, 0x00, 0x00, 0x00, 0x00, 0x00, 0x86]))
  File "/var/mycodo-root/env/lib/python3.7/site-packages/serial/serialposix.py", line 636, in write
    raise SerialTimeoutException('Write timeout')
serial.serialutil.SerialTimeoutException: Write timeout

kizniche, 2021-06-21:
You're likely not setting the proper device.

goldilauchs, 2021-06-21:
can you elaborate. The input is set to the specific Co2 sensor. I am using all the same default options as the Pi2.
Everything is replicated perfectly between the Pis. Im not sure how to proceed.

goldilauchs, 2021-06-21:
Why respond at all if you have no help to offer?
I'm honestly shocked that you yourself scoffed this off with a one sentence reply giving me nothing to work with.
I've donated in the past and was planning on doing so again as i use this software everyday.
But this response has truly shocked me.

kizniche, 2021-06-21:
It would only work on one and not the other if you didn't change the Device, because serial is set to different default devices on the Pi 2 and Pi 4. What do you have the device set to?

kizniche, 2021-06-21:
I don't have to respond at all. I typed my last response before seeing yours. And frankly, I'm not dissuaded from replying again. I'm currently on vacation and I don't need to respond to any of this.

kizniche, 2021-06-21:
>Why respond at all if you have no help to offer?

You don't think telling you what the likely issue is isn't helping?

goldilauchs, 2021-06-21:
I apologize. Its my fault for relying on software to run my greenhouse. Im frantic and my words were childish.

goldilauchs, 2021-06-21:
the serial is Ama0 for pi 2, and Ama0 for pi4

kizniche, 2021-06-21:
Try /dev/ttyS0

goldilauchs, 2021-06-21:
Warning: Invalid device or improper permissions to read device

kizniche, 2021-06-21:
You used ```raspi-config``` to enable Serial and also answered No enabling login shell over serial?

goldilauchs, 2021-06-21:
im navigating to the docker-compose.yml to ensure permissions is set to true

goldilauchs, 2021-06-21:
no to shell, yes to serial port hardware

kizniche, 2021-06-21:
You shouldn't be running Mycodo in Docker if it's on a Pi.

goldilauchs, 2021-06-21:
i meant, i saw this error in a past post, and you recommended then to edit the docker file permissions.

kizniche, 2021-06-21:
try /dev/serial0 and if that doesn't work, try /dev/serial1. You can also ```ls /dev/serial*``` to see what serial devices there are if 0 or 1 doesn't work.

goldilauchs, 2021-06-21:
everything looks better, all i  get now is a Bad Checksum

kizniche, 2021-06-21:
Bad Checksum for every measurement? That indicates potentially the wrong device.

goldilauchs, 2021-06-21:
**2021-06-21 12:12:06,620 - ERROR - mycodo.inputs.mh_z19b_0e85654c - Bad checksum**

goldilauchs, 2021-06-21:
im getting live data, but its not recording any to a graphable format

kizniche, 2021-06-21:
If it's showing on the Live page, it's being recorded in the database, which the graphs also query from. So it's likely that you may need to refresh your page for the graph to show the data or you might not have the measurement selected for the graph widget.

goldilauchs, 2021-06-21:
thank you so much for the help, and i apologize again for my anger. This software you pioneered has changed my farming life in ways i cannot express. Thank you and i will surely donate again soon.

kizniche, 2021-06-21:
No problem, I know the frustration when things don't work as you expect them to. I'm on vacation and replying on my phone so it's easier to first explain the issue briefly because many times people only need guidance in the right direction. I'd rather not waste both our time writing a detailed response if it's unnecessary.

closed 2021-08-30 by kizniche