kizniche/Mycodo #1036: Feature Request: Setpoint Tracking for Bang-Bang Hysteretic Controller

DarthPleurotus, 2021-06-22:
_**Is your feature request related to a problem? Please describe.**_

I really like the setpoint tracking option for PID Controllers and want to find more ways to control outputs based on time of day. However, I can't find a way to apply the Method I created for setpoint tracking to anything other than a PID setpoint. In particular, I'd like to be able to use the Method with a Bang-Bang Hysteretic Controller, because it seems like some of the outputs I'm trying to control aren't really good candidates for a PID.

_**Describe the solution you'd like**_

A setpoint tracking option for Bang-Bang Hysteretic Controllers

**Describe alternatives you've considered**

I've tried to find a way to include Methods in conditional controllers to achieve a similar result. The only feasible way I could see was to also create a PID controller, enable setpoint tracking for that controller using the Method I created, and then add the PID setpoint as one of the Measurements for the custom Conditional. This seems like an unnecessarily complicated and inefficient solution. Especially if one wants to create multiple controllers this way, you'd need to use multiple "dummy" PIDs that were blank apart from including information on setpoint tracking, creating a lot of unnecessary clutter.

kizniche, 2021-07-06:
It's a bit complex to handle the starting and stopping of using a method in a function, as you can see in the example code I provided at https://forum.kylegabriel.com/t/using-methods-in-conditionals/236/2

You must be able to detect when the function is deactivated/activated by the user and a reboot, since you only want the method restarting (if a duration method) in the former scenario. So building in the ability to select a method to set the bang-bang setpoint isn't as straightforward as it seems. Just commenting to let you know I see this feature request and it's on my radar.

ethicsbro, 2021-07-06:
Thanks for letting me know!

DarthPleurotus, 2021-07-06:
Thanks for letting me know!