kizniche/Mycodo #1037: First time Relay Issue.

GarthButt, 2021-06-22:
First time using a relay, Simply trying to setup an On/Off test. 
I cant even get that right.
Heres the error i get. 

ERROR - mycodo.outputs.on_off_gpio_b505ff69 - Output was unable to be setup on pin 11 with trigger=1: Only '=', '!=', 'is_()', 'is_not()', 'is_distinct_from()', 'is_not_distinct_from()' operators can be used with None/True/False
Traceback (most recent call last):

Ive tried many different GPIO pins. I cant seem to get any of them working. 
Please send help :(

kizniche, 2021-06-23:
Please paste the entire traceback

GarthButt, 2021-06-23:
2021-06-22 16:07:09,056 - ERROR - mycodo.outputs.on_off_gpio_b505ff69 - Output was unable to be setup on pin 25 with trigger=1: Only '=', '!=', 'is_()', 'is_not()', 'is_distinct_from()', 'is_not_distinct_from()' operators can be used with None/True/False
Traceback (most recent call last):
  File "/home/pi/Mycodo/mycodo/outputs/on_off_gpio.py", line 187, in setup_output
    self.check_triggers(self.unique_id, output_channel=0)
  File "/var/mycodo-root/mycodo/outputs/base_output.py", line 552, in check_triggers
    amount > Trigger.output_duration)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/sqlalchemy/sql/operators.py", line 341, in __lt__
    return self.operate(lt, other)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/sqlalchemy/orm/attributes.py", line 301, in operate
    return op(self.comparator, *other, **kwargs)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/sqlalchemy/sql/operators.py", line 341, in __lt__
    return self.operate(lt, other)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/sqlalchemy/orm/properties.py", line 424, in operate
    return op(self.__clause_element__(), *other, **kwargs)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/sqlalchemy/sql/operators.py", line 341, in __lt__
    return self.operate(lt, other)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/sqlalchemy/sql/elements.py", line 831, in operate
    return op(self.comparator, *other, **kwargs)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/sqlalchemy/sql/operators.py", line 341, in __lt__
    return self.operate(lt, other)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/sqlalchemy/sql/type_api.py", line 75, in operate
    return o[0](self.expr, op, *(other + o[1:]), **kwargs)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/sqlalchemy/sql/default_comparator.py", line 92, in _boolean_compare
    "Only '=', '!=', 'is_()', 'is_not()', "
sqlalchemy.exc.ArgumentError: Only '=', '!=', 'is_()', 'is_not()', 'is_distinct_from()', 'is_not_distinct_from()' operators can be used with None/True/False

kizniche, 2021-06-24:
Are there any Triggers in use and if so, can you provide a screenshot of their settings?

Mibu92000, 2021-06-24:
There's several things it still in use Samsung I with an canna get rid of the bar sits in the phone and China figure out how to get rid of the Confederate program that they put in the funk I don't know how to get this completely done but I'm doing with and can't figure it outSent from my Galaxy

GarthButt, 2021-06-24:
[image]

GarthButt, 2021-06-24:
this is the only function im trying to perform, a simple on/off

kizniche, 2021-06-24:
That's the Output page... Is there nothing on the Function page?

GarthButt, 2021-06-24:
no i have not set up any functions. im merely trying to turn on and off an LED now.
do i have to use functions to test the pins activation of relays?

kizniche, 2021-06-25:
I have never seen this issue before and I cannot reproduce the error. I just pushed a change that adds try/except blocks to the trigger checking function that is causing an error in your case. To be clear, this is not a fix for the error you're experiencing, but the error will now no longer prevent the output from operating. I encourage you to read the [New Bug Report Issue Template](https://github.com/kizniche/Mycodo/issues/new?assignees=&labels=&template=problem-bug-report.md) and provide the necessary information that will help in diagnosing and fixing the actual issue. If you can provide this information, do not open a new report, just put the information in a new comment in this issue.

kizniche, 2021-07-02:
Also curious, why did you delete your user GoldiLauchs from #1033 only to create another one for this issue?

closed 2021-08-30 by kizniche