kizniche/Mycodo #1038: Input type cast error when calibrating Atlas PH

Hazmit, 2021-06-23:
### Describe the problem/bug

When performing the calibration process on a new Atlas PH sensor I am receiving an error when entering the temperature of the first solution. It appears to be a type cast problem where the code is saving the value as a str and not formatting it properly. I've tried both using a float and integer value. This hard stops you from being able to perform a calibration.

### Versions:

 - Mycodo Version: 8.11.0
 - Raspberry Pi Version: 3B+
 - Raspbian OS Version: Buster Lite fresh install apt-get updated

### Reproducibility

Please list specific setup details that are involved and the steps to reproduce the behavior:

Perform PH calibration for atlas sensor, input temperature, see error on submit.

### Expected behavior

Expected that it will accept the value and continue with calibration.

### Screenshots

Attached
[image]

kizniche, 2021-06-23:
Duplicate of #1022

closed 2021-06-23 by kizniche

Hazmit, 2021-06-23:
Thanks for marking this so quickly!
Since you marked it I referenced the original bug and was able to calibrate on the input screen! Seems to be a good work around. I see that the other calibration will be removed soon. Appreciate the eyes on the bug report. I owe you a beer.