kizniche/Mycodo #1040: Submit forms without page refresh using AJAX

kizniche, 2021-06-25:
Replace many of the settings forms with AJAX submission to prevent a page reload. Some forms require a page reload, such as those that change available settings based on what has been selected, but most do not.

First, I will work on a proof of concept that works with the currently-used WTForms and Flask.
Next, I will implement for all the simple forms.
Last, I will attempt to refractor forms that change options based on the selection, so all form submissions can operate without refreshing the page.

kizniche, 2021-06-27:
While I was getting this to work for Inputs, I also improved the sorting, so now Inputs can be sorted by drag and drop like the widgets on the dashboard. I also changed the configuration menu from a collapsible element to a modal dialogue. So far, everything is feeling very smooth and I'm sure this will save a lot of time not having to reload the page.

[image]

[image]

[image]

jcastro, 2021-06-27:
So nice! Can’t wait

kizniche, 2021-06-29:
I just had a tour de force learning how to manipulate gridstack and got every single action on the Input page to be able to be manipulated without refreshing the page, without sacrificing any features! I'm so proud of it, I made a short video demo.

https://www.youtube.com/watch?v=MuKDRfjcwGc

jcastro, 2021-06-29:
> I just had a tour de force learning how to manipulate gridstack and got every single action on the Input page to be able to be manipulated without refreshing the page, without sacrificing any features! I'm so proud of it, I made a short video demo.
> 
> https://www.youtube.com/watch?v=MuKDRfjcwGc

awesome, is this launching soon? it will save me a lot of time when doing the final setup.

Also, related to this... sometimes when trying to modify for example an input and I forgot to disable it first, it makes me lose all the changes as the page refreshes. Maybe it would be great to either avoid the refresh and show an alert or actually implement a procedure where if you make changes to an enabled input, it asks if you want to continue as it's enabled and do the actual work for you (disable, apply new changes, enable it again)

kizniche, 2021-06-29:
There are no page refreshes on the Input page any more, so there's not the possibility of your changes being erased.

kizniche, 2021-07-02:
I may be going a little overboard with using ajax, but I made dependency installs not require a page change either.

[image]

[image]

kizniche, 2021-07-04:
Whew! Massive refactor of the Function page to make it completely refresh-less. That seems to be the bulk of what needed changing. Master should be stable now if anyone wants to [upgrade](https://github.com/kizniche/Mycodo/wiki/Updating-to-Github-Master-Branch) and test the new system.

jcastro, 2021-07-04:
@kizniche I've made an upgrade to master and I'm getting the following error after installing and restarting the whole raspberry

```Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/databases/utils.py", line 26, in session_scope
    yield session
  File "/var/mycodo-root/mycodo/utils/database.py", line 74, in db_retrieve_table_daemon
    return_table = return_table.all()
  File "/var/mycodo-root/env/lib/python3.7/site-packages/sqlalchemy/orm/query.py", line 2685, in all
    return self._iter().all()
  File "/var/mycodo-root/env/lib/python3.7/site-packages/sqlalchemy/orm/query.py", line 2823, in _iter
    execution_options={"_sa_orm_load_options": self.load_options},
  File "/var/mycodo-root/env/lib/python3.7/site-packages/sqlalchemy/orm/session.py", line 1670, in execute
    result = conn._execute_20(statement, params or {}, execution_options)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/sqlalchemy/engine/base.py", line 1520, in _execute_20
    return meth(self, args_10style, kwargs_10style, execution_options)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/sqlalchemy/sql/elements.py", line 314, in _execute_on_connection
    self, multiparams, params, execution_options
  File "/var/mycodo-root/env/lib/python3.7/site-packages/sqlalchemy/engine/base.py", line 1399, in _execute_clauseelement
    cache_hit=cache_hit,
  File "/var/mycodo-root/env/lib/python3.7/site-packages/sqlalchemy/engine/base.py", line 1749, in _execute_context
    e, statement, parameters, cursor, context
  File "/var/mycodo-root/env/lib/python3.7/site-packages/sqlalchemy/engine/base.py", line 1930, in _handle_dbapi_exception
    sqlalchemy_exception, with_traceback=exc_info[2], from_=e
  File "/var/mycodo-root/env/lib/python3.7/site-packages/sqlalchemy/util/compat.py", line 211, in raise_
    raise exception
  File "/var/mycodo-root/env/lib/python3.7/site-packages/sqlalchemy/engine/base.py", line 1706, in _execute_context
    cursor, statement, parameters, context
  File "/var/mycodo-root/env/lib/python3.7/site-packages/sqlalchemy/engine/default.py", line 692, in do_execute
    cursor.execute(statement, parameters)
sqlalchemy.exc.OperationalError: (sqlite3.OperationalError) no such column: pid.position_y
[SQL: SELECT pid.id AS pid_id, pid.unique_id AS pid_unique_id, pid.name AS pid_name, pid.position_y AS pid_position_y, pid.is_activated AS pid_is_activated, pid.is_held AS pid_is_held, pid.is_paused AS pid_is_paused, pid.is_preset AS pid_is_preset, pid.log_level_debug AS pid_log_level_debug, pid.preset_name AS pid_preset_name, pid.period AS pid_period, pid.start_offset AS pid_start_offset, pid.max_measure_age AS pid_max_measure_age, pid.measurement AS pid_measurement, pid.direction AS pid_direction, pid.setpoint AS pid_setpoint, pid.band AS pid_band, pid.p AS pid_p, pid.i AS pid_i, pid.d AS pid_d, pid.integrator_min AS pid_integrator_min, pid.integrator_max AS pid_integrator_max, pid.raise_output_id AS pid_raise_output_id, pid.raise_output_type AS pid_raise_output_type, pid.raise_min_duration AS pid_raise_min_duration, pid.raise_max_duration AS pid_raise_max_duration, pid.raise_min_off_duration AS pid_raise_min_off_duration, pid.raise_always_min_pwm AS pid_raise_always_min_pwm, pid.lower_output_id AS pid_lower_output_id, pid.lower_output_type AS pid_lower_output_type, pid.lower_min_duration AS pid_lower_min_duration, pid.lower_max_duration AS pid_lower_max_duration, pid.lower_min_off_duration AS pid_lower_min_off_duration, pid.lower_always_min_pwm AS pid_lower_always_min_pwm, pid.send_lower_as_negative AS pid_send_lower_as_negative, pid.store_lower_as_negative AS pid_store_lower_as_negative, pid.setpoint_tracking_type AS pid_setpoint_tracking_type, pid.setpoint_tracking_id AS pid_setpoint_tracking_id, pid.setpoint_tracking_max_age AS pid_setpoint_tracking_max_age, pid.method_start_time AS pid_method_start_time, pid.method_end_time AS pid_method_end_time, pid.autotune_activated AS pid_autotune_activated, pid.autotune_noiseband AS pid_autotune_noiseband, pid.autotune_outstep AS pid_autotune_outstep 
FROM pid]
(Background on this error at: http://sqlalche.me/e/14/e3q8)
2021-07-04 11:23:36,873 - ERROR - mycodo.database - The Mycodo database is locked. Trying to access again in 1 second...
2021-07-04 11:23:37,904 - ERROR - mycodo.databases.utils - Error raised in session_scope.  Session will be rolled back: db_uri='sqlite:////home/pi/Mycodo/databases/mycodo.db', error='(sqlite3.OperationalError) no such column: pid.position_y
[SQL: SELECT pid.id AS pid_id, pid.unique_id AS pid_unique_id, pid.name AS pid_name, pid.position_y AS pid_position_y, pid.is_activated AS pid_is_activated, pid.is_held AS pid_is_held, pid.is_paused AS pid_is_paused, pid.is_preset AS pid_is_preset, pid.log_level_debug AS pid_log_level_debug, pid.preset_name AS pid_preset_name, pid.period AS pid_period, pid.start_offset AS pid_start_offset, pid.max_measure_age AS pid_max_measure_age, pid.measurement AS pid_measurement, pid.direction AS pid_direction, pid.setpoint AS pid_setpoint, pid.band AS pid_band, pid.p AS pid_p, pid.i AS pid_i, pid.d AS pid_d, pid.integrator_min AS pid_integrator_min, pid.integrator_max AS pid_integrator_max, pid.raise_output_id AS pid_raise_output_id, pid.raise_output_type AS pid_raise_output_type, pid.raise_min_duration AS pid_raise_min_duration, pid.raise_max_duration AS pid_raise_max_duration, pid.raise_min_off_duration AS pid_raise_min_off_duration, pid.raise_always_min_pwm AS pid_raise_always_min_p [... cut ...]  pid.setpoint_tracking_max_age AS pid_setpoint_tracking_max_age, pid.method_start_time AS pid_method_start_time, pid.method_end_time AS pid_method_end_time, pid.autotune_activated AS pid_autotune_activated, pid.autotune_noiseband AS pid_autotune_noiseband, pid.autotune_outstep AS pid_autotune_outstep 
FROM pid]
(Background on this error at: http://sqlalche.me/e/14/e3q8)
2021-07-04 11:23:37,909 - ERROR - mycodo.database - The Mycodo database is locked. Trying to access again in 1 second...
^C
```

jcastro, 2021-07-04:
And if trying to access the web panel I get this
```Error 500: Internal Server Error
Something bad happened but it's probably not your fault. Letting the developers know about these issues is crucial to supporting Mycodo. Please submit a new issue on GitHub with the following diagnostic information and error traceback (copy the entire traceback):

Version: 8.11.0
Database: bdc03b708ac6
Model: Raspberry Pi 4 Model B Rev 1.1

Release:
Distributor ID: Raspbian
Description: Raspbian GNU/Linux 10 (buster)
Release: 10
Codename: buster

Firmware:
May 27 2021 14:03:06
Copyright (c) 2012 Broadcom
version 7d9a298cda813f747b51fe17e1e417e7bf5ca94d (clean) (release) (start_x)

Error (Full Traceback):

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/engine/base.py", line 1706, in _execute_context
    cursor, statement, parameters, context
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/engine/default.py", line 692, in do_execute
    cursor.execute(statement, parameters)
sqlite3.OperationalError: no such column: conditional.position_y

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_restx/api.py", line 652, in error_router
    return original_handler(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_login/utils.py", line 272, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_page.py", line 611, in page_dashboard
    conditional = Conditional.query.all()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/orm/query.py", line 2685, in all
    return self._iter().all()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/orm/query.py", line 2823, in _iter
    execution_options={"_sa_orm_load_options": self.load_options},
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/orm/session.py", line 1670, in execute
    result = conn._execute_20(statement, params or {}, execution_options)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/engine/base.py", line 1520, in _execute_20
    return meth(self, args_10style, kwargs_10style, execution_options)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/sql/elements.py", line 314, in _execute_on_connection
    self, multiparams, params, execution_options
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/engine/base.py", line 1399, in _execute_clauseelement
    cache_hit=cache_hit,
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/engine/base.py", line 1749, in _execute_context
    e, statement, parameters, cursor, context
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/engine/base.py", line 1930, in _handle_dbapi_exception
    sqlalchemy_exception, with_traceback=exc_info[2], from_=e
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/util/compat.py", line 211, in raise_
    raise exception
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/engine/base.py", line 1706, in _execute_context
    cursor, statement, parameters, context
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/engine/default.py", line 692, in do_execute
    cursor.execute(statement, parameters)
sqlalchemy.exc.OperationalError: (sqlite3.OperationalError) no such column: conditional.position_y
[SQL: SELECT conditional.id AS conditional_id, conditional.unique_id AS conditional_unique_id, conditional.name AS conditional_name, conditional.position_y AS conditional_position_y, conditional.is_activated AS conditional_is_activated, conditional.log_level_debug AS conditional_log_level_debug, conditional.conditional_statement AS conditional_conditional_statement, conditional.conditional_status AS conditional_conditional_status, conditional.period AS conditional_period, conditional.start_offset AS conditional_start_offset, conditional.pyro_timeout AS conditional_pyro_timeout, conditional.message_include_code AS conditional_message_include_code, conditional.custom_options AS conditional_custom_options 
FROM conditional]
(Background on this error at: http://sqlalche.me/e/14/e3q8)```

kizniche, 2021-07-04:
Did you recently upgrade to master while I was working on the new form submission system? If so, that is the issue. If not, then read on...

Please attach the excerpt of the upgrade log for your upgrade.

Looks like your database didn't upgrade when you updated to master. I upgraded 2 of my systems without issue, so I'm not sure what the issue is. Can you restore your previous version, then upgrade again? You can restore from the command line with mycodo_backup_restore.sh. You must enter the full path to a backup, e.g. ```sudo ~Mycodo/mycodo/scripts/mycodo_backup_restore.sh /var/Mycodo-backups/backup-to-restore```

https://github.com/kizniche/Mycodo/blob/master/mycodo/scripts/mycodo_backup_restore.sh

kizniche, 2021-07-04:
If you did upgrade some time in the middle of this feature addition, you will need to restore to a point in time before and then upgrade to master again.

jcastro, 2021-07-04:
> Did you recently upgrade to master while I was working on the new form submission system? If so, that is the issue. If not, then read on...
>
> Please attach the excerpt of the upgrade log for your [...]

Hey! No, this is the first time I updated so I moved from 8.11 stable to this master repo. I will try to do the restore and try again. Thanks!

kizniche, 2021-07-04:
Can you attach the upgrade log excerpt? That will likely show what the issue in the upgrade was.

jcastro, 2021-07-04:
> Can you attach the upgrade log excerpt? That will likely show what the issue in the upgrade was.

Sure, but I'm not sure where to find it. Do you have any steps that I can follow to get it?

kizniche, 2021-07-04:
It's located at /var/log/mycodo/mycodoupgrade.log but don't sttach the entire thing, it includes all upgrades. Save the file locally, then prune all other upgrades from it before attaching.

kizniche, 2021-07-04:
Do you need assistance getting the file from your Pi?

jcastro, 2021-07-04:
[mycodoupgrade.log](https://github.com/kizniche/Mycodo/files/6760015/mycodoupgrade.log)
Here it is! Btw now that I'm thinking about it I've upgraded from master before to install and tests the fixes you did for the OLED screen, etc. Probably it was that then.

kizniche, 2021-07-04:
Does your upgrade just end abruptly on line 235 or did you accidentally cut the log there?

jcastro, 2021-07-04:
No that's the last thing on the log, I'm now restoring to a previous version

jcastro, 2021-07-04:
alright I got it back! I'm going to try to upgrade to master again, unless I need to do something else in between to upgrade the database?

kizniche, 2021-07-04:
I just checked the commit log and if you upgraded after the OLED fix was pushed, then that was the issue, because that was in the middle of this feature addition.

> I'm going to try to upgrade to master again, unless I need to do something else in between to upgrade the database?

Nope, as long as you restored to a version before the feature addition (e.g. don't go back 1 version, which was in th emiddle of the feature addition, go back 2 versions).

jcastro, 2021-07-04:
When there's a new public release available will it work? Or do I still have to go back to a point where I didn't upgraded from master? Asking this because I have quite a few changes so I only want to revert it if it won't even work using a new public release version. thanks!

kizniche, 2021-07-04:
This is the issue. Normally, for every time I want to change the database, such as add a column to a table, I create a new script, so when I push that, any user who upgrades to master, gets that new script and it will upgrade their database. For example:

Script A will upgrade database to version ABC1, which adds the column position_y to table output.
Script B will upgrade database to version ABC2, which adds the column position_y to table function.

The user can upgrade to master after script A is pushed, or after script B is pushed, and both scripts will run, and their database will always end up at version ABC2 with potision_y columns on both the output and function tables.

However, if I create and push the following script:

Script A will upgrade database to version ABC1, which adds the column position_y to table output.

Then a user upgrades to master, and their DB is now at ABC1 with position_y in the output table.

Then I alter that script to be the following and push it to github:

Script A will upgrade database to version ABC1, which adds the column position_y to table output and table function.

When the user then upgrades to master again, the script will not run, because the user's DB is already at version ABC1 (DB scripts will not rerun). This is what I learned a long time ago that users will always upgrade to master without my instruction, so I need to always create a new upgrade script for every time I want to push to master, otherwise this issue can occur. Sometimes changing the script is more convenient than creating a new one, and if I can edit and push it quick enough, there's little chance someone could upgrade to master in the time between the pushes. I happened to have changed the script because I thought I would be pushing soon, but it ended up being a huge amount of work that lasted over a day.

> When there's a new public release available will it work?

If you restored back to a point before the feature, then your upgrade system will work perfectly.

jcastro, 2021-07-04:
I see! I will keep this in mind for future testing.

I'm not restoring to a version that I think it's the correct one, but is there a better way to check this is indeed the release version and not the master repo upgrade?

kizniche, 2021-07-04:
It can be any prior version, release or master upgrade, just make sure the timestamp of the backup is before Jul 1

jcastro, 2021-07-04:
> It can be any prior version, release or master upgrade. When you upgraded to master that caused the issue, a backup was made of your system before the upgrade. Restore that backup version and you should be fine.

ah nice, then it was a backup from today that's when I upgraded to master to test the new changes. Thanks Kyle!

kizniche, 2021-07-04:
>I will keep this in mind for future testing.

I normally don't tell people not to upgrade to master anymore when it's crucial they don't, but whenever I specifically say to upgrade to master, it is likely safe. So if in doubt, ask if it's okay to upgrade to master if you want to and I don't state to do so myself.

jcastro, 2021-07-05:
Updated to master without problems, loving the new changes. Thanks!

kizniche, 2021-07-06:
Great. Let me know if you find any issues. This will need a significant amount of testing before release, so I'd like to catch as many bugs as possible. There are a few more pages I want to implement this, including the Dashboard (which I anticipate posing issues) and settings pages.

kizniche, 2021-07-06:
Also any suggestions are welcome. This is only the first layout/styles I came up with, and there are a few style/layout tweaks I still want to make (such as the PID entry buttons are crowded in mobile view), but I like hearing how things can be improved.

jcastro, 2021-07-06:
Hey Kyle! I actually found a bug on the relay module on dashboard, when trying to reduce its size it will have a weird behavior. I can’t send a video because I’m on holidays but let me know if you can’t reproduce it

kizniche, 2021-07-06:
I was meaning bugs related to this refreshless form submission feature. I haven't touched the Dashboard yet.

jcastro, 2021-07-13:
This is the bug https://d.pr/v/jpiSwy

kizniche, 2021-07-14:
I came across that bug recently and fixed it, but not sure where in the commits.

jcastro, 2021-07-14:
nice! I will just upgrade from Master then, unless that's unsafe DB-wise

kizniche, 2021-07-14:
Should be good at the moment.

jcastro, 2021-07-14:
✅  Fixed!

kizniche, 2021-07-23:
I think I've come to the conclusion that it's going to be way too complex to apply this to the dashboard. Loading/unloading various javascript code that's needed for widget types and individual widgets has been a nightmare. So I'm going to leave the dashboard as it is, but with a few code cleanups I've done along the way.

closed 2021-08-30 by kizniche