kizniche/Mycodo #1041: Camera stream issue

joaozorro, 2021-06-27:
### Describe the problem/bug

Streaming from the pi camera freezes the whole system 

### Versions:

 - Mycodo Version: [8.11.0]
 - Raspberry Pi Version: [4]
 - Raspbian OS Version: [Raspian]

### Reproducibility

Please list specific setup details that are involved and the steps to reproduce the behavior:

1. Setup pi camera for streaming 
2. start streaming
3. try to navigate somewhere 
4. Nothing happens and at some point the web seems to lose the communication with the daemon
5. the system becomes completely irresponsive

### Expected behavior

being able to navigate.

### Screenshots

[image]
[image]

### Additional context

Is there anything that should be added to make it easier to address this issue?

kizniche, 2021-06-27:
Try reducing the frame rate to 1 FPS

joaozorro, 2021-06-27:
same issue. any other ideas?

kizniche, 2021-06-27:
Which camera and how long is your camera ribbon cable?

joaozorro, 2021-06-27:
Pi Camera, not very long. see the image attached 
[image]

kizniche, 2021-06-28:
try streaming with the opencv library instead of picamera

closed 2021-08-30 by kizniche