kizniche/Mycodo #1042: AtlasScientificI2C raised an exception when initializing: [Errno 2] No such file or directory: '/dev/i2c-1'

Ikzy, 2021-06-30:
Following your hydroponics tutorial I'm trying to set new i2c addresses for my EZO_PMP's  .. unfortunately I'm encountering an error: 

pi@raspberrypi:~ $ ~/Mycodo/env/bin/python ~/Mycodo/mycodo/devices/atlas_scientific_i2c.py
AtlasScientificI2C raised an exception when initializing: [Errno 2] No such file or directory: '/dev/i2c-1'
Traceback (most recent call last):
  File "/home/pi/Mycodo/mycodo/devices/atlas_scientific_i2c.py", line 35, in __init__
    self.file_read = io.open("/dev/i2c-" + str(i2c_bus), "rb", buffering=0)
FileNotFoundError: [Errno 2] No such file or directory: '/dev/i2c-1'
>> Atlas Scientific sample code
>> Any commands entered are passed to the board via I2C except:
>>   List_addr lists the available I2C addresses.
>>   Address,xx changes the I2C address the Raspberry Pi communicates with.
>>   Poll,xx.x command continuously polls the board every xx.x seconds
 where xx.x is longer than the 1.50 second timeout.
>> Pressing ctrl-c will stop the polling
Enter command: list_addr
Traceback (most recent call last):
  File "/home/pi/Mycodo/mycodo/devices/atlas_scientific_i2c.py", line 187, in <module>
    main()
  File "/home/pi/Mycodo/mycodo/devices/atlas_scientific_i2c.py", line 141, in main
    devices = device.list_i2c_devices()
  File "/home/pi/Mycodo/mycodo/devices/atlas_scientific_i2c.py", line 116, in list_i2c_devices
    self.set_i2c_address(i)
  File "/home/pi/Mycodo/mycodo/devices/atlas_scientific_i2c.py", line 61, in set_i2c_address
    fcntl.ioctl(self.file_read, i2c_slave, addr)
AttributeError: 'AtlasScientificI2C' object has no attribute 'file_read'

Here's my setup.log. 
[setup.log](https://github.com/kizniche/Mycodo/files/6743108/setup.log)

please help :)

kizniche, 2021-06-30:
Did you enable I2C with ```raspi-config```?

Ikzy, 2021-06-30:
ok! I totally missed that in the tutorial! (if it's in there). 

thanks for the Ultra quick feedback!

closed 2021-06-30 by Ikzy