kizniche/Mycodo #1043: Input mqtt_paho_json should handle nested json

maecki-maecki, 2021-06-30:
My temperature sensors run on tasmota-flashed ESP8266 mcu.
Tasmota publishes the sensor readings in the following form to mqtt:
`{
  "TempUnit": "C",
  "Time": "2021-06-30T20:47:47",
  "DS18B20-2": {
    "Id": "031797798E9C",
    "Temperature": 21.2
  },
  "DS18B20-1": {
    "Id": "0317221700FF",
    "Temperature": 18.5
  }
} `

I would like to parse both sensors into two channels of the input, which I didn‘t succeed to because the input just is able to select a top-level key directly as a value.

Possible solutions that I see:
- use flatten library to flatten json bevor parsing and use a key like `DS18B20-1_Temperature` as json_name in the above example
- use jmespath library and use a path selector like  `DS18B20-1.Temperature` as json_name

maecki-maecki, 2021-07-01:
Opened PR #1044

maecki-maecki, 2021-07-01:
replaced with PR #1045

kizniche, 2021-07-02:
This issue has been mentioned on **DIY Automation Forum**. There might be relevant details there:

https://forum.kylegabriel.com/t/best-practices-for-off-pi-inputs-measurements/143/5

maecki-maecki, 2021-07-24:
Fixed & committed?

closed 2021-08-30 by kizniche