kizniche/Mycodo #1046: OLED display doesn‘t show measurements

maecki-maecki, 2021-07-02:
I am on today‘s  master version since I want to use SSD3306 displays (128x64, i2c, 4 lines)

Text, date, ip-address work but ‚last measurement‘ does not but says ‚ERROR‘ instead.
Logfile:

`
2021-07-02 16:19:24,509 - ERROR - mycodo.function.display_ssd1306_oled_128x64_i2c_4lines_592f1c2a - Error generating channel 0 line: 'display_unit'
`

Seems ‚display_unit‘ has not been stored in channel  config or is not retrievable (if I replace line 466 in the inputfile with a ‚True‘, then it works)

maecki-maecki, 2021-07-02:
Regression?
Your commit #1524fd4994721d8d8d5a3b61aadeb949257979c1  for #1030 introduced the line that throws the exception ….

kizniche, 2021-07-02:
Thanks for the bug report. Just pushed a fix. It was supposed to be ```self.options_channels['display_unit'][current_channel]``` not ```self.options_channels['select_measurement'][current_channel]['display_unit']```

maecki-maecki, 2021-07-02:
Thank you for the lightning fast fix, fix works for me!

kizniche, 2021-07-04:
I try to avoid these situations that lead to database issues by always creating new DB upgrade scripts, but this time I didn't and as a result, anyone that upgraded to master in a small window will have critical DB issues when upgrading next. If you did upgrade to master to apply this fix, you will likely not be able to upgrade in the future unless you restore to a previous version before this, then upgrade again. Read more about this [here](https://github.com/kizniche/Mycodo/issues/1040#issuecomment-873553311).

closed 2021-08-30 by kizniche