kizniche/Mycodo #1047: mycodo daemon crashes with error in Pyro5 lib

maecki-maecki, 2021-07-04:
I am running on current Master version and had two crashes in the last days that seem like the Pyro endpoint not listening anymore …

`pi@raspberrypi:/var/log/mycodo $ journalctl -u mycodo
-- Logs begin at Sat 2021-07-03 01:23:42 CEST, end at Sun 2021-07-04 10:30:34 CEST. --
Jul 03 08:57:01 raspberrypi mycodo_daemon[23832]: Caught signal 15. Stopping daemon.
Jul 03 08:57:01 raspberrypi mycodo_daemon[23832]: Stopping daemon.
Jul 03 08:57:05 raspberrypi python[21834]: 2021-07-03 08:57:05,529 [Remote command] Terminate daemon...
Jul 03 08:57:05 raspberrypi python[21834]: Traceback (most recent call last):
Jul 03 08:57:05 raspberrypi python[21834]:   File "/var/mycodo-root/env/lib/python3.7/site-packages/Pyro5/client.py", line 282, in connect_and_handshake
Jul 03 08:57:05 raspberrypi python[21834]:     sslContext=sslContext)
Jul 03 08:57:05 raspberrypi python[21834]:   File "/var/mycodo-root/env/lib/python3.7/site-packages/Pyro5/socketutil.py", line 283, in create_socket
Jul 03 08:57:05 raspberrypi python[21834]:     sock.connect(connect)
Jul 03 08:57:05 raspberrypi python[21834]: ConnectionRefusedError: [Errno 111] Connection refused
Jul 03 08:57:05 raspberrypi python[21834]: The above exception was the direct cause of the following exception:
Jul 03 08:57:05 raspberrypi python[21834]: Traceback (most recent call last):
Jul 03 08:57:05 raspberrypi python[21834]:   File "/var/mycodo-root/mycodo/mycodo_client.py", line 630, in <module>
Jul 03 08:57:05 raspberrypi python[21834]:     if daemon.terminate_daemon():
Jul 03 08:57:05 raspberrypi python[21834]:   File "/var/mycodo-root/mycodo/mycodo_client.py", line 153, in terminate_daemon
Jul 03 08:57:05 raspberrypi python[21834]:     return self.proxy().terminate_daemon()
Jul 03 08:57:05 raspberrypi python[21834]:   File "/var/mycodo-root/env/lib/python3.7/site-packages/Pyro5/client.py", line 92, in __getattr__
Jul 03 08:57:05 raspberrypi python[21834]:     self._pyroGetMetadata()
Jul 03 08:57:05 raspberrypi python[21834]:   File "/var/mycodo-root/env/lib/python3.7/site-packages/Pyro5/client.py", line 358, in _pyroGetMetadata
Jul 03 08:57:05 raspberrypi python[21834]:     self.__pyroCreateConnection()
Jul 03 08:57:05 raspberrypi python[21834]:   File "/var/mycodo-root/env/lib/python3.7/site-packages/Pyro5/client.py", line 342, in __pyroCreateConnection
Jul 03 08:57:05 raspberrypi python[21834]:     connect_and_handshake(conn)
Jul 03 08:57:05 raspberrypi python[21834]:   File "/var/mycodo-root/env/lib/python3.7/site-packages/Pyro5/client.py", line 304, in connect_and_handshake
Jul 03 08:57:05 raspberrypi python[21834]:     raise errors.CommunicationError(err) from x
Jul 03 08:57:05 raspberrypi python[21834]: Pyro5.errors.CommunicationError: cannot connect to ('127.0.0.1', 9090): [Errno 111] Connection refused
Jul 03 08:57:06 raspberrypi systemd[1]: mycodo.service: Control process exited, code=exited, status=1/FAILURE
Jul 03 08:57:06 raspberrypi systemd[1]: mycodo.service: Failed with result 'exit-code'.
pi@raspberrypi:/var/log/mycodo $  `

maecki-maecki, 2021-07-04:
As a somewhat crude workaround I let systemd restart the daemon on crashes, with the following additions to mycodo.service: 
```
….
StartLimitIntervalSec=500
StartLimitBurst=5

[Service]
Restart=on-failure
RestartSec=5s
….
```

kizniche, 2021-07-04:
Never seen that before. You may have a filesystem issue. I'd reinstall the OS and Mycodo and see if that resolves it.

maecki-maecki, 2021-07-04:
Really?

I‘ll give the mycodo reinstall a try as soon as it works again to switch to master, but why should I reinstall raspbian when mycodo crashes?

kizniche, 2021-07-04:
I've never seen the issue before, and there are currently hundreds (probably thousands) of Mycodo instances currently in the wild, so if you're the only one experiencing this issue, it could be due to a hardware issue (faulty SD card/filesystem corruption).

maecki-maecki, 2021-07-04:
Could be but not the most probable cause.
As mentioned, maybe my mycodo installation is corrupted due to the master install … I‘ll retry as soon as I can get to master again or to a working oled … (After release installation from scartch, upgrade to master is still broken ‚Something bad happened …‘)

kizniche, 2021-07-04:
upgrading to master could not cause that issue.

kizniche, 2021-07-04:
I would do a stepwise set of tests.. Reinstall Mycodo. If that doesn't work. Reinstall OS and Mycodo. Take the path of least resistance.

maecki-maecki, 2021-07-04:
Hey, sorry to  take your time, but rereading the journalctl output above tells me this might be a completely different topic:
who sends a SIGTERM to the daemon at a time I definitly wasn‘t worrking on my raspberry … ?

But thinking through what happens leads me to the following:

The above error is probably triggered by the mycodo.service file:

- ExecStop using mycodo_client is executed but not waited for ( TimeoutSec=0 ), Systemd then sends a SIGTERM to the process
- mycodo_client still tries to connect the then already terminated daemon (4 Seconds after SIGTERM in the above example)
- Because of this error, systemd marks the service as failed instead of restarting it  

So there is a hickup in systemd file, I would recommend using a  `TimeoutStopSec=5` or something alike to give the mycodo_client some time to do its work before systemd takes over killing the process.

Remaining question is: Who did trigger my mycodo.service stop or restart?
My first suspect was restart_daemon.py but I can see no entry in mycodokeepup.log

So I keep searching for the one stopping or restarting my service …

kizniche, 2021-07-04:
Signal 15 is system reboot

maecki-maecki, 2021-07-04:
Hey Kyle,

There is no such thing as a signal for reboot.

Signal 15 is SIGTERM and is used to terminate processes gracefully, like systemd does 

***@***.***:~ $ kill -l
 1) SIGHUP       2) SIGINT       3) SIGQUIT      4) SIGILL       5) SIGTRAP
 6) SIGABRT      7) SIGBUS       8) SIGFPE       9) SIGKILL     10) SIGUSR1
11) SIGSEGV     12) SIGUSR2     13) SIGPIPE     14) SIGALRM     15) SIGTERM
…

Additionally, my pi has not booted for 3 days

***@***.***:~ $ uptime
 22:17:56 up 3 days,  9:31,  1 user,  load average: 0,28, 0,22, 0,22

Kind regards
Maecki

maecki-maecki, 2021-07-04:
Until 5 seconds before the crash, I have data activity in my daemon.log:
```
…

Jul  3 08:56:57 raspberrypi influxd[2506]: [httpd] ::1 - mycodo [03/Jul/2021:08:56:57 +0200] "POST /write?db=mycodo_db HTTP/1.1" 204 0 "-" "python-requests/2.25.1" db95b1f7-dbcb-11eb-9de0-b827ebe84621 17843
Jul  3 08:56:57 raspberrypi influxd[2506]: [httpd] ::1 - mycodo [03/Jul/2021:08:56:57 +0200] "POST /write?db=mycodo_db HTTP/1.1" 204 0 "-" "python-requests/2.25.1" dbd5c606-dbcb-11eb-9de1-b827ebe84621 12479
Jul  3 08:57:00 raspberrypi influxd[2506]: [httpd] ::1 - mycodo [03/Jul/2021:08:57:00 +0200] "POST /write?db=mycodo_db HTTP/1.1" 204 0 "-" "python-requests/2.25.1" ddaed3c5-dbcb-11eb-9de2-b827ebe84621 14261
Jul  3 08:57:05 raspberrypi python[21834]: 2021-07-03 08:57:05,529 [Remote command] Terminate daemon...
Jul  3 08:57:05 raspberrypi python[21834]: Traceback (most recent call last):
Jul  3 08:57:05 raspberrypi python[21834]:   File "/var/mycodo-root/env/lib/python3.7/site-packages/Pyro5/client.py", line 282, in connect_and_handshake
Jul  3 08:57:05 raspberrypi python[21834]:     sslContext=sslContext)
…
``

kizniche, 2021-07-04:
Which Raspberry Pi OS release?

maecki-maecki, 2021-07-04:
```
pi@raspberrypi:/etc/cron.daily $ cat /etc/issue
Raspbian GNU/Linux 10 \n \l

pi@raspberrypi:/etc/cron.daily $ uname -a
Linux raspberrypi 5.10.17-v7+ #1421 SMP Thu May 27 13:59:01 BST 2021 armv7l GNU/Linux
```

kizniche, 2021-07-04:
And the OS release?

maecki-maecki, 2021-07-04:
Raspbian 10 (buster) as mentioned above

kizniche, 2021-07-04:
Have you tried a new SD card?

kizniche, 2021-07-04:
Have you already deleted all inputs/outputs/functions and confirmed the behavior still exists?

maecki-maecki, 2021-07-04:
No crash since yesterday morning …

kizniche, 2021-07-04:
Are you using any atypical configurations or custom code?

maecki-maecki, 2021-07-04:
just the mqtt input I built the PR for

maecki-maecki, 2021-07-06:
```
…
Jul  6 01:48:09 raspberrypi influxd[26918]: ts=2021-07-05T23:48:09.389141Z lvl=info msg="Executing query" log_id=0V8XskcW000 service=query query="SELECT last(value) FROM mycodo_db.autogen.C WHERE device_id = 'bac23a50-6732-4ff1-bcba-1f35b70d5a0c' AND channel = '0' AND measure = 'temperature' AND time > now() - 6m"
Jul  6 01:48:09 raspberrypi influxd[26918]: [httpd] ::1 - mycodo [06/Jul/2021:01:48:09 +0200] "GET /query?db=mycodo_db&q=SELECT+LAST%28value%29+FROM+C+WHERE+device_id%3D%27bac23a50-6732-4ff1-bcba-1f35b70d5a0c%27+AND+channel%3D%270%27+AND+measure%3D%27temperature%27+AND+time+%3E+now%28%29+-+360s HTTP/1.1" 200 58 "-" "python-requests/2.25.1" 73bf018f-ddeb-11eb-917a-b827ebe84621 2159
Jul  6 01:48:16 raspberrypi mycodo_daemon: Caught signal 15. Stopping daemon.
Jul  6 01:48:16 raspberrypi mycodo_daemon: Stopping daemon.
Jul  6 01:48:20 raspberrypi python[5094]: 2021-07-06 01:48:20,209 [Remote command] Terminate daemon...
Jul  6 01:48:20 raspberrypi python[5094]: Traceback (most recent call last):
Jul  6 01:48:20 raspberrypi python[5094]:   File "/var/mycodo-root/env/lib/python3.7/site-packages/Pyro5/client.py", line 282, in connect_and_handshake
Jul  6 01:48:20 raspberrypi python[5094]:     sslContext=sslContext)
Jul  6 01:48:20 raspberrypi python[5094]:   File "/var/mycodo-root/env/lib/python3.7/site-packages/Pyro5/socketutil.py", line 283, in create_socket
Jul  6 01:48:20 raspberrypi python[5094]:     sock.connect(connect)
Jul  6 01:48:20 raspberrypi python[5094]: ConnectionRefusedError: [Errno 111] Connection refused
Jul  6 01:48:20 raspberrypi python[5094]: The above exception was the direct cause of the following exception:
Jul  6 01:48:20 raspberrypi python[5094]: Traceback (most recent call last):
…
```

kizniche, 2021-07-06:
Can you disable all inputs/functions and delete all outputs and see if the issue persists?

maecki-maecki, 2021-07-06:
Switched to a virgin installation (release version) on fresh sd this morning. Without the changes in master, I will see whether the problem persists …

maecki-maecki, 2021-07-07:
;-(
``` 
Jul  7 16:10:12 raspberrypi mycodo_daemon: Caught signal 15. Stopping daemon.
Jul  7 16:10:12 raspberrypi mycodo_daemon: Stopping daemon.
Jul  7 16:10:16 raspberrypi python[1101]: 2021-07-07 16:10:16,660 [Remote command] Terminate daemon...
Jul  7 16:10:16 raspberrypi python[1101]: Traceback (most recent call last):
Jul  7 16:10:16 raspberrypi python[1101]:   File "/var/mycodo-root/env/lib/python3.7/site-packages/Pyro5/client.py", line 282, in connect_and_handshake
Jul  7 16:10:16 raspberrypi python[1101]:     sslContext=sslContext)
Jul  7 16:10:16 raspberrypi python[1101]:   File "/var/mycodo-root/env/lib/python3.7/site-packages/Pyro5/socketutil.py", line 283, in create_socket
Jul  7 16:10:16 raspberrypi python[1101]:     sock.connect(connect)
Jul  7 16:10:16 raspberrypi python[1101]: ConnectionRefusedError: [Errno 111] Connection refused
Jul  7 16:10:16 raspberrypi python[1101]: The above exception was the direct cause of the following exception:
Jul  7 16:10:16 raspberrypi python[1101]: Traceback (most recent call last):
Jul  7 16:10:16 raspberrypi python[1101]:   File "/var/mycodo-root/mycodo/mycodo_client.py", line 630, in <module>
Jul  7 16:10:16 raspberrypi python[1101]:     if daemon.terminate_daemon():
Jul  7 16:10:16 raspberrypi python[1101]:   File "/var/mycodo-root/mycodo/mycodo_client.py", line 153, in terminate_daemon
Jul  7 16:10:16 raspberrypi python[1101]:     return self.proxy().terminate_daemon()
Jul  7 16:10:16 raspberrypi python[1101]:   File "/var/mycodo-root/env/lib/python3.7/site-packages/Pyro5/client.py", line 92, in __getattr__
Jul  7 16:10:16 raspberrypi python[1101]:     self._pyroGetMetadata()
Jul  7 16:10:16 raspberrypi python[1101]:   File "/var/mycodo-root/env/lib/python3.7/site-packages/Pyro5/client.py", line 358, in _pyroGetMetadata
Jul  7 16:10:16 raspberrypi python[1101]:     self.__pyroCreateConnection()
Jul  7 16:10:16 raspberrypi python[1101]:   File "/var/mycodo-root/env/lib/python3.7/site-packages/Pyro5/client.py", line 342, in __pyroCreateConnection
Jul  7 16:10:16 raspberrypi python[1101]:     connect_and_handshake(conn)
Jul  7 16:10:16 raspberrypi python[1101]:   File "/var/mycodo-root/env/lib/python3.7/site-packages/Pyro5/client.py", line 304, in connect_and_handshake
Jul  7 16:10:16 raspberrypi python[1101]:     raise errors.CommunicationError(err) from x
Jul  7 16:10:16 raspberrypi python[1101]: Pyro5.errors.CommunicationError: cannot connect to ('127.0.0.1', 9090): [Errno 111] Connection refused
Jul  7 16:10:17 raspberrypi systemd[1]: mycodo.service: Control process exited, code=exited, status=1/FAILURE
Jul  7 16:10:17 raspberrypi systemd[1]: mycodo.service: Failed with result 'exit-code'.
Jul  7 16:10:41 raspberrypi kernel: [  314.641251] w1_master_driver w1_bus_master1: Attaching one wire slave 00.a00000000000 crc af
Jul  7 16:10:41 raspberrypi kernel: [  314.651777] w1_master_driver w1_bus_master1: Family 0 for 00.a00000000000.af is not registered.
Jul  7 16:11:19 raspberrypi kernel: [  352.561849] w1_master_driver w1_bus_master1: Attaching one wire slave 00.600000000000 crc 65
Jul  7 16:11:19 raspberrypi kernel: [  352.576349] w1_master_driver w1_bus_master1: Family 0 for 00.600000000000.65 is not registered.
```

kizniche, 2021-07-07:
Weird! Do you happen to have a different Pi to test if it's a hardware issue?

kizniche, 2021-07-07:
I'll do a fresh install on a Pi 3 this evening and see if it suffers from the same issue.

maecki-maecki, 2021-07-07:
I have a  passively cooled pi4 but didn’t wanna use it because it gets warmer than  the pi 3 when continously on ….

kizniche, 2021-07-07:
All my Pis are passively cooled, without heatsinks, and never had a problem. Several Pi 4s and a Compute Module 4, too.

maecki-maecki, 2021-07-07:
Don‘t need the power or the 8G RAM either for two PIDs in mycodo, a mosquitto and an nginx …

maecki-maecki, 2021-07-08:
I setup a trap with auditd to see who initiates the SIGTERM, hope to thus gain some intel …

maecki-maecki, 2021-07-08:
Okay I found the offender by the hints auditd gave me:

Bug can be triggered by a single Input module of type ‘bash command (mycodo)’ and give it a commandline of `sleep 300;echo 1` 

The timeout this brings kills the daemon.
My suspect is  the timer killing a whole processgroup in cmd_output of `mycodo/utils/system_pi.py` 

How is this triggered in my system?
I have a ‘bash command‘ input with the command line `curl -sS http://tasmota-temperatur-1/cm?cmnd=status%2010 | jq -r '.StatusSNS."DS18B20-1".Temperature' ` 
That seemingly doesn’t always return in time ….

kizniche, 2021-07-09:
It's likely this:

https://github.com/kizniche/Mycodo/blob/9d6e803f26b76d89c771c0ae39f980927750b250/mycodo/utils/system_pi.py#L542

This was added to prevent processes from running indefinitely. I suppose it wasn't thoroughly tested. It's supposed to only kill the thread. I'm open to suggestions for how to fix/improve it.

maecki-maecki, 2021-07-09:
Why kill the process group? Is os.kill(cmd.pid) not enough?

BTW.: I am leaving for a one-week vacation tonight, so I will not be on this topic for that time.

kizniche, 2021-07-09:
Fix (change to ```os.kill(cmd.pid, signal.SIGTERM)```) tested and committed. Thanks for the great investigatory work and suggested change.

maecki-maecki, 2021-07-09:
Thanks goes to you: incredible commitment to mycodo and a whole lot of work! I will test the fix when I am back…

closed 2021-08-30 by kizniche